espressif/esp-idf #3340: Example getting Started Station dont work

arnold-b, 2019-04-21:
Hello 

The simple example from GettingStarted Station don´t connect to wifi. 
I also tried a few other examples none connect to wifi.

I´m using VisualStudio and the current master   
If i try arduino example it works. It means it can not be a hardware problem. 

what can i do ?

thanks

I (0) cpu_start: App cpu up.
I (374) heap_init: Initializing. RAM available for dynamic allocation:
I (381) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (387) heap_init: At 3FFB8E98 len 00027168 (156 KiB): DRAM
I (393) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (400) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (406) heap_init: At 4008FCB8 len 00010348 (64 KiB): IRAM
I (412) cpu_start: Pro cpu start user code
I (95) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (163) wifi station: ESP_WIFI_MODE_STA
I (173) wifi: wifi driver task: 3ffc0824, prio:23, stack:3584, core=0
I (173) wifi: wifi firmware version: 7240fb7
I (173) wifi: config NVS flash: enabled
I (173) wifi: config nano formating: disabled
I (173) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (183) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (223) wifi: Init dynamic tx buffer num: 32
I (223) wifi: Init data frame dynamic rx buffer num: 32
I (223) wifi: Init management frame dynamic rx buffer num: 32
I (223) wifi: Init management short buffer num: 32
I (233) wifi: Init static rx buffer size: 1600
I (233) wifi: Init static rx buffer num: 10
I (243) wifi: Init dynamic rx buffer num: 32
I (333) phy: phy_version: 4100, 2a5dd04, Jan 23 2019, 21:00:07, 0, 0
I (343) wifi: mode : sta (84:0d:8e:3a:73:a8)
I (343) wifi station: wifi_init_sta finished.
I (343) wifi station: connect to ap SSID:xxx password:$12345$
I (2393) wifi station: retry to connect to the AP
I (2393) wifi station: connect to the AP fail
I (4443) wifi station: retry to connect to the AP
I (4443) wifi station: connect to the AP fail
I (6493) wifi station: retry to connect to the AP
I (6493) wifi station: connect to the AP fail
I (8543) wifi station: retry to connect to the AP
I (8553) wifi station: connect to the AP fail
I (10603) wifi station: retry to connect to the AP

arnold-b, 2019-04-24:
Hello, 
its strange. Till now i played with 3.2 and everything is ok. Now i gave IDF Master (4) a new try and Wifi every example works. Strange. I would like to know the problem. I´m afraid it happend again. 

thanks

closed 2019-04-24 by arnold-b

tangqingcai, 2019-05-09:
@arnold-b hello;  I met the same issue likely;
How about your issue ?
You change the ESP-IDF version to 3.2 ,then the example work is ok?

simonliu009, 2019-07-26:
I ran into the same issue today. The same ssid/password are working fine in idf 3.2/3.3 but not in idf 4.0.  I fixed it by erasing the flash.  `esptool.py erase_flash`