espressif/esp-idf #3342: how to send data from SD card to android via BT classic.

ROYMURLI, 2019-04-22:
I tried to transfer the data which is about 140 KB from SD card to android via BT classic. But after transferring 4 KB it crashes. 
So, in "case ESP_SPP_DATA_IND_EVT: " I am sending GETLOG from android via BT classsic and I used   different task "task_getlog" for fetching from file. Presently I am trying by copying 1024 bytes from 1 file to another file to fetch, I also tried with the original file (means not using of another file) but in both the cases it crashed.
Debug logs is below:-
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6380
load:0x40078000,len:11152
load:0x40080400,len:6692
entry 0x40080764
[0;32mI (28) boot: ESP-IDF v3.3-beta1-136-g97eecfa1b-dirty 2nd stage bootloader[0m
[0;32mI (28) boot: compile time 13:16:55[0m
[0;32mI (29) boot: Enabling RNG early entropy source...[0m
[0;32mI (35) boot: SPI Speed      : 40MHz[0m
[0;32mI (39) boot: SPI Mode       : DIO[0m
[0;32mI (43) boot: SPI Flash Size : 16MB[0m
[0;32mI (47) boot: Partition Table:[0m
[0;32mI (51) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (58) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (65) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (73) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (80) boot:  3 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (88) boot:  4 ota_0            OTA app          00 10 00110000 00100000[0m
[0;32mI (95) boot:  5 ota_1            OTA app          00 11 00210000 00100000[0m
[0;32mI (103) boot: End of partition table[0m
[0;32mI (107) boot: Defaulting to factory image[0m
[0;32mI (112) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2f9a8 (194984) map[0m
[0;32mI (189) esp_image: segment 1: paddr=0x0003f9d0 vaddr=0x3ff80000 size=0x00000 (     0) load[0m
[0;32mI (189) esp_image: segment 2: paddr=0x0003f9d8 vaddr=0x3ff80000 size=0x00000 (     0) load[0m
[0;32mI (196) esp_image: segment 3: paddr=0x0003f9e0 vaddr=0x3ffbdb60 size=0x00630 (  1584) load[0m
[0;32mI (205) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x91564 (595300) map[0m
[0;32mI (422) esp_image: segment 5: paddr=0x000d1584 vaddr=0x3ffbe190 size=0x02780 ( 10112) load[0m
[0;32mI (426) esp_image: segment 6: paddr=0x000d3d0c vaddr=0x3ffc0910 size=0x00000 (     0) load[0m
[0;32mI (429) esp_image: segment 7: paddr=0x000d3d14 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (438) esp_image: segment 8: paddr=0x000d411c vaddr=0x40080400 size=0x13aec ( 80620) load[0m
[0;32mI (480) esp_image: segment 9: paddr=0x000e7c10 vaddr=0x400c0000 size=0x00000 (     0) load[0m
[0;32mI (480) esp_image: segment 10: paddr=0x000e7c18 vaddr=0x50000000 size=0x00000 (     0) load[0m
[0;32mI (487) esp_image: segment 11: paddr=0x000e7c20 vaddr=0x50000000 size=0x00000 (     0) load[0m
[0;32mI (507) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (507) boot: Disabling RNG early entropy source...[0m
[0;32mI (510) cpu_start: Pro cpu up.[0m
[0;32mI (513) cpu_start: Application information:[0m
[0;32mI (518) cpu_start: Project name:     bt_spp_vfs_acceptor_demo[0m
[0;32mI (525) cpu_start: App version:      b4d1559-dirty[0m
[0;32mI (530) cpu_start: Compile time:     13:57:53[0m
[0;32mI (535) cpu_start: Compile date:     Apr 19 2019[0m
[0;32mI (540) cpu_start: ESP-IDF:          v3.3-beta1-136-g97eecfa1b-dirty[0m
[0;32mI (547) cpu_start: Starting app cpu, entry point is 0x400811f4[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (558) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (565) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
[0;32mI (571) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM[0m
[0;32mI (577) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM[0m
[0;32mI (583) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM[0m
[0;32mI (589) heap_init: At 3FFD7FF8 len 00008008 (32 KiB): DRAM[0m
[0;32mI (595) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (602) heap_init: At 3FFE4350 len 00013CB0 (79 KiB): D/IRAM[0m
[0;32mI (608) heap_init: At 40093EEC len 0000C114 (48 KiB): IRAM[0m
[0;32mI (614) cpu_start: Pro cpu start user code[0m
[0;32mI (184) esp_apptrace: Initialized TRAX on CPU0[0m
[0;32mI (185) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (28) esp_apptrace: Initialized TRAX on CPU1[0m
[0;32mI (4) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (898) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (1112) BTDM_INIT: BT controller compile version [0b60040][0m
[0;32mI (1113) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[... 214 lines cut ...]

                if(nofile == true)
                {
                    esp_spp_write(param->data_ind.handle, 6, "NOFILE");
                    nofile = false;
                    upload_log = false;
                }
                vTaskDelay(DELAY_IN_MS(10));
            }
        }
        break;
    case ESP_SPP_CONG_EVT:
        break;
    case ESP_SPP_WRITE_EVT:
        ESP_LOGI(SPP_TAG, "ESP_SPP_WRITE_EVT");
        ESP_LOGI(SPP_TAG, "ESP_SPP_WRITE_EVT len=%d cong=%d", param->write.len , param->write.cong);
        break;  
    case ESP_SPP_SRV_OPEN_EVT:
        ESP_LOGI(SPP_TAG, "ESP_SPP_SRV_OPEN_EVT");
        esp_spp_write(param->write.handle, 4, "PASS");  
        break;
    default:
        break;
    }
}

blueMoodBHD, 2019-04-25:
@ROYMURLI Could you please provide the `.elf` file. Note that the `.elf` file needs to match the log.

ROYMURLI, 2019-04-25:
Thanks for reply, The .elf file is attached here. 
[applicaton.zip](https://github.com/espressif/esp-idf/files/3116877/applicaton.zip)

blueMoodBHD, 2019-04-26:
According to the log, there may be a pointer error in function `esp_spp_cb`, causing a crash.

ROYMURLI, 2019-04-26:
In callback 
case ESP_SPP_DATA_IND_EVT:
printf("got a read event, writing %d values\n", write_ptr);
param->data_ind.data[param->data_ind.len-2] = '\0';
printf("%d bytes | %s\n",param->data_ind.len, (char *) param->data_ind.data);
//
//
//(After couple of different condition for different functions)
//
//
/*This is for "GETLOG" command which I send over serial BT to get data. One task is also running which I already mentioned above*/ 
else if(strstr((char *)param->data_ind.data, "GETLOG") != NULL)
        {
            recording = false;
            write_log = false;
            wait = false;
            upload_log = true;
            getlog_variable = true;
            nofile = false;
            
            while(upload_log)
            {   
                if(log_spp_write == true)
                {
                    printf("hi7 \n");
                    printf("bufFile %s",bufFile);
                    esp_spp_write(param->data_ind.handle, strlen(bufFile), (uint8_t *) bufFile);
                    memset(bufFile, 0, 1024);
                    log_spp_write = false;
            /*        if((k+1)%4==0)
                    {   
                        printf("getlog 1 \n");
                        break;
                    }*/
                    printf("hi7 \n");
                }

                if(nofile == true)
                {
                    esp_spp_write(param->data_ind.handle, 6, "NOFILE");
                    nofile = false;
                    upload_log = false;
                }
                vTaskDelay(DELAY_IN_MS(10));
            }
        }
        break;

Can you please tell me where I am stuck. I tried mny times by removing some elements and adding some in code but in all the case after 4 KB of transfer it crashed.

ROYMURLI, 2019-04-26:
Task for getlog:-
void task_getlog(void *args)
{
while(1)
{
if(getlog_variable == true)
{
char bufFile1[1024];
int n;
long sz = 0;
int total = 0;
int remainder = 0;
FILE *fp;
FILE *fp1;
fp = fopen("/sdcard/dozee.log", "rb");
printf("\n%s buffers:\n", "/sdcard/dozee.log");
if (fp == NULL)
{
printf("Error, file %s not found.", "/sdcard/dozee.log");
recording = true;
nofile = true;
vTaskDelay(DELAY_IN_MS(10));
upload_log = false;
}
else
{
fseek(fp, 0L, SEEK_END);
sz = ftell(fp);
total = sz/1024;
remainder = sz%1024;
if(remainder>0)
{
total = total + 1;
printf("total is : %d",total);
}
fseek(fp, 0L, SEEK_SET);
ESP_LOGI(TAG, ">> Start sending file:%s", fileName);
int i = 0;
while(upload_log)
{
printf("hi1 \n");
for(k=0; k<total; k++)
{
printf("k = %d \n",k);
printf("hi2 \n");
FILE *fp1;
fp1 = fopen("/sdcard/tmp.log", "w");
if(fp1==NULL)
{
fclose(fp1);
vTaskDelay(DELAY_IN_MS(100));
fp1 = fopen("/sdcard/tmp.log", "w+");
printf("hi3 \n");
}
if(fp1==NULL)
{
printf("sadness");
}
memset(bufFile1,0, 1024);
vTaskDelay(DELAY_IN_MS(10));
fread(bufFile1,1,1024,fp);
vTaskDelay(DELAY_IN_MS(50));
printf("hi41 \n");
fwrite(bufFile1,1024,1,fp1);
vTaskDelay(DELAY_IN_MS(50));
fclose(fp1);
printf("hi4 \n");

                        fp1 = fopen("/sdcard/tmp.log", "r");
                        memset(bufFile, 0, 1024);
                        vTaskDelay(DELAY_IN_MS(10));
                        n = fread(bufFile, 1, 1024, fp1);
                        i++;
                        printf("n is %d - i is %d \n", n, i);
                        log_spp_write = true;
                        while(log_spp_write)
                        {
                            vTaskDelay(DELAY_IN_MS(1));
                        }
                        printf("hi5 \n");
                        printf("heapsize %d \n",xPortGetFreeHeapSize());
                        remove("/sdcard/tmp.log");
                        fseek(fp,1024+(k*1024),SEEK_SET);
                        printf("hi6 \n");

                        if(k == total-1)
                        {
                            ESP_LOGI(TAG, "<< Done sending file:%s", fileName);
                            vTaskDelay(DELAY_IN_MS(50));
                            recording= true;
                            upload_log = false;
                            remove("/sdcard/dozee.log");
                            printf("file deleted \n");
                        }

                    /*if((k+1)%4==0 && k!=total-1)
                    {
                        wait=true;
                        printf("grt 1 \n");
                        while(wait)
                            vTaskDelay(DELAY_IN_MS(1));
                    }*/
                }
            }      
        }
        getlog_variable = false;
    }
vTaskDelay(DELAY_IN_MS(10));
}
}

blueMoodBHD, 2019-04-26:
It seems like that, you fopen `fp` each times but without fclose. After some times, there will be many opened file point. Causing `fp1` open fail , then `fp1` is null and causing crash.

ROYMURLI, 2019-04-26:
Ok, In "task_getlog" before going into "for loop" the FILE fp is opened once and in "for loop" I opened fp1 each time because I am copying the 1024 bytes from fp to fp1 and transferring the fp1 data via BT classic and then delete that fp1 file accordingly. After successful completion of file I deleted fp file also. So, where should I close the fp file in this task. Actually before this I only used fp file and used fread and fseek to set the pointer to read and tranfer in the same way but it was unsuccesssful too.

ROYMURLI, 2019-04-26:
Should I close fp when I am deleting fp1 and then open inside the for loop?

blueMoodBHD, 2019-04-26:
To figure out whether this cause the crash or not, may be you could check if `fp1` is NULL after open file each time. If opening file failed do cause the crash. Please make sure close the file after use it.

ROYMURLI, 2019-04-26:
Yes, You are correct. But every time before opening fp1 I remov the fp1 inside the loop. For the first four time it opened correctly but 5th time when I tried to open it, it didn't open and crashed. For cross check I gave condition to print "sadness". It prints it on 5th time. As per the debug logs. I think if I am deleting the fp1 each time there is no need of closing the file. I will try with closing the file and deleting it also.
for(k=0; k<total; k++)
{
printf("k = %d \n",k);
printf("hi2 \n");
FILE *fp1;
fp1 = fopen("/sdcard/tmp.log", "w");
if(fp1==NULL)
{
fclose(fp1);
vTaskDelay(DELAY_IN_MS(100));
fp1 = fopen("/sdcard/tmp.log", "w+");
printf("hi3 \n");
}
if(fp1==NULL)
{
printf("sadness");
}

blueMoodBHD, 2019-04-26:
Maybe you don't need to open the file so many times, if you just want send data to SPP. Just as follows:
```
EILF *fp = fopen(...);
if (!fp) {
    //ERROR
}
while(1){
    /*
   * your code for  GETLOG or others
   */
    n = fread(bufFile, 1, 1024, fp);
    esp_spp_write(handle, n, bufFile);
    ...
    ...
    ...
    if(/*file end*/) {
       break;
    }
}
fclose(fp);
```

ROYMURLI, 2019-04-26:
Thanks for your suggestion. I deleted the task and did all the coding inside the callback. I only used one file which is already present in the SD card directly. 
[debug.txt](https://github.com/espressif/esp-idf/files/3120576/debug.txt)
 

else if(strstr((char *)param->data_ind.data, "GETLOG") != NULL)
        {
            recording = false;
            write_log = false;

            bool upload_log = true;
            bool file_complete = false;
	        char bufFile[1024];
            int n;
	        long sz = 0;
	        FILE *fp;
	        fp = fopen("/sdcard/dozee.log", "rb");
	        printf("\n%s buffers:\n", "/sdcard/dozee.log");
	        if (fp == NULL) 
            {
		        printf("Error, file %s not found.", "/sdcard/dozee.log");
                recording = true;
                upload_log = false;
                esp_spp_write(param->data_ind.handle, 6, "NOFILE");
            }
            else
            {
                fseek(fp, 0L, SEEK_END);
                sz = ftell(fp);
                fseek(fp, 0L, SEEK_SET);
                ESP_LOGI(TAG, ">> Start sending file:%s", fileName);
                int i = 0;
                while(upload_log)
                {
                    if (!file_complete) 
                    {
                        if ((n = fread(bufFile, 1, 1024, fp)) > 0) 
                        {
                            i++;
                            printf(" %d - %d\n", n, i);
                            esp_spp_write(param->data_ind.handle, strlen(bufFile), (uint8_t *) bufFile);
                            fseek(fp,1024+((i-1)*1024),SEEK_SET);
                        } 
                        else 
                        {
                            file_complete = true;
                        }
                    } 
                    else 
                    {
                        ESP_LOGI(TAG, "<< Done sending file:%s", fileName);
                        vTaskDelay(DELAY_IN_MS(50));
                        recording= true;
                        upload_log = false;
                        fclose(fp);
                        remove("/sdcard/dozee.log");
                        printf("file deleted \n");
                        
                    }
                }     
           
            }
        }

It is still crashing. The debug log is attached below.

ROYMURLI, 2019-04-26:
This is inside "case ESP_SPP_DATA_IND_EVT:"

blueMoodBHD, 2019-04-26:
No,  you could call `esp_spp_write` only once in a `ESP_SPP_DATA_IND_EVT`,  suggest to open file in other place such as `ESP_SPP_SRV_OPEN_EVT`. Then read a piece of file and send it(only once) in  `ESP_SPP_DATA_IND_EVT`.

ROYMURLI, 2019-04-26:
But the file is about 350 KB. So, If I want to send the file I have to send it in chunks repetedly of 1 KB in a loop. I am sending command "GETLOG" from my app (currently through a python script) to get the data. So I used "(strstr((char *)param->data_ind.data, "GETLOG") != NULL)" inside ESP_SPP_DATA_IND_EVT. If I open the file in "ESP_SPP_SRV_OPEN_EVT" then also I have to use 'esp_spp_write' inside the ESP_SPP_DATA_IND_EVT because it is totally depends upon upcoming command. If apps send "GETLOG" command esp should send whole file and delete at last.

blueMoodBHD, 2019-04-26:
When you get  "GETLOG", you can sent 1KB data, Then, you will get a `ESP_SPP_WRITE_EVT`, it means the write is completed, you can set anther 1KB here. Then you will get anther `ESP_SPP_WRITE_EVT`... Loop this until send all data.  Pay attention to the congest, if it is congest, you should wait until it is not congest, just like the demo.

ROYMURLI, 2019-04-26:
Yeah Got it, Thanks for your help. SPP_CONG_EVT comes several times but automatically it is completed. thanks again.

closed 2019-04-29 by ROYMURLI