espressif/esp-idf #3346: pyelftools requirement

KaeLL, 2019-04-22:
I refuse to believe I'm the only one affected by this annoyance. Plus the fact that suggested ways to fix it involve related issues with msys' pacman, which is useless because of some error with duplicated database records.

So... help? I'm using the latest version of the master branch. Here's the build output.

make -j all 
Toolchain path: /c/msys32/opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
The following Python requirements are not satisfied:
pyelftools>=0.22
Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.
Alternatively, you can run "C:/msys32/mingw32/bin/python.exe -m pip install --user -r C:/Users/devfw01/eclipse-workspace/esp-idf\\requirements.txt" for resolving the issue.
make: *** [C:\Users\devfw01\eclipse-workspace\esp-idf/make/project.mk:245: check_python_dependencies] Error 1

JoaoBerlese, 2019-04-23:
Hello @KaeLL =]

I had the same issue.
To find the problem, I decide to restart the "Installation Step by Step".
I corrected it by doing the following ([Get Started link](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#get-started-setup-toolchain)):

Step 1. Set up the Toolchain >> Install Prerequisites (linux) >> Ubuntu and Debian:

```
sudo apt-get install gcc git wget make libncurses-dev flex bison gperf python python-pip python-setuptools python-serial python-cryptography python-future python-pyparsing python-pyelftools
```

After this command, everything worked again.

dobairoland, 2019-04-23:
Hi @KaeLL!

When a new functionality is added which depends on a new Python library then one encounters similar failure to yours. The error message tells you to install the highlighted package in order to get the new functionality. This is an expected behavior for now. We are working on an improved approach which will install such packages in the background. Sorry for the inconvenience and please be patient while this new feature becomes available.

Until then, you can solve your issue by running:
```
C:/msys32/mingw32/bin/python.exe -m pip install --user -r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt
```
or
```
pip install --user pyelftools
```

Does this solve your issue?

Note: Usually we recommend to install Python packages with `pacman` but it doesn't have a `pyelftool` package. Therefore, it should be installed with `pip`.

KaeLL, 2019-04-23:
@JoaoBerlese I'm using Windows.
@dobairoland It didn't work. Like I said, it's already installed. Here's the output:

	C:\Windows\system32>C:/msys32/mingw32/bin/python.exe -m pip install --user -r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt
	DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
	Requirement already satisfied: setuptools in c:\msys32\mingw32\lib\python2.7\site-packages (from -r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 4)) (41.0.0)
	Requirement already satisfied: pyserial>=3.0 in c:\msys32\mingw32\lib\python2.7\site-packages (from -r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 8)) (3.4)
	Requirement already satisfied: future>=0.15.2 in c:\msys32\mingw32\lib\python2.7\site-packages (from -r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 9)) (0.16.0)
	Requirement already satisfied: cryptography>=2.1.4 in c:\msys32\mingw32\lib\python2.7\site-packages (from -r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 10)) (2.3.1)
	Requirement already satisfied: pyparsing>=2.0.3 in c:\msys32\mingw32\lib\python2.7\site-packages (from -r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 11)) (2.2.0)
	Requirement already satisfied: idna>=2.1 in c:\msys32\mingw32\lib\python2.7\site-packages (from cryptography>=2.1.4->-r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 10)) (2.7)
	Requirement already satisfied: asn1crypto>=0.21.0 in c:\msys32\mingw32\lib\python2.7\site-packages (from cryptography>=2.1.4->-r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 10)) (0.24.0)
	Requirement already satisfied: six>=1.4.1 in c:\msys32\mingw32\lib\python2.7\site-packages (from cryptography>=2.1.4->-r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 10)) (1.11.0)
	Requirement already satisfied: cffi!=1.11.3,>=1.7 in c:\msys32\mingw32\lib\python2.7\site-packages (from cryptography>=2.1.4->-r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 10)) (1.10.0)
	Requirement already satisfied: enum34 in c:\msys32\mingw32\lib\python2.7\site-packages (from cryptography>=2.1.4->-r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 10)) (1.1.6)
	Requirement already satisfied: ipaddress in c:\msys32\mingw32\lib\python2.7\site-packages (from cryptography>=2.1.4->-r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 10)) (1.0.22)
	Requirement already satisfied: pycparser in c:\msys32\mingw32\lib\python2.7\site-packages (from cffi!=1.11.3,>=1.7->cryptography>=2.1.4->-r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 10)) (2.19)

dobairoland, 2019-04-23:
@KaeLL I don't see any `pyelftools` in your output. I don't understand how that could happen. That means that our checker can see it in `C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt` but `pip` cannot.

Could you please make sure that you don't have any modifications done on the `requirements.txt` file since you've posted the issue (`cd $IDF_PATH && git checkout -- requirements.txt`)?

KaeLL, 2019-04-23:
Apologies. I ran the cmd but my esp-idf was on the 4.0-dev which works fine, not on the master branch.
Here's the output after checking out to master branch:
```
C:\Users\devfw01>C:/msys32/mingw32/bin/python.exe -m pip install --user -r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Requirement already satisfied: setuptools in c:\msys32\mingw32\lib\python2.7\site-packages (from -r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 4)) (41.0.0)
Requirement already satisfied: pyserial>=3.0 in c:\msys32\mingw32\lib\python2.7\site-packages (from -r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 8)) (3.4)
Requirement already satisfied: future>=0.15.2 in c:\msys32\mingw32\lib\python2.7\site-packages (from -r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 9)) (0.16.0)
Requirement already satisfied: cryptography>=2.1.4 in c:\msys32\mingw32\lib\python2.7\site-packages (from -r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 10)) (2.3.1)
Requirement already satisfied: pyparsing>=2.0.3 in c:\msys32\mingw32\lib\python2.7\site-packages (from -r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 11)) (2.2.0)
Requirement already satisfied: pyelftools>=0.22 in c:\users\devfw01\.local\lib\python2.7\site-packages (from -r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 12)) (0.25)
Requirement already satisfied: idna>=2.1 in c:\msys32\mingw32\lib\python2.7\site-packages (from cryptography>=2.1.4->-r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 10)) (2.7)
Requirement already satisfied: asn1crypto>=0.21.0 in c:\msys32\mingw32\lib\python2.7\site-packages (from cryptography>=2.1.4->-r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 10)) (0.24.0)
Requirement already satisfied: six>=1.4.1 in c:\msys32\mingw32\lib\python2.7\site-packages (from cryptography>=2.1.4->-r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 10)) (1.11.0)
Requirement already satisfied: cffi!=1.11.3,>=1.7 in c:\msys32\mingw32\lib\python2.7\site-packages (from cryptography>=2.1.4->-r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 10)) (1.10.0)
Requirement already satisfied: enum34 in c:\msys32\mingw32\lib\python2.7\site-packages (from cryptography>=2.1.4->-r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 10)) (1.1.6)
Requirement already satisfied: ipaddress in c:\msys32\mingw32\lib\python2.7\site-packages (from cryptography>=2.1.4->-r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 10)) (1.0.22)
Requirement already satisfied: pycparser in c:\msys32\mingw32\lib\python2.7\site-packages (from cffi!=1.11.3,>=1.7->cryptography>=2.1.4->-r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 10)) (2.19)

```
And I'm not sure if it's still needed, but here's the `requirements.txt` file:
```
C:\Windows\system32>pip install --user pyelftools
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Requirement already satisfied: pyelftools in c:\users\devfw01\.local\lib\python2.7\site-packages (0.25)

C:\Windows\system32>type %IDF_PATH%\requirements.txt
# This is a list of python packages needed for ESP-IDF. This file is used with pip.
# Please see the Get Started section of the ESP-IDF Programming Guide for further information.
#
setuptools
# The setuptools package is required to install source distributions and on some systems is not installed by default.
# Please keep it as the first item of this list.
#
pyserial>=3.0
future>=0.15.2
cryptography>=2.1.4
pyparsing>=2.0.3
pyelftools>=0.22
```

negativekelvin, 2019-04-24:
Requirement already satisfied: pyparsing>=2.0.3 in **c:\msys32\mingw32\lib\python2.7**\site-packages (from -r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 11)) (2.2.0)
Requirement already satisfied: pyelftools>=0.22 in **c:\users\devfw01\.local\lib\python2.7**\site-packages (from -r C:/Users/devfw01/eclipse-workspace/esp-idf\requirements.txt (line 12)) (0.25)

dobairoland, 2019-04-24:
Yes, @negativekelvin is right and you have your pyelftools installed elsewhere (`c:\users...`).

@KaeLL, can you please try to reinstall it?
```
c:\msys32\mingw32\bin\python -m pip uninstall pyelftools
c:\msys32\mingw32\bin\python -m pip install pyelftools
```
Now `pyelftools` should be in the right place:
`Requirement already satisfied: pyelftools in c:\msys32\mingw32\lib\python2.7\site-packages (0.25)`

If this doesn't help then please share the output of `echo %PATH%`.

KaeLL, 2019-04-24:
@dobairoland it worked, though I'm now having another issue with the lack of a rule to build some files. But that's for another thread. Thanks.

dobairoland, 2019-04-25:
I'm happy that it got solved! Please feel free to open another issue if you experience some other (unrelated) issue.

closed 2019-04-25 by dobairoland