espressif/esp-idf #3348: Wifi Problem

arnold-b, 2019-04-23:
IDF 3.2 release

Hello, 

who can tell me how i can find my problem. 

All Versions over 3.2 i can not get any WIFI connection. 3.2 Worked for 2 Days and suddenly i could not get any connection again without any change. i didn´t succeed to solve this, but after 1 hour suddenly it worked again. 

Other Wifi Samples are ok. 

I´m using fritzbox as accesspoint. 

static esp_err_t event_handler(void *ctx, system_event_t *event)
{
	httpd_handle_t *hd = (httpd_handle_t *) ctx;

	switch (event->event_id) {
	case SYSTEM_EVENT_STA_START:
		ESP_LOGI(TAG, "SYSTEM_EVENT_STA_START");
		ESP_ERROR_CHECK(esp_wifi_connect());
		break;
	case SYSTEM_EVENT_STA_GOT_IP:
		ESP_LOGI(TAG, "SYSTEM_EVENT_STA_GOT_IP");
		ESP_LOGI(TAG,
			"Got IP: '%s'",
			ip4addr_ntoa(&event->event_info.got_ip.ip_info.ip));

		// Start webserver tests
		if(*hd == NULL) {
			*hd = start_tests();
		}

		break;
	case SYSTEM_EVENT_STA_DISCONNECTED:
		ESP_LOGI(TAG, "SYSTEM_EVENT_STA_DISCONNECTED");
		ESP_ERROR_CHECK(esp_wifi_connect());

		// Stop webserver tests
		if(*hd) {
			stop_tests(*hd);
			*hd = NULL;
		}

		break;
	default:
		break;
	}
	return ESP_OK;
}
extern "C" { static void initialise_wifi(void)
{
	
	
	ESP_ERROR_CHECK(nvs_flash_init());
	tcpip_adapter_init();
static httpd_handle_t hd = NULL;
	ESP_ERROR_CHECK(esp_event_loop_init(event_handler, &hd));

	wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	ESP_ERROR_CHECK(esp_wifi_init(&cfg));
	ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));

    wifi_config_t wifi_config;
	strcpy((char*)wifi_config.sta.ssid , "mysid" );
	
	strcpy((char*)wifi_config.sta.password ,  "xxxx" ) ;
    wifi_config.sta.bssid_set = false;

	ESP_LOGI(TAG, "Setting WiFi configuration SSID %s.%s..", wifi_config.sta.ssid,wifi_config.sta.password);
	ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
	ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config));
	ESP_ERROR_CHECK(esp_wifi_start());
}
}

extern "C" {
	
 void app_main(void)
{
	//ESP_ERROR_CHECK(nvs_flash_init());
	initialise_wifi();
	
	//ESP_ERROR_CHECK(init_spiffs());

	//ESP_ERROR_CHECK(start_file_server("/spiffs"));
}
}

on debug i can not see any error.

Alvin1Zhang, 2019-04-24:
@arnold-b Hi, thanks for reporting the issue. Would you please help provide more details as in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Thanks.

igrr, 2019-04-24:
@arnold-b I am not able to reproduce the issue with v3.2 release and the code attached below (this is wifi/getting_started/station example, with the code from your post added).
Please attach the source code of your initialize_wifi function.
I suspect that the issue is that some fields of Wi-Fi configuration structure are not zeroed out by your initialization code.
<details>
<summary>Example:</summary>

```c++
/* WiFi station Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "nvs_flash.h"

#include "lwip/err.h"
#include "lwip/sys.h"

/* The examples use WiFi configuration that you can set via 'make menuconfig'.

   If you'd rather not, just change the below entries to strings with
   the config you want - ie #define EXAMPLE_WIFI_SSID "mywifissid"
*/
#define EXAMPLE_ESP_WIFI_SSID      CONFIG_ESP_WIFI_SSID
#define EXAMPLE_ESP_WIFI_PASS      CONFIG_ESP_WIFI_PASSWORD
#define EXAMPLE_ESP_MAXIMUM_RETRY  CONFIG_ESP_MAXIMUM_RETRY

/* FreeRTOS event group to signal when we are connected*/
static EventGroupHandle_t s_wifi_event_group;

/* The event group allows multiple bits for each event, but we only care about one event 
 * - are we connected to the AP with an IP? */
const int WIFI_CONNECTED_BIT = BIT0;

static const char *TAG = "wifi station";

static int s_retry_num = 0;

static esp_err_t event_handler(void *ctx, system_event_t *event)
{
    switch(event->event_id) {
    case SYSTEM_EVENT_STA_START:
        esp_wifi_connect();
        break;
    case SYSTEM_EVENT_STA_GOT_IP:
        ESP_LOGI(TAG, "got ip:%s",
                 ip4addr_ntoa(&event->event_info.got_ip.ip_info.ip));
        s_retry_num = 0;
        xEventGroupSetBits(s_wifi_event_group, WIFI_CONNECTED_BIT);
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
        {
            if (s_retry_num < EXAMPLE_ESP_MAXIMUM_RETRY) {
                esp_wifi_connect();
                xEventGroupClearBits(s_wifi_event_group, WIFI_CONNECTED_BIT);
                s_retry_num++;
                ESP_LOGI(TAG,"retry to connect to the AP");
            }
            ESP_LOGI(TAG,"connect to the AP fail\n");
            break;
        }
    default:
        break;
    }
    return ESP_OK;
}

void wifi_init_sta()
{
    s_wifi_event_group = xEventGroupCreate();

    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL) );

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    wifi_config_t wifi_config = {
        .sta = {
            .ssid = EXAMPLE_ESP_WIFI_SSID,
            .password = EXAMPLE_ESP_WIFI_PASS
        },
    };

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA) );
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config) );
    ESP_ERROR_CHECK(esp_wifi_start() );

    ESP_LOGI(TAG, "wifi_init_sta finished.");
    ESP_LOGI(TAG, "connect to ap SSID:%s password:%s",
             EXAMPLE_ESP_WIFI_SSID, EXAMPLE_ESP_WIFI_PASS);
}

void app_main()
{
    //Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
      ESP_ERROR_CHECK(nvs_flash_erase());
      ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    setenv("TZ", "CET-1CEST", 0);
    tzset();
    time_t now;
    struct tm timeinfo;
    time(&now);

    ESP_LOGI(TAG, "ESP_WIFI_MODE_STA");
    wifi_init_sta();
}
```
</details>

arnold-b, 2019-04-24:
Hello, 
there is the code. if i comment out in app_main
               time_t now;
		struct tm timeinfo;
		time(&now);
then it works

#include "sdkconfig.h"

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include <string.h>
#include <sys/time.h>
#include <errno.h>
#include <sys/fcntl.h>
#include <unistd.h>
#include <ctype.h>

#include "esp_vfs.h"
#include "esp_log.h"
#include "esp_spiffs.h"

static const char tag[] = "[SPIFFS example]";
static const char *TAG_WIFI = "TEST_WIFI";
#define TAG TAG_WIFI

#ifdef CONFIG_EXAMPLE_USE_WIFI
#include "esp_wifi.h"
#include "esp_system.h"
#include "esp_event.h"
#include "esp_event_loop.h"
#include "freertos/event_groups.h"
#include "esp_attr.h"
#include "lwip/err.h"
#include "lwip/apps/sntp.h"

//#include "tests.h"
//#include "Webserver.h"

#include "file_server.cpp"
#include "nvs_flash.h"
#define EXAMPLE_WIFI_SSID  CONFIG_WIFI_SSID
#define EXAMPLE_WIFI_PASS  CONFIG_WIFI_PASSWORD
/* FreeRTOS event group to signal when we are connected & ready to make a request */
static EventGroupHandle_t wifi_event_group;

/* The event group allows multiple bits for each event,
   but we only care about one event - are we connected
   to the AP with an IP? */
const int CONNECTED_BIT = 0x00000001;

/*SPIFFS*/
esp_vfs_spiffs_conf_t conf = {
	.base_path = "/spiffs",
	.partition_label = NULL,
	.max_files = 5,
	.format_if_mount_failed = true
};

//-------------------------------
static void initialize_sntp(void)
{
	ESP_LOGI(tag, "Initializing SNTP");
	sntp_setoperatingmode(SNTP_OPMODE_POLL);
	sntp_setservername(0,(char *) "pool.ntp.org");
	sntp_init();
}

//---------------------------
extern "C" {
[... 688 lines cut ...]
}

//================
extern "C" {
	int app_main(void)
	{

		initialise_wifi();

//This is the Problem
		time_t now;
		struct tm timeinfo;
		time(&now);

//	

		printf("\r\n\n");
		ESP_LOGI(tag, "==== STARTING SPIFFS TEST ====\n");

		ESP_ERROR_CHECK(init_spiffs());

		return 0;
	}

}

igrr, 2019-04-24:
As I have (correctly) guessed above, you are not initializing all the fields of the configuration structure:

```c++
    wifi_config_t wifi_config;
	strcpy((char*)wifi_config.sta.ssid , "braun" );
	
	strcpy((char*)wifi_config.sta.password ,  "$0909115112257039$" ) ;
    wifi_config.sta.bssid_set = false;
```

Change
```c++
    wifi_config_t wifi_config;
```

to 
```c++
    wifi_config_t wifi_config = {};
```

to zero-initialize the rest of the fields.

arnold-b, 2019-04-24:
Hey, thanks for your help. 
sorry for this. 
after wifi_config_t wifi_config = {}; it works.

closed 2019-04-24 by arnold-b