espressif/esp-idf #3349: Undocumented wifi reasons?

sebirdman, 2019-04-23:
I'm seeing that my wifi disconnect is reason 34, which i wasn't expecting so my code didn't handle it properly. Are there reasons that are not currently in the wifi_err_reason_t enum and if so, what are they?

liuzfesp, 2019-04-23:
Hi @sebirdman , the reason code 34 is from the AP. If the AP disconnects the ESP32 STA, it notifies the reason code to ESP32 STA in deauth/disassociation frame.

According to WiFi spec <ieee80211-2016>, the 34 means: Disassociated because excessive number of frames need to be acknowledged, but are not acknowledged due to AP transmissions and/or poor channel conditions.

Other reason code defined in section 9.4.1.7:

Reason code | Name | Meaning
-- | -- | --
0 |   | Reserved
1 | UNSPECIFIED_REASON | Unspecified reason
2 | INVALID_AUTHENTICATION | Previous authentication no longer valid
3 | LEAVING_NETWORK_DEAUT H | Deauthenticated because sending STA is leaving (or has left) IBSS or ESS
4 | REASON_INACTIVITY | Disassociated due to inactivity
5 | NO_MORE_STAS | Disassociated because AP is unable to handle all currently associated STAs
6 | INVALID_CLASS2_FRAME | Class 2 frame received from nonauthenticated STA
7 | INVALID_CLASS3_FRAME | Class 3 frame received from nonassociated STA
8 | LEAVING_NETWORK_DISASS OC | Disassociated because sending STA is leaving (or has left) BSS
9 | NOT_AUTHENTICATED | STA requesting (re)association is not authenticated with responding STA
10 | UNACCEPTABLE_POWER_CA PABILITY | Disassociated because the information in the Power Capability element is unacceptable
11 | UNACCEPTABLE_SUPPORTED _CHANNELS | Disassociated because the information in the Supported Channels element is unacceptable
12 | BSS_TRANSITION_DISASSOC | Disassociated due to BSS transition management
13 | REASON_INVALID_ELEMENT | Invalid element, i.e., an element defined in this standard for which the content does not meet the specifications in Clause 9
14 | MIC_FAILURE | Message integrity code (MIC) failure
15 | 4WAY_HANDSHAKE_TIMEOU T | 4-way handshake timeout
16 | GK_HANDSHAKE_TIMEOUT | Group key handshake timeout
17 | HANDSHAKE_ELEMENT_MIS MATCH | Element in 4-way handshake different from (Re)Association Request/Probe Response/Beacon frame
18 | REASON_INVALID_GROUP_CI PHER | Invalid group cipher
19 | REASON_INVALID_PAIRWISE _CIPHER | Invalid pairwise cipher
20 | REASON_INVALID_AKMP | Invalid AKMP
21 | UNSUPPORTED_RSNE_VERSI ON | Unsupported RSNE version
22 | INVALID_RSNE_CAPABILITIE S | Invalid RSNE capabilities
23 | 802_1_X_AUTH_FAILED | IEEE 802.1X authentication failed
24 | REASON_CIPHER_OUT_OF_P OLICY | Cipher suite rejected because of the security policy
25 | TDLS_PEER_UNREACHABLE | TDLS direct-link teardown due to TDLS peer STA unreachable via the TDLS direct link
26 | TDLS_UNSPECIFIED_REASON | TDLS direct-link teardown for unspecified reason
27 | SSP_REQUESTED_DISASSOC | Disassociated because session terminated by SSP request
28 | NO_SSP_ROAMING_AGREEM ENT | Disassociated because of lack of SSP roaming agreement
29 | BAD_CIPHER_OR_AKM | Requested service rejected because of SSP cipher suite or AKM requirement
30 | NOT_AUTHORIZED_THIS_LO CATION | Requested service not authorized in this location
31 | SERVICE_CHANGE_ PRECLUDES_TS | TS deleted because QoS AP lacks sufficient bandwidth for this QoS STA due to a change in BSS service characteristics or operational mode (e.g., an HT BSS change from 40 MHz channel to 20 MHz channel)
32 | UNSPECIFIED_QOS_REASON | Disassociated for unspecified, QoS-related reason
33 | NOT_ENOUGH_BANDWIDTH | Disassociated because QoS AP lacks sufficient bandwidth for this QoS STA
34 | MISSING_ACKS | Disassociated because excessive number of frames need to be acknowledged, but are not acknowledged due to AP transmissions and/or poor channel conditions
35 | EXCEEDED_TXOP | Disassociated because STA is transmitting outside the limits of its TXOPs
36 | STA_LEAVING | Requesting STA is leaving the BSS (or resetting)
37 | END_TS END_BA END_DLS | Requesting STA is no longer using the stream or session
38 | UNKNOWN_TS UNKNOWN_BA | Requesting STA received frames using a mechanism for which a setup has not been completed
39 | TIMEOUT | Requested from peer STA due to timeout
45 | PEERKEY_MISMATCH | Peer STA does not support the requested cipher suite
46 | PEER_INITIATED | In a DLS Teardown frame: The teardown was initiated by the DLS peer  								In a Disassociation frame: Disassociated because authorized access limit reached
47 | AP_INITIATED | In a DLS Teardown frame: The teardown was initiated by the AP  								In a Disassociation frame: Disassociated due to external service requirements
48 | REASON_INVALID_FT_ACTIO N_FRAME_COUNT | Invalid FT Action frame count
49 | REASON_INVALID_PMKID | Invalid pairwise master key identifier (PMKID)
50 | REASON_INVALID_MDE | Invalid MDE
51 | REASON_INVALID_FTE | Invalid FTE
52 | MESH-PEERING-CANCELED | Mesh peering canceled for unknown reasons
53 | MESH-MAX-PEERS | The mesh STA has reached the supported maximum number of peer mesh STAs
54 | MESH-CONFIGURATION- POLICY-VIOLATION | The received information violates the Mesh Configuration policy configured in the mesh STA profile
55 | MESH-CLOSE-RCVD | The mesh STA has received a Mesh Peering Close frame requesting to close the mesh peering.
56 | MESH-MAX-RETRIES | The mesh STA has resent dot11MeshMaxRetries Mesh Peering Open frames, without receiving a Mesh Peering Confirm frame.
57 | MESH-CONFIRM-TIMEOUT | The confirmTimer for the mesh peering instance times out.
58 | MESH-INVALID-GTK | The mesh STA fails to unwrap the GTK or the values in the wrapped contents do not match
[... cut ...]
63 | MESH-PATH-ERROR- DESTINATION- UNREACHABLE | The mesh STA determines that the link to the next hop of an active path in its forwarding information is no longer usable.
64 | MAC-ADDRESS-ALREADY- EXISTS-IN-MBSS | The Deauthentication frame was sent because the MAC address of the STA already exists in the mesh BSS. See 11.3.6.
65 | MESH-CHANNEL-SWITCH- REGULATORY- REQUIREMENTS | The mesh STA performs channel switch to meet regulatory requirements.
66 | MESH-CHANNEL-SWITCH- UNSPECIFIED | The mesh STA performs channel switching with unspecified reason.
67–65 535 |   | Reserved

sebirdman, 2019-04-23:
awsome! thank you @liuzfesp

closed 2019-04-23 by sebirdman