espressif/esp-idf #3350: spi_master example stops updating display after a few seconds

wulph111, 2019-04-23:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v3]
- Module or chip used:  [ESP32-WROVER]
- IDF version (run ``git describe --tags`` to find it): 
     v4.0-dev-402-ga20d02b7f (also was occurring in v3.3-beta1-4-g23b6d40c5)
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description

SPI_MASTER example display freezes after a few seconds. I added a printf("Frame %d\n",frame);
 to display_pretty_colors which shows the frame counter continuing to increase although the display is frozen. The display appears to fleeze somewhere around frame 45.

### Expected Behavior

Sinusoidal functions in pretty_effect_calc_lines() should update the display indefinitely.

### Actual Behavior

For some reason the updating stops, although the loop continues to execute.

### Steps to repropduce

Copy  $IDF_PATH/examples/peripherals/spi_master to working folder
make flash && make monitor

### Code to reproduce this issue

```cpp
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/spi_master
```

## Debug Logs

```
$ make monitor
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Python requirements from C:/esp32/esp-idf/requirements.txt are satisfied.
MONITOR
--- idf_monitor on com14 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3a (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6296
load:0x40078000,len:11308
load:0x40080400,len:6700
entry 0x40080764
I (28) boot: ESP-IDF v4.0-dev-402-ga20d02b7f 2nd stage bootloader
I (28) boot: compile time 22:41:42
I (43) boot: Enabling RNG early entropy source...
I (43) boot: SPI Speed      : 40MHz
I (43) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (50) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot: End of partition table
I (87) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1ca08 (1172
56) map
I (137) esp_image: segment 1: paddr=0x0002ca30 vaddr=0x3ffb0000 size=0x02190 (  8
592) load
I (141) esp_image: segment 2: paddr=0x0002ebc8 vaddr=0x40080000 size=0x00400 (  1
024) load
0x40080000: _WindowOverflow4 at C:/esp32/esp-idf/components/freertos/xtensa_vecto
rs.S:1779

I (144) esp_image: segment 3: paddr=0x0002efd0 vaddr=0x40080400 size=0x01040 (  4
160) load
I (154) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x15bb4 ( 89
012) map
0x400d0018: _flash_cache_start at ??:?

I (193) esp_image: segment 5: paddr=0x00045bd4 vaddr=0x40081440 size=0x07f48 ( 32
584) load
0x40081440: esp_dport_access_stall_other_cpu_start at C:/esp32/esp-idf/components
/esp32/dport_access.c:154

I (212) boot: Loaded app from partition at offset 0x10000
I (212) boot: Disabling RNG early entropy source...
I (213) cpu_start: Pro cpu up.
I (216) cpu_start: Application information:
I (221) cpu_start: Project name:     spi_master
I (226) cpu_start: App version:      1
I (231) cpu_start: Compile time:     Apr 22 2019 22:42:57
I (237) cpu_start: ELF file SHA256:  3844daf3cda66e3e...
I (243) cpu_start: ESP-IDF:          v4.0-dev-402-ga20d02b7f
I (249) cpu_start: Starting app cpu, entry point is 0x40080f38
0x40080f38: call_start_cpu1 at C:/esp32/esp-idf/components/esp32/cpu_start.c:267

I (0) cpu_start: App cpu up.
I (260) heap_init: Initializing. RAM available for dynamic allocation:
I (267) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (273) heap_init: At 3FFB3730 len 0002C8D0 (178 KiB): DRAM
I (279) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (285) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (292) heap_init: At 40089388 len 00016C78 (91 KiB): IRAM
I (298) cpu_start: Pro cpu start user code
I (316) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
LCD ID: 00FFFFFF
ST7789V detected.
LCD ST7789V initialization.
Frame 1
Frame 2
Frame 3
...
```

## Other items if possible

[spi_master_files.zip](https://github.com/espressif/esp-idf/files/3105696/spi_master_files.zip)

wulph111, 2019-04-30:
How would I go about troubleshooting this? The main loop is still running. Do I have defective hardware, or has anyone else seen this issue?

wulph111, 2019-05-10:
No views?

learnerzcz, 2024-12-15:
hi ,solve it? i meet also meet it. i want to keep dispay ,but no idea to do