espressif/esp-idf #3353: iBeacon demo fails to detect beacons other than Kontakt.io's Beacon Pro

marjon-halter, 2019-04-23:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4.1]
- Module or chip used:  [ESP32-WROVER]
- IDF version: v4.0-dev-402-ga20d02b7f
- Build System:[Make]
- Compiler version: crosstool-NG crosstool-ng-1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description
iBeacon Demo project only detects Kontakt.io's Beacon Pro (https://store.kontakt.io/next-generation/32-beacon-pro-bp16-3.html), but no other beacon transmitting devices within it's proximity.

//Detailed problem description:
Downloaded the Beacon Simulator app (https://play.google.com/store/apps/details?id=net.alea.beaconsimulator&hl=en) on my phone to see other beacon transmitting devices within proximity, and they show up. But the ESP32 WROVER dev kit could not see them, except for the Kontakt.io's Beacon Pro. I tried setting up the Beacon Simulator app to be a transmitter, and detect it's beacon using another mobile phone with the same app, and it seems to be transmitting correctly, but the ESP32 WROVER dev kit still could not see it's beacon. I tried to move the Kontakt.io's Beacon Pro outside of the proximity to see if the ESP32 stops detecting it, and yes, it stopped detecting it, but could not see any other beacon transmitter (even though the Beacon Simulator app can see a lot).

### Expected Behavior
ESP32-WROVER dev kit detects iBeacons within its proximity.

### Actual Behavior
ESP32-WROVER dev kit can only detect Kontakt.io's Beacon Pro transmitter.

### Steps to repropduce
1. Compile the ble_ibeacon demo project and program to the ESP32-WROVER dev kit.
2. Do a "make flash monitor" and check the results of the BLE scan to see if it detects any iBeacon devices.

### Code to reproduce this issue
```cpp
static esp_ble_scan_params_t ble_scan_params = {
    .scan_type              = BLE_SCAN_TYPE_ACTIVE,
    .own_addr_type          = BLE_ADDR_TYPE_PUBLIC,
    .scan_filter_policy     = BLE_SCAN_FILTER_ALLOW_ALL,
    .scan_interval          = 0x50,
    .scan_window            = 0x30,
    .scan_duplicate         = BLE_SCAN_DUPLICATE_DISABLE
};
```
```
static void esp_gap_cb(esp_gap_ble_cb_event_t event, esp_ble_gap_cb_param_t *param)
{
    esp_err_t err;

    switch (event) {
    case ESP_GAP_BLE_ADV_DATA_RAW_SET_COMPLETE_EVT:{
#if (IBEACON_MODE == IBEACON_SENDER)
        esp_ble_gap_start_advertising(&ble_adv_params);
#endif
        break;
    }
    case ESP_GAP_BLE_SCAN_PARAM_SET_COMPLETE_EVT: {
#if (IBEACON_MODE == IBEACON_RECEIVER)
        //the unit of the duration is second, 0 means scan permanently
        uint32_t duration = 0;
        esp_ble_gap_start_scanning(duration);
        ESP_LOGI(DEMO_TAG, "Starting BLE scan...");
#endif
        break;
    }
    case ESP_GAP_BLE_SCAN_START_COMPLETE_EVT:
        //scan start complete event to indicate scan start successfully or failed
        if ((err = param->scan_start_cmpl.status) != ESP_BT_STATUS_SUCCESS) {
            ESP_LOGE(DEMO_TAG, "Scan start failed: %s", esp_err_to_name(err));
        } else {
            ESP_LOGI(DEMO_TAG, "Scan start successful.");
        }
        break;
    case ESP_GAP_BLE_ADV_START_COMPLETE_EVT:
        //adv start complete event to indicate adv start successfully or failed
        if ((err = param->adv_start_cmpl.status) != ESP_BT_STATUS_SUCCESS) {
            ESP_LOGE(DEMO_TAG, "Adv start failed: %s", esp_err_to_name(err));
        } else {
            ESP_LOGI(DEMO_TAG, "Adv scan start successful.");
        }
        break;
    case ESP_GAP_BLE_SCAN_RESULT_EVT: {
        esp_ble_gap_cb_param_t *scan_result = (esp_ble_gap_cb_param_t *)param;
        switch (scan_result->scan_rst.search_evt) {
        case ESP_GAP_SEARCH_INQ_RES_EVT:
            /* Search for BLE iBeacon Packet */
            if (esp_ble_is_ibeacon_packet(scan_result->scan_rst.ble_adv, scan_result->scan_rst.adv_data_len)){
                esp_ble_ibeacon_t *ibeacon_data = (esp_ble_ibeacon_t*)(scan_result->scan_rst.ble_adv);
                ESP_LOGI(DEMO_TAG, "----------iBeacon Found----------");
                esp_log_buffer_hex("IBEACON_DEMO: Device address:", scan_result->scan_rst.bda, ESP_BD_ADDR_LEN );
                esp_log_buffer_hex("IBEACON_DEMO: Proximity UUID:", ibeacon_data->ibeacon_vendor.proximity_uuid, ESP_UUID_LEN_128);

                uint16_t major = ENDIAN_CHANGE_U16(ibeacon_data->ibeacon_vendor.major);
                uint16_t minor = ENDIAN_CHANGE_U16(ibeacon_data->ibeacon_vendor.minor);
                ESP_LOGI(DEMO_TAG, "Major: 0x%04x (%d)", major, major);
                ESP_LOGI(DEMO_TAG, "Minor: 0x%04x (%d)", minor, minor);
                ESP_LOGI(DEMO_TAG, "Measured power (RSSI at a 1m distance):%d dbm", ibeacon_data->ibeacon_vendor.measured_power);
                ESP_LOGI(DEMO_TAG, "RSSI of packet:%d dbm", scan_result->scan_rst.rssi);
            }
            break;
        default:
            break;
        }
        break;
    }

    case ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT:
        if ((err = param->scan_stop_cmpl.status) != ESP_BT_STATUS_SUCCESS){
            ESP_LOGE(DEMO_TAG, "Scan stop failed: %s", esp_err_to_name(err));
        }
        else {
            ESP_LOGI(DEMO_TAG, "Stop scan successfully");
        }
        break;

    case ESP_GAP_BLE_ADV_STOP_COMPLETE_EVT:
        if ((err = param->adv_stop_cmpl.status) != ESP_BT_STATUS_SUCCESS){
            ESP_LOGE(DEMO_TAG, "Adv stop failed: %s", esp_err_to_name(err));
        }
        else {
            ESP_LOGI(DEMO_TAG, "Stop adv successfully");
        }
        break;

    default:
        break;
    }
}
```
```
I (0) cpu_start: App cpu up.
I (865) heap_init: Initializing. RAM available for dynamic allocation:
I (872) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (878) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (884) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (890) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (896) heap_init: At 3FFD4D80 len 0000B280 (44 KiB): DRAM
I (903) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (909) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (915) heap_init: At 4009188C len 0000E774 (57 KiB): IRAM
I (922) cpu_start: Pro cpu start user code
I (270) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (299) BTDM_INIT: BT controller compile version [484c057]
I (309) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (419) phy: phy_version: 4100, 2a5dd04, Jan 23 2019, 21:00:07, 0, 0
I (819) IBEACON_DEMO: register callback
I (829) IBEACON_DEMO: Starting BLE demo.

I (839) IBEACON_DEMO: Starting BLE scan...
I (839) IBEACON_DEMO: Scan start successful.
```

## Other items if possible

- [ ] sdkconfig file attached.
- [ ] elf file in the ``build`` folder attached

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3109641/sdkconfig.txt)

[ble_ibeacon_demo.elf.zip](https://github.com/espressif/esp-idf/files/3109644/ble_ibeacon_demo.elf.zip)

Weijian-Espressif, 2019-05-20:
ibeacon example only detects apple beacon,  please refer to esp_ibeacon_api.c :

/* For iBeacon packet format, please refer to Apple "Proximity Beacon Specification" doc */
/* Constant part of iBeacon data */
esp_ble_ibeacon_head_t ibeacon_common_head = {
    .flags = {0x02, 0x01, 0x06},
    .length = 0x1A,
    .type = 0xFF,
    .company_id = 0x004C,
    .beacon_type = 0x1502
};

bool esp_ble_is_ibeacon_packet (uint8_t *adv_data, uint8_t adv_data_len){
    bool result = false;

    if ((adv_data != NULL) && (adv_data_len == 0x1E)){
        if (!memcmp(adv_data, (uint8_t*)&ibeacon_common_head, sizeof(ibeacon_common_head))){
            result = true;
        }
    }

    return result;
}

Alvin1Zhang, 2019-07-05:
@marjon-halter Thanks for reporting the issue, feel free to reopen the issue if it still exists. Thanks.

closed 2019-07-05 by Alvin1Zhang

krone-trailer-franzhoepfinger, 2021-05-17:
Hello, 

i can CONFIRM. a real iBeacon works. 
ALL Apps in the Google Playstore i tested send out this Signature for a iBeacon (simulated)
02 01 06 1a 16 ee fd 03 05
while the REAL iBeacon according to Apples Spec would be: 
02 01 06 1a ff 4c 00 02  15