espressif/esp-idf #3354: NVS non deterministic behavior depending on delay

Mair, 2019-04-24:
## Environment

- Development Kit:     Devkit1 and Lolin32

- Module or chip used:  [ESP32-WROOM-32]
- IDF version v4.0-dev-402-ga20d02b7f
- Build System:         [Make]
- Compiler version: (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description

I'm getting some strange behavior with NVS.
Reading from NVS  and then immediately writing to it has a different outcome if reading then delaying then writing.
It seems like its almost running asynchronously without blocking. 
Note the commented line of code where I have a delay 
`// vTaskDelay(1000 / portTICK_PERIOD_MS);` If that line is commented the output is 
**_NVS: Value not set_**
However, un-commented the output is 
**_NVS: Value is 1_**

To flash the chip I am using `make -j8 erase_flash flash monitor` Which should erase the flash each time.

Can anyone explain to me why the code behaves differently with a delay. It seems like its almost running asynchronously without blocking. Is this perhaps a bug?

### Expected Behavior
erasing the chip consistency produces **_NVS: Value not set_** regardless of delay  
### Actual Behavior
inconsistency depending on the delay 
### Steps to reproduce

1. setup a simple project and copy in the below code
2. use the following make command `make -j8 erase_flash flash monitor`
3.  note the output
4. uncomment the delay
5.  use the following make command `make -j8 erase_flash flash monitor`
6. note the output

### Code to reproduce this issue

```cpp
#include <stdio.h>
#include <string.h>
#include "esp_log.h"
#include "nvs.h"
#include "nvs_flash.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"

#define TAG "NVS"

void app_main()
{
  ESP_LOGW(TAG, "Started!!!");
  esp_err_t err = nvs_flash_init();
  ESP_ERROR_CHECK(err);
  
  nvs_handle handle;
  err = nvs_open("store", NVS_READWRITE, &handle);
  ESP_ERROR_CHECK(err);

  int32_t val = 0;
  err = nvs_get_i32(handle,"val1", &val);
  printf("err is %s\n",esp_err_to_name(err));
  // depending if the below line is commented the output is different 
  // vTaskDelay(800 / portTICK_PERIOD_MS);

  switch (err)
  {
    case ESP_ERR_NVS_NOT_FOUND:
      ESP_LOGE(TAG, "Value not set yet");
      break;
    case ESP_OK:
      ESP_LOGI(TAG,"Value is %d", val);
      break;
    default:
      ESP_LOGE(TAG, "Error (%s) opening NVS handle!\n", esp_err_to_name(err));
      break;
  }
  printf("completed reading\n");
  val++; 

  nvs_set_i32(handle,"val1",val);
  nvs_commit(handle);
  nvs_close(handle);
  ESP_LOGI(TAG,"Value is now set to %d", val);
}
```

## Debug Logs

> With Delay

```
W (289) NVS: Started!!!
err is ESP_ERR_NVS_NOT_FOUND
E (1109) NVS: Value not set yet
completed reading
I (1119) NVS: Value is now set to 1
```

> Without Delay

```
W (289) NVS: Started!!!
err is ESP_OK
I (309) NVS: Value is 1
completed reading
I (319) NVS: Value is now set to 2

```

negativekelvin, 2019-04-24:
Are you sure the flash is being erased?  If you run make without the parallel jobs does it change anything?

KonstantinKondrashov, 2019-04-24:
Hi @Mair. I think this is not an issue linked with NVS component. Here we see that the log was lost during the first boot.  Can you put a delay in the start of the code to make sure in it?
```
void app_main()
{
   vTaskDelay(10000 / portTICK_PERIOD_MS);
  ESP_LOGW(TAG, "Started!!!");
....
```

KonstantinKondrashov, 2019-04-24:
> Are you sure the flash is being erased? If you run make without the parallel jobs does it change anything?

I have reproduced this behavior like ` make erase_flash` then `make flash monitor`. 

I think the monitor misses some first logs and then reset happened `After flashing (Reset after flashing)`.

jsobell, 2019-04-24:
It appears that the code is being executed twice. It looks like when the initial download is completed the device boots, runs your code (initialising `val1`), then the device is reset for the monitoring process and during the monitor execution you see that `val1` has already been written to.
When you add the delay the reset occurs before the write of 'val1' to NVS, and the system starts again with the data still blank.
If you move the `vTaskDelay(800 / portTICK_PERIOD_MS);` to be the first instruction in `main()` it will also work as expected, so this isn't a system or library issue but an initialisation 'artifact' of the way the device is programmed then reset.
I'd suggest adding an NVS flag to indicate whether initialisation has already been performed, and if not you delay for 1 second in case it's the first ever run and the IF tools are likely to trigger an additional reset. You really need something like this anyway to indicate that initialisation completed (and it didn't get reset half way through) and you should probably reset the NVS variables in code or even look at `esp_partition_erase` during that initialisation instead of depending on the `erase_flash` command.
Having a `initialised` flag in NVS is also a nice way of allowing factory resets, as you can just wipe that value and it will auto-initialise next time you boot the device :)

negativekelvin, 2019-04-25:
It might be worthwhile to add intelligence to the makefile that only resets the chip once after flashing.

closed 2019-12-20 by Alvin1Zhang