espressif/esp-idf #3355: ESP32- SPIFFS issue

SampathReddyP, 2019-04-24:
Hi

I have developed a application in which I have store data in file for that I used Spiffs file system.
**Development Environment:**
Development Kit: ESP32-DEVKITV1
Module: ESP32-WROOM-32D
IDF version:  v3.3-beta1-432-g5ec58c316
Operation System: Windows
Power Supply: USB

I have worked on basic spiffs example for testing, In that i observed one case i.e, while i try to write 30kb data continuously to a file (ex:"/spiffs/hello.txt") for 30-40 times i getting fwrite error or fwrite() function return 0.
Here every-time i am overwriting the file /spiffs/hello.txt. I also tried with the different mode option like "w","wb" and "wb+". But in all cases i am getting same issue.
In which scenario the fwrite() function will return 0 please explain?

**Here is the code:**
               FILE* f = fopen("/spiffs/hello.txt", "wb");
               if (f == NULL) {
	           ESP_LOGE(TAG, "Failed to open file for writing");
	           return;
               }
               char *buf = "0123-----abc";  **### ->30kb in size**
               unsigned int wc = fwrite((char *)buf,1,strlen(buf),f);
               fclose(f);
               ESP_LOGI(TAG, "File written-%d",wc);   **### ->Here wc is 0 in some cases**

The major problem here is if i write data in chucks like I have 30kb file but each time i will write 3kb data to it. If in middle i get fwrite() return 0 means my entire file is getting corrupted. So please suggest some solution related to this issue.

And also i have to store 60kb data to a file in my application so please suggest whether spiffs file system is better option or not if not suggest any other options?

Regards,
Sampath

MatthiasJentsch, 2019-10-06:
I've the same problem with this environment:

Development Kit: none
Kit version: none
Module or chip used: ESP32-WROVER-B
IDF version: v4.1-dev-369-g4dac7c7df
Build System: CMake
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
Operating System: Windows
Power Supply: external 3.3V

Has somebody an explanation for this behavior?

Jorge-Crespo-Celdran, 2021-12-13:
@SampathReddyP and @MatthiasJentsch 

I'm having the same issue. Did you manage to solve it?