espressif/esp-idf #3356: ESP32 BLE connection - Dunno how to make loops

happynet95, 2019-04-24:
Hello guys. I'm trying to make a device that is connected by other ESP32 chip and sends (and also recieves) data ceaselessly. However, I can't even make a loop : (
I was making the code w/th Arduino IDE(and it was almost completed) but I figured out that it was unable a BLE client to connect 3 or more BLE servers simiultaneously. So I moved to esp-idf and faced those problems. There was BLE_GATT_SERVER example(which can make multiconnections, so I used it.)

Of course I know that I should run void loop(){} function using while(1){}, so I tried several times. At first, I put the whole code in void main_app() in a while(1) loop, . the device didnt work as I expected. (So the problem is also that I dont know which function should be put in Setup and which should be put in loop.) Can you guys help?

My codes are based in gatt_client and gatt_server.

https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_server/main/gatts_demo.c

https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_client/main/gattc_demo.c

AmrutaCh, 2019-04-24:
You just need to create a task and perform your actions in while(1) loop there. Its basic operation in RTOS

happynet95, 2019-05-14:
@AmrutaCh  Thank You I got it!

Alvin1Zhang, 2019-05-16:
@happynet95 Thanks for reporting the issue. Has your issue been fixed? Thanks.

happynet95, 2019-05-17:
@Alvin1Zhang Yeah. Although struggling with other issues.

```

void app_main()
{
    esp_err_t ret;

    // Initialize NVS.
    ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK( ret );

    ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT));

    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    ret = esp_bt_controller_init(&bt_cfg);
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s initialize controller failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }

    ret = esp_bt_controller_enable(ESP_BT_MODE_BLE);
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s enable controller failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }
    ret = esp_bluedroid_init();
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s init bluetooth failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }
    ret = esp_bluedroid_enable();
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s enable bluetooth failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }

    ret = esp_ble_gatts_register_callback(gatts_event_handler);
    if (ret){
        ESP_LOGE(GATTS_TAG, "gatts register error, error code = %x", ret);
        return;
    }
    ret = esp_ble_gap_register_callback(gap_event_handler);
    if (ret){
        ESP_LOGE(GATTS_TAG, "gap register error, error code = %x", ret);
        return;
    }
    ret = esp_ble_gatts_app_register(PROFILE_A_APP_ID);
    if (ret){
        ESP_LOGE(GATTS_TAG, "gatts app register error, error code = %x", ret);
        return;
    }
    ret = esp_ble_gatts_app_register(PROFILE_B_APP_ID);
    if (ret){
        ESP_LOGE(GATTS_TAG, "gatts app register error, error code = %x", ret);
        return;
    }
    esp_err_t local_mtu_ret = esp_ble_gatt_set_local_mtu(500);
    if (local_mtu_ret){
        ESP_LOGE(GATTS_TAG, "set local  MTU failed, error code = %x", local_mtu_ret);
    }

    return;
}
```

This is the code of GATT_SERVER. and I thought that I had to control this with a loop. However, this is just a setup. think about Arduino IDE. There are void setup() and void loop. If you want to send some signals, you can just use 

```
while(1){
esp_ble_gatts_send_indicate(GATTS_IF, Id, gl_profile_tab_server[PROFILE_A_APP_ID].char_handle,4, value, false);
}
```

GATTS_IF and Id are the variables you have to change. You can easily find them from the codes.

Weijian-Espressif, 2019-05-17:
@happynet95 , please not use `while(1)`, you can create a task to send data periodically. you can refer to our ble spp server demo and ble spp client demo.

Alvin1Zhang, 2019-06-11:
@happynet95 Thanks for reporting the issue. Feel free to reopen if the issue still exists, and help open another issue for new issues. Thanks.

closed 2019-06-11 by Alvin1Zhang