espressif/esp-idf #3357: Feature request: compile-time way to get IDF version components

martijnthe, 2019-04-24:
I know there's the `IDF_VER` preprocessor define. But it's a string, so cannot be used in preprocessor conditionals. I would like to have `IDF_VER_MAJOR`, `IDF_VER_MINOR`, `IDF_VER_PATCH`, so that something like this would be possible:

```
#if IDF_VER_MAJOR == 3
#  if IDF_VER_MINOR >= 3
// Code for IDF v3.2.x and below
# else
// Code for IDF v3.3.x and above
#  endif
#else
#  error "Unsupported IDF version"
#endif
```

I know you can parse the `IDF_VER` string at runtime, but that would drag in all the code paths and bloat the binary size.

igrr, 2019-04-24:
This is tracked in https://github.com/espressif/esp-idf/issues/2482, please follow the link and click "Subscribe" to receive updates. Closing as duplicate.

closed 2019-04-24 by igrr

martijnthe, 2019-04-24:
@igrr thanks for pointing out the existing ticket. I’d searched for it but didn’t find it.