espressif/esp-idf #3358: Esp32 wont update time through sntp with gsm module

Arthedian, 2019-04-24:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):  v4.0-dev-402-ga20d02b7f
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     Debian
- Power Supply: external 5V

## Problem Description

I am trying to get time through sntp with gsm module sim800l. The module is succesfully connected to the internet, but the time will not be updated (through WIFI its ok). Even I cannot connect to the HTTP protocol.

### Expected Behavior

update time through sntp on esp32 with sim800L

### Actual Behavior

nothing happend

### Steps to repropduce

1. clone https://github.com/loboris/ESP32-PPPOS-EXAMPLE.git
2. set tx/rx/internet settings

### Code to reproduce this issue
code [here](https://github.com/loboris/ESP32-PPPOS-EXAMPLE.git)

## Other items if possible

> I (15211) [SNTP]: Waiting for system time to be set... (1/10)
I (17218) [SNTP]: Waiting for system time to be set... (2/10)
I (19218) [SNTP]: Waiting for system time to be set... (3/10)
I (21218) [SNTP]: Waiting for system time to be set... (4/10)
I (23218) [SNTP]: Waiting for system time to be set... (5/10)
I (25218) [SNTP]: Waiting for system time to be set... (6/10)
I (27218) [SNTP]: Waiting for system time to be set... (7/10)
I (29218) [SNTP]: Waiting for system time to be set... (8/10)
I (31218) [SNTP]: Waiting for system time to be set... (9/10)
I (33218) [SNTP]: ERROR OBTAINING TIME
 (34240) [HTTP]: ===== HTTP GET REQUEST =========================================
E (34240) [HTTP]: DNS lookup failed err=202 res=0x0
I (35252) [HTTP]: ===== HTTP GET REQUEST =========================================
E (35252) [HTTP]: DNS lookup failed err=202 res=0x0

[this](https://github.com/espressif/esp-idf/issues/2648) looks the same, but it was closed without answering it. 

I have my custom FTP code and it is working, so I have internet on my sim card.

Arthedian, 2019-04-24:
I was further testing, and If I choose branch 3.0, than the code is working, but If I choose branch 3.3 or 4.0, than it is not.

suda-morris, 2019-04-25:
Hi @Arthedian It seems like you didn't setup the DNS server correctly. BTW, we have a PPPoS client example in IDF example [here](https://github.com/espressif/esp-idf/tree/master/examples/protocols/pppos_client) which also support SIM800L. Feel free to try it and check if you still have similiar issue.

Arthedian, 2019-04-25:
I am trying this example code from version 3.3. I changed the pins in the config and added this code to the begging of the main, so I enable power to the gsm module on my board:
```
    gpio_pad_select_gpio(19);
    gpio_set_direction(19, GPIO_MODE_OUTPUT);
    gpio_set_level(19, 0);
    gpio_set_level(19, 1);
```
Why the app is falling?
> I (63) cpu_start: Starting scheduler on PRO CPU.
> D (0) intr_alloc: Connected src 25 to int 2 (cpu 1)
> I (0) cpu_start: Starting scheduler on APP CPU.
> D (78) heap_init: New heap initialised at 0x3ffe0440
> D (88) heap_init: New heap initialised at 0x3ffe4350
> D (88) intr_alloc: Connected src 16 to int 12 (cpu 0)
> I (98) uart: queue free spaces: 30
> D (98) intr_alloc: Connected src 35 to int 13 (cpu 0)
> D (108) event: created event loop 0x3ffb98c4
> E (118) esp-modem: esp_dte_handle_line(84): handle line failed
> D (118) event: created post for event ESP_MODEM_EVENT:4
> D (128) event: posted ESP_MODEM_EVENT:4 to loop 0x3ffb98c4
> D (128) event: running post ESP_MODEM_EVENT:4 with handler 0x400d4034 on loop 0x3ffb98c4
> W (138) pppos_example: Unknow line received: AT
> 
> D (188) dce_service: sync ok
> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
> Core 0 register dump:
> PC      : 0x40104591  PS      : 0x00060d30  A0      : 0x800d44f9  A1      : 0x3ffbabf0  
> A2      : 0x00000000  A3      : 0x3ffb9418  A4      : 0x0000000b  A5      : 0x3ffbaf2c  
> A6      : 0x3ffb961c  A7      : 0x0000000b  A8      : 0x8010458e  A9      : 0x3ffbabd0  
> A10     : 0x00000000  A11     : 0x00000001  A12     : 0x401047a8  A13     : 0x00000000  
> A14     : 0x00000000  A15     : 0x0000000b  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
> EXCVADDR: 0x0000002c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
> 
> ELF file SHA256: cebbf38ef94db7e9dc02c65db5d3a7fba91d6c0b71137bb45a396c3513db3901
> 
> Backtrace: 0x40104591:0x3ffbabf0 0x400d44f6:0x3ffbac10 0x400d464e:0x3ffbac40 0x40086911:0x3ffbac70

Arthedian, 2019-04-25:
so after further testing it is sometimes working with exception throwing, but still it wont connect to the mqtt server which is in the example:

> D (347256) TRANS_TCP: [sock=55],connecting to server IP:198.41.30.241,Port:1883...
> E (347896) MQTT_CLIENT: Error transport connect
> D (347896) MQTT_CLIENT: Reconnect after 10000 ms
> I (347896) pppos_example: MQTT_EVENT_DISCONNECTED

suda-morris, 2019-04-25:
@Arthedian Powering on GSM module will cost a long time before it's ready to process AT commands. So in your application you may need to add this delay before you send the first AT command.

And please make sure 2G network is still working in your country.

Arthedian, 2019-04-25:
@suda-morris Even If I add 10 second wait, it is still falling. And If it is not responding to the AT command, than it should write it to the console, not to write an exception right?

2G network is working in my country because my FTP client code is working with gsm.

suda-morris, 2019-05-06:
I tested the example just in my local (Shanghai, China) again, it still works fine here.
I wonder if your 2G module print out the IP address it obtained from *base station* like following? Just want to ensure that your sim card is not out of date.
```bash
I (2656) pppos_example: ~~~~~~~~~~~~~~
I (2656) pppos_example: IP          : 10.65.71.127
I (2656) pppos_example: Netmask     : 255.255.255.255
I (2666) pppos_example: Gateway     : 10.64.64.64
I (2666) pppos_example: Name Server1: 211.136.112.50
I (2676) pppos_example: Name Server2: 211.136.150.66
I (2676) pppos_example: ~~~~~~~~~~~~~~
```

Arthedian, 2019-05-22:
@suda-morris 

so currently idf example is succesfully connecting to the mqtt server, but when I added obtain_time, before powering down the module, I still cannot get the time.

> I (30076) uart: queue free spaces: 30
> I (31226) pppos_example: Module: SIMCOM_SIM800L
> I (31226) pppos_example: Operator: "OSKAR"
> I (31226) pppos_example: IMEI: 867856031777128
> I (31226) pppos_example: IMSI: 230030156526940
> I (31426) pppos_example: rssi: 30, ber: 0
> I (31626) pppos_example: Battery voltage: 4153 mV
> I (31826) pppos_example: Modem PPP Started
> E (32076) esp-modem: Frame Error
> E (32126) esp-modem: Frame Error
> E (32176) esp-modem: Frame Error
> E (32226) esp-modem: Frame Error
> I (32596) pppos_example: Modem Connect to PPP Server
> I (32596) pppos_example: ~~~~~~~~~~~~~~
> I (32596) pppos_example: IP          : 10.169.206.145
> I (32596) pppos_example: Netmask     : 255.255.255.255
> I (32606) pppos_example: Gateway     : 192.168.254.254
> I (32606) pppos_example: Name Server1: 217.77.165.211
> I (32616) pppos_example: Name Server2: 217.77.165.81
> I (32616) pppos_example: ~~~~~~~~~~~~~~
> I (32626) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
> I (32636) pppos_example: MQTT other event id: 7
> I (33146) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
> I (34476) pppos_example: MQTT_EVENT_CONNECTED
> I (34476) pppos_example: sent subscribe successful, msg_id=29270
> I (35386) pppos_example: MQTT_EVENT_SUBSCRIBED, msg_id=29270
> I (35386) pppos_example: sent publish successful, msg_id=0
> I (36386) pppos_example: MQTT_EVENT_DATA
> TOPIC=/topic/esp-pppos
> DATA=esp32-pppos
> I (37406) pppos_example: Modem PPP Stopped
> I (37836) pppos_example: Initializing SNTP
> I (37836) pppos_example: Waiting for system time to be set... (1/10)
> I (39836) pppos_example: Waiting for system time to be set... (2/10)
> I (41836) pppos_example: Waiting for system time to be set... (3/10)
> I (43836) pppos_example: Waiting for system time to be set... (4/10)
> I (45836) pppos_example: Waiting for system time to be set... (5/10)
> I (47836) pppos_example: Waiting for system time to be set... (6/10)
> I (49836) pppos_example: Waiting for system time to be set... (7/10)
> I (51836) pppos_example: Waiting for system time to be set... (8/10)
> I (53836) pppos_example: Waiting for system time to be set... (9/10)
> E (55866) esp-modem: Frame Error
> E (55916) esp-modem: Frame Error
> I (56366) pppos_example: Power down

Arthedian, 2019-05-22:
Huray, it is working, I placed the function after PPP stopped. Thank you for help

Arthedian, 2019-05-22:
I have one last issue with the example. How can i set pin for the sim card, so the code can connect to the sim?

suda-morris, 2019-05-23:
@Arthedian Congratulations!

The sim card is connected to your modem, not esp32, so I'm afraid you cannot set pin for the sim card.

negativekelvin, 2019-05-23:
You can unlock sim using AT+CPIN command

Arthedian, 2019-05-23:
Thank you, I just wasnt sure if there is any api for it. I succesfully integrated it into my code. My current problem is when I add Wifi connection. I want to be able to automatically switch between this two interfaces. If the wifi is working, that the internet should be travelling through wifi, but if the wifi is off, than it should be switch to the gsm. In the esp-idf 3.1 I had code which successfully was switching priority between this two interfaces, but in the current esp-idf 3.3 its not working because of this error:

> 'netif_list' undeclared (first use in this function)

```
if(config->wifi->status==3){
        if (DEBUGPRINT) printf("case2\n");
        if (canChange | (config->conn->connection!=1))
        {
            for (struct netif *pri = netif_list; pri != NULL; pri = pri->next)
            {
                if ((pri->name[0] == 's') && (pri->name[1] == 't'))
                {
                    //pausenetworkdependanttasks();
                    netif_set_default(pri);
                    //resumenetworkdependanttasks();
                    ESP_LOGD("connCareDefault", "Interface priority has been set to %c%c%d (" IPSTR "/" IPSTR " gateway " IPSTR ")",
                         pri->name[0], pri->name[1], pri->num,
                         IP2STR(&pri->ip_addr.u_addr.ip4), IP2STR(&pri->netmask.u_addr.ip4), IP2STR(&pri->gw.u_addr.ip4));
                }
            }
        }
        config->conn->connection=1;
        
    }
    else if(config->conn->hasGsm && config->gsm->status){
        if (DEBUGPRINT) printf("case3\n");
        for (struct netif *pri = netif_list; pri != NULL; pri = pri->next)
        {
            if ((pri->name[0] == 'p') && (pri->name[1] == 'p'))
            {
                //pausenetworkdependanttasks();
                netif_set_default(pri);
                //resumenetworkdependanttasks();
                ESP_LOGD("connCareDefault", "Interface priority has been set to %c%c%d (" IPSTR "/" IPSTR " gateway " IPSTR ")",
                     pri->name[0], pri->name[1], pri->num,
                     IP2STR(&pri->ip_addr.u_addr.ip4), IP2STR(&pri->netmask.u_addr.ip4), IP2STR(&pri->gw.u_addr.ip4));
            }
        }
        config->conn->connection=2;        
    }
    else{
        config->conn->connection=0;
    }
```

suda-morris, 2019-05-23:
@Arthedian Missing `#include "lwip/netif.h"` ?

Arthedian, 2019-05-23:
Thank you, but still I am not sure, why it worked in the old idf.

closed 2019-05-23 by Arthedian

suda-morris, 2019-05-23:
@Arthedian Maybe some include dependency has been changed and become cleaner in the ESP-IDF.