espressif/esp-idf #3359: httpd_req_delete dumps raw data to terminal

tsichevski, 2019-04-24:
## Environment

- IDF version v4.0-dev-402-ga20d02b7f

## Problem Description

The function httpd_req_delete() reads the http request bytes by 32-byte long portions and logs every portion unencoded to terminal with the following command:

ESP_LOGD(TAG, LOG_FMT("purging data : %s"), dummy);

where dummy points to the buffer containing the request data portion.

The problems:

1. The data the dummy points to is not a zero-terminated C string, so you may see a raw processor RAM dumped either;
2. This data is raw binary request data, in must not be send to a terminal unencoded;
3. The request may be of any length, dumping it to a terminal is a very bad idea.

### Expected Behavior

Not to dump the request contents to terminal

### Actual Behavior

Dumps raw request contents to terminal

### Steps to repropduce

1. Send a POST request with any binary payload;
2.  In the request handler code do NOT read the request contents, for example, just call httpd_resp_send_500() and return;
3. See all the request binary data along with some RAM contents dumped directly to your terminal :(

tsichevski, 2019-04-24:
UPD: in the latest esp-idf release the code writes the terminating zero, so the RAM contents should not be dumped anymore. The other issues remain.

anurag-kar, 2019-04-24:
@tsichevski 

> This data is raw binary request data, in must not be send to a terminal unencoded

This is only for debug purpose. In fact the log shouldn't event appear on the monitor unless log level is set to debug or verbose from `menuconfig`. Could you check if that's the case?

> The request may be of any length, dumping it to a terminal is a very bad idea.

If you look closely it is limited by the length of the dummy array, which is 32 in this case

Now coming back to your original issue:

> The data the dummy points to is not a zero-terminated C string, so you may see a raw processor RAM dumped either

As far as I recall the terminating zero has always been there since the component was introduced. You can check the git blame here https://github.com/espressif/esp-idf/blame/a20d02b7f196c407bc9f39b781e31a0a4f665968/components/esp_http_server/src/httpd_parse.c#L724

If you do suspect that there was this issue before you updated, would it be possible for you to attach the log? Because it could be that it was happening due of some other uncaught reason.

tsichevski, 2019-04-24:
Hi anurag-kar,

> @tsichevski
> 
> > This data is raw binary request data, in must not be send to a terminal unencoded
> 
> This is only for debug purpose. In fact the log shouldn't event appear on the monitor unless log level is set to debug or verbose from `menuconfig`. Could you check if that's the case?

I know. I have intentionally set my log level to DEBUG, because I need to use this level of logging to debug my programs. And dumping any raw binary data to a terminal usually breaks it very soon.

> 
> > The request may be of any length, dumping it to a terminal is a very bad idea.
> 
> If you look closely it is limited by the length of the dummy array, which is 32 in this case

The 'request length', not the 'dummy' buffer length. If I have a HTTP POST request, say, of 32 megabytes length, it will be split into more than one million of 32-byte chunks, and I will see more than 4 million messages in my terminal :-(
> Now coming back to your original issue:
>
>
> As far as I recall the terminating zero has always been there since the component was introduced. You can check the git blame here [...]
As I wrote in my previous comment, I have noticed it already, so this part RAM-related part is not an issue anymore, sorry. It would be nice if I can edit the original post to delete this very part of my complaint.

> If you do suspect that there was this issue before you updated, would it be possible for you to attach the logs from before? Because it could be that it was happening due to some other uncaught reason.

No, I don't.

Regards,
Vladimir

anurag-kar, 2019-04-24:
@tsichevski Thanks for the clarification. Now I understand the issue.

I guess a config option has to be added for deep debugging, disabled by
default, which will only display the raw data when enabled.

Will come up with a bug fix as soon as possible. Till then you’ll have to
manage by commenting out the line in the source code.

Again, thanks for cooperating :)

tsichevski, 2019-04-24:
> @tsichevski Thanks for the clarification. Now I understand the issue. I guess a config option has to be added for deep debugging, disabled by default, which will only display the raw data when enabled.
No, unknown binary bytes shall never be send to any text stream.
As to logging, I would propose the following:

1. Be more accurate choosing the severity level for events. I noticed currently both the 'fatal protocol error' and 'some data bytes received' situations are logged with the same DEBUG level :-( I would propose the usual scheme: FATAL then the program or a component is about to crash; WARNING if some input data is bad and will be rejected by the software; INFO the something countable happened, i.e. some component started, FINE when a request arrived/processed; DEBUG for request processing stages; TRACE for data processing details on the chunk level;
2. Introduce at least one more log level, say, TRACE, more verbose than debug for message that could really flood the log. The developer will switch this level on only for a short period of time, and only if he really needs it.
3. Make it possible to configure the log level on the per-component basis.

> Will come up with a bug fix as soon as possible. Till then you’ll have to manage by commenting out the line in the source code. Again, thanks for cooperating :)

Done it already, thanks!

negativekelvin, 2019-04-24:
>Make it possible to configure the log level on the per-component basis.

You can do this already

https://github.com/espressif/esp-idf/blob/3bb7dba9957ee71a236058322ae5e70f96f9a104/examples/protocols/mqtt/ws/main/app_main.c#L101-L107

tsichevski, 2019-04-25:
Did not know that. Thanks!

closed 2019-05-15 by igrr