espressif/esp-idf #3360: Example http_server / file_serving not able to set content-encoding gzip

arnold-b, 2019-04-24:
Hello, 

i´m using espidf master and i try to send z.b. menu.html.gz with encoding gzip. 
what ever i try in chrome i always get transfer-encoding: Chunked and i don´t get Content-Encoding gzip
 i tried to set  httpd_resp_set_type(req, "application/x-gzip"); and also 
tried  httpd_resp_set_type(req, "text/html");

the result is, that in chrome the request will be showed compressed. 

How to do ?

thanks

[image]

[image]

I would expect something like this.
[image]

static esp_err_t set_content_type_from_file(httpd_req_t *req,char * turi)
{
	
	
ESP_LOGI(TAG_FILESERVER, "Set TYPE ----------------------------------");

    if (IS_FILE_EXT(turi, ".pdf")) {
        return httpd_resp_set_type(req, "application/x-pdf");
    } else if (IS_FILE_EXT(turi, ".html")) {
        return httpd_resp_set_type(req, "text/html");
	} else if (IS_FILE_EXT(turi, ".htm")) {
		return httpd_resp_set_type(req, "text/html");
	} else if (IS_FILE_EXT(turi, ".css")) {
		return httpd_resp_set_type(req, "text/css");
	} else if (IS_FILE_EXT(turi, ".js")) {
		return httpd_resp_set_type(req, "application/javascript");
	} else if (IS_FILE_EXT(turi, ".jpeg")) {
        return httpd_resp_set_type(req, "image/jpeg");
    } else if (IS_FILE_EXT(turi, ".png")) {
        
        return httpd_resp_set_type(req, "image/png");        
    } else if (IS_FILE_EXT(turi, ".jpg")) {
        return httpd_resp_set_type(req, "image/jpeg");        
    } else if (IS_FILE_EXT(turi, ".ico")) {
        return httpd_resp_set_type(req, "image/x-icon");
    } else if (IS_FILE_EXT(turi, ".xml")) {
        return httpd_resp_set_type(req, "text/xml");
    } else if (IS_FILE_EXT(turi, ".zip")) {
        return httpd_resp_set_type(req, "application/x-zip");
    } else if (IS_FILE_EXT(turi, ".html.gz")) {
        ESP_LOGI(TAG_FILESERVER, "html.gz found \n");
        return httpd_resp_set_type(req, "text/html");

    } else if (IS_FILE_EXT(turi, ".gz")) {
        return httpd_resp_set_type(req, "application/x-gzip");   
    }
    /* This is a limited set only */
    /* For any other type always set as plain text */
    return httpd_resp_set_type(req, "text/html");
	
}

/* Send HTTP response with the contents of the requested file */
static esp_err_t http_resp_file(httpd_req_t *req)
{
    char filepath[FILE_PATH_MAX];
    FILE *fd = NULL;
    struct stat file_stat;

    /* Retrieve the base path of file storage to construct the full path */
    strcpy(filepath, ((struct file_server_data *)req->user_ctx)->base_path);

    /* Concatenate the requested file path */
	char * turi;  // Temporaty because didnt find why sometimes ? on the end
	turi = strdup(req->uri);
	
    ESP_LOGI(TAG_FILESERVER, "%s\n", req->uri);
	
	
    strcat(filepath, turi);

    if (stat(filepath, &file_stat) == -1) {
        strcat(filepath ,".gz");
        if ((stat(filepath, &file_stat) == -1)) {
        ESP_LOGE(TAG_FILESERVER, "Failed to stat file : %s", filepath);
        /* Respond with 404 Not Found */
        httpd_resp_send_err(req, HTTPD_404_NOT_FOUND, "File does not exist");
        return ESP_FAIL;
        }
    }

    fd = fopen(filepath, "r");
    if (!fd) {
        ESP_LOGE(TAG_FILESERVER, "Failed to read existing file : %s", filepath);
        /* Respond with 500 Internal Server Error */
        httpd_resp_send_err(req, HTTPD_500_INTERNAL_SERVER_ERROR, "Failed to read existing file");
        return ESP_FAIL;
    }
	set_content_type_from_file(req,filepath);
	//struct httpd_req_aux *ra = (httpd_req_aux*) req->aux;
	//char * content = ra->content_type;
	//ESP_LOGI(TAG_FILESERVER, "Sending file : %s type:%s (%ld bytes)...", filepath, content,file_stat.st_size);
	ESP_LOGI(TAG_FILESERVER, "Sending file : %s  (%ld bytes)...", filepath,file_stat.st_size);
    

    /* Retrieve the pointer to scratch buffer for temporary storage */
    char *chunk = ((struct file_server_data *)req->user_ctx)->scratch;
    size_t chunksize;
    do {
        /* Read file in chunks into the scratch buffer */
        chunksize = fread(chunk, 1, SCRATCH_BUFSIZE, fd);

        /* Send the buffer contents as HTTP response chunk */
        if (httpd_resp_send_chunk(req, chunk, chunksize) != ESP_OK) {
            fclose(fd);
            ESP_LOGE(TAG_FILESERVER, "File sending failed!");
            /* Abort sending file */
            httpd_resp_sendstr_chunk(req, NULL);
            /* Respond with 500 Internal Server Error */
            httpd_resp_send_err(req, HTTPD_500_INTERNAL_SERVER_ERROR, "Failed to send file");
            return ESP_FAIL;
        }

        /* Keep looping till the whole file is sent */
    } while (chunksize != 0);

    /* Close file after sending complete */
    fclose(fd);
    ESP_LOGI(TAG_FILESERVER, "File sending complete");

    /* Respond with an empty chunk to signal HTTP response completion */
    httpd_resp_send_chunk(req, NULL, 0);
    return ESP_OK;
}
D (16041) httpd_uri: httpd_find_uri_handler: [1] = /*
I (16051) file_server: /menu.html

I (16131) file_server: Set TYPE ----------------------------------
I (16131) file_server: html.gz found 

I (16131) file_server: Sending file : /spiffs/menu.html.gz  (119368 bytes)...
D (16211) httpd_txrx: httpd_send_all: sent = 70
D (16211) httpd_txrx: httpd_send_all: sent = 2
D (16211) httpd_txrx: httpd_send_all: sent = 6
D (16221) httpd_txrx: httpd_send_all: sent = 8192
D (16221) httpd_txrx: httpd_send_all: sent = 2
D (16241) httpd_txrx: httpd_send_all: sent = 6
D (16241) httpd_txrx: httpd_send_all: sent = 8192
D (16241) httpd_txrx: httpd_send_all: sent = 2
D (16731) httpd_txrx: httpd_send_all: sent = 6
D (16741) httpd_txrx: httpd_send_all: sent = 8192
D (16741) httpd_txrx: httpd_send_all: sent = 2
D (16771) httpd_txrx: httpd_send_all: sent = 6
D (16771) httpd_txrx: httpd_send_all: sent = 8192
D (16771) httpd_txrx: httpd_send_all: sent = 2
D (16781) httpd_txrx: httpd_send_all: sent = 6
D (16781) httpd_txrx: httpd_send_all: sent = 8192
D (16791) httpd_txrx: httpd_send_all: sent = 2
D (16831) httpd_txrx: httpd_send_all: sent = 6
D (16831) httpd_txrx: httpd_send_all: sent = 8192
D (16831) httpd_txrx: httpd_send_all: sent = 2
D (16861) httpd_txrx: httpd_send_all: sent = 6
D (16861) httpd_txrx: httpd_send_all: sent = 8192
D (16861) httpd_txrx: httpd_send_all: sent = 2
D (16871) httpd_txrx: httpd_send_all: sent = 6
D (16881) httpd_txrx: httpd_send_all: sent = 8192
D (16881) httpd_txrx: httpd_send_all: sent = 2
D (16921) httpd_txrx: httpd_send_all: sent = 6
D (16921) httpd_txrx: httpd_send_all: sent = 8192
D (16921) httpd_txrx: httpd_send_all: sent = 2
D (16931) httpd_txrx: httpd_send_all: sent = 6
D (16931) httpd_txrx: httpd_send_all: sent = 8192
D (16931) httpd_txrx: httpd_send_all: sent = 2
D (16961) httpd_txrx: httpd_send_all: sent = 6
D (16971) httpd_txrx: httpd_send_all: sent = 8192
D (16971) httpd_txrx: httpd_send_all: sent = 2
D (16981) httpd_txrx: httpd_send_all: sent = 6
D (16981) httpd_txrx: httpd_send_all: sent = 8192
D (16981) httpd_txrx: httpd_send_all: sent = 2
D (17001) httpd_txrx: httpd_send_all: sent = 6
D (17011) httpd_txrx: httpd_send_all: sent = 8192
D (17011) httpd_txrx: httpd_send_all: sent = 2
D (17011) httpd_txrx: httpd_send_all: sent = 6
D (17021) httpd_txrx: httpd_send_all: sent = 8192
D (17021) httpd_txrx: httpd_send_all: sent = 2
D (17041) httpd_txrx: httpd_send_all: sent = 6
D (17041) httpd_txrx: httpd_send_all: sent = 4680
D (17041) httpd_txrx: httpd_send_all: sent = 2
D (17041) httpd_txrx: httpd_send_all: sent = 3
D (17041) httpd_txrx: httpd_send_all: sent = 2
I (17051) file_server: File sending complete
D (17051) httpd_txrx: httpd_send_all: sent = 3
D (17061) httpd_txrx: httpd_send_all: sent = 2
D (17061) httpd_sess: httpd_sess_process: httpd_req_delete
D (17071) httpd_sess: httpd_sess_process: success
D (17071) httpd: httpd_server: processing listen socket 54
D (17081) httpd: httpd_accept_conn: newfd = 58
D (17081) httpd_sess: httpd_sess_new: fd = 58
D (17091) httpd: httpd_accept_conn: complete
D (17091) httpd: httpd_server: doing select maxfd+1 = 59
D (17301) httpd: httpd_server: processing socket 57
D (17301) httpd_sess: httpd_sess_process: httpd_req_new
D (17301) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (17311) httpd_txrx: httpd_recv_with_opt: received length = 0
D (17311) httpd_parse: read_block: connection closed
D (17321) httpd: httpd_server: closing socket 57
D (17321) httpd_sess: httpd_sess_delete: fd = 57
D (17331) httpd: httpd_server: doing select maxfd+1 = 59
D (107001) httpd: httpd_server: processing socket 58
D (107001) httpd_sess: httpd_sess_process: httpd_req_new
D (107001) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (107011) httpd_txrx: httpd_recv_with_opt: received length = 0
D (107021) httpd_parse: read_block: connection closed
D (107021) httpd: httpd_server: closing socket 58
D (107031) httpd_sess: httpd_sess_delete: fd = 58
D (107031) httpd: httpd_server: doing select maxfd+1 = 56

negativekelvin, 2019-04-24:
You could cheat and try
`return httpd_resp_set_type(req, "text/html\r\nContent-Encoding: gzip");`

anurag-kar, 2019-04-25:
@arnold-b Following the solution by @negativekelvin should work

arnold-b, 2019-04-25:
Hey, thanks.  It works 
Before i just used ESP-arduino. I need to learn a lot to get my application running again

closed 2019-04-25 by arnold-b