espressif/esp-idf #3361: mcpwm basic config - no synchronization happening, why?

bobkasterop, 2019-04-25:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ x] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      |ESP32-DevKitC
- Kit version (: [v4]
- Module or chip used:  [ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows]
- Power Supply:         [USB|
## Problem Description
im using the example code "MCPWM basic config" to generate 3 PWM signals. i need these 3 signals to be synchronized but when i use the enable_sync function, nothing happens. the signals are still not synchronized. 
see code.
//Detailed problem description goes here.

### Expected Behavior
3 synchronized pwm signals
### Actual Behavior
3 pwm signals not starting the same place (not synchronized)
### Steps to repropduce

1. step1
generate pwm signals
2. ...
synchronize using the sync enable function

// It helps if you attach a picture of your setup/wiring here.
[image]

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
static void mcpwm_example_config(void *arg)
{
    //1. mcpwm gpio initialization
    mcpwm_example_gpio_initialize();

    //2. initialize mcpwm configuration
    printf("Configuring Initial Parameters of mcpwm...\n");

    mcpwm_config_t pwm_config;
    pwm_config.frequency = 5000;    //frequency = 1000Hz
    pwm_config.cmpr_a = 50.0;    //duty cycle of PWMxA = 50.0%
    pwm_config.cmpr_b = 50.0;    //duty cycle of PWMxb = 50.0%
    pwm_config.counter_mode = MCPWM_UP_DOWN_COUNTER;
    pwm_config.duty_mode = MCPWM_DUTY_MODE_0;
    mcpwm_init(MCPWM_UNIT_0, MCPWM_TIMER_0, &pwm_config);    //Configure PWM0A & PWM0B with above settings
    
    mcpwm_init(MCPWM_UNIT_0, MCPWM_TIMER_1, &pwm_config);    //Configure PWM1A & PWM1B with above settings
  
    mcpwm_init(MCPWM_UNIT_0, MCPWM_TIMER_2, &pwm_config);    //Configure PWM2A & PWM2B with above settings
    
    printf("Configuration Done \n");

#if MCPWM_EN_SYNC
    //6. Syncronization configuration
    //comment if you don't want to use sync submodule, also u can comment the sync gpio signals
    //here synchronization occurs on PWM1A and PWM1B
    mcpwm_sync_enable(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_SELECT_SYNC0, 200);    //Load counter value with 20% of period counter of mcpwm timer 1 when sync 0 occurs
    mcpwm_sync_enable(MCPWM_UNIT_0, MCPWM_TIMER_1, MCPWM_SELECT_SYNC0, 200);
    mcpwm_sync_enable(MCPWM_UNIT_0, MCPWM_TIMER_2, MCPWM_SELECT_SYNC0, 200);
#endif

    vTaskDelete(NULL);
}

void app_main()
{
    printf("Testing MCPWM...\n");
    xTaskCreate(mcpwm_example_config, "mcpwm_example_config", 4096, NULL, 5, NULL);
    printf("Exit main \n");
}
```

there is some initial #define lines of code that i didnt post but these are exactly the same as for the example code im using.

koobest, 2019-04-25:
Hi, @bobkasterop 
You should connect a external sync signal to the pin  `GPIO_SYNC0_IN` . The sync signal is active low.

[image]

thanks !!

negativekelvin, 2019-04-25:
I think he wants to use the internal sync feature, not external gpio.

bobkasterop, 2019-04-26:
yes im trynna use internal sync feature.

koobest, 2019-04-28:
Hi, @bobkasterop
The following code can get the synchronous PWM, please  try it. Since we don't have an API to do this, we operate the registers directly.

```
static void mcpwm_example_config(void *arg)
{
    //1. mcpwm gpio initialization
    mcpwm_example_gpio_initialize();

    //2. initialize mcpwm configuration
    printf("Configuring Initial Parameters of mcpwm...\n");

    mcpwm_config_t pwm_config;
    pwm_config.frequency = 5000;    //frequency = 1000Hz
    pwm_config.cmpr_a = 0.0;    //duty cycle of PWMxA = 0%
    pwm_config.cmpr_b = 0.0;    //duty cycle of PWMxb = 0%
    pwm_config.counter_mode = MCPWM_UP_DOWN_COUNTER;
    pwm_config.duty_mode = MCPWM_DUTY_MODE_0;
    mcpwm_init(MCPWM_UNIT_0, MCPWM_TIMER_0, &pwm_config);
    mcpwm_init(MCPWM_UNIT_0, MCPWM_TIMER_1, &pwm_config);
    mcpwm_init(MCPWM_UNIT_0, MCPWM_TIMER_2, &pwm_config);
    printf("Configuration Done \n");
    mcpwm_sync_enable(MCPWM_UNIT_0, MCPWM_TIMER_0, 1, 200);
    mcpwm_sync_enable(MCPWM_UNIT_0, MCPWM_TIMER_1, 1, 200);
    mcpwm_sync_enable(MCPWM_UNIT_0, MCPWM_TIMER_2, 1, 200);
    MCPWM0.timer[0].sync.out_sel = 1;
    vTaskDelay(1000/portTICK_PERIOD_MS);
    MCPWM0.timer[0].sync.out_sel = 0;
    mcpwm_set_duty(MCPWM_UNIT_0, MCPWM_TIMER_0,  MCPWM_OPR_A, 50);
    mcpwm_set_duty(MCPWM_UNIT_0, MCPWM_TIMER_0,  MCPWM_OPR_B, 50);
    mcpwm_set_duty(MCPWM_UNIT_0, MCPWM_TIMER_1,  MCPWM_OPR_A, 50);
    mcpwm_set_duty(MCPWM_UNIT_0, MCPWM_TIMER_1,  MCPWM_OPR_B, 50);
    mcpwm_set_duty(MCPWM_UNIT_0, MCPWM_TIMER_2,  MCPWM_OPR_A, 50);
    mcpwm_set_duty(MCPWM_UNIT_0, MCPWM_TIMER_2,  MCPWM_OPR_B, 50);
    vTaskDelete(NULL);
}
```
thanks !!

bobkasterop, 2019-04-28:
it works perfectly! thank you very much :+1:

Alvin1Zhang, 2019-05-20:
@bobkasterop Thanks for reporting the issue. Feel free to reopen if your issue still exists. Thanks.

closed 2019-05-20 by Alvin1Zhang

JamshidJD, 2023-06-23:
I'm trying to use MCPWM_UP_DOWN_COUNTER
and my frequency range is 10K - 100K, i know i have to provide double the frequency to generate the center aligned output. but the factor of twice the frequency changes when i above 12K, for instance if i want to generate 100K i'm providing 200K as frequency but the output is 125K instead. I've also tried to make a loop to see the linearity but by decreasing the value 200K then next jumps comes at around 83K and so on...., this only happens in MCPWM_UP_DOWN_COUNTER if i am using the other mode its fine. Even if i try to generate 50kHz it generate 55kHz
The code snippet is attached below, any help would be appreciated. thanks

mcpwm_config_t pwm_configA;
pwm_configA.frequency = 100000; // frequência = 500Hz,
pwm_configA.cmpr_a = 50; // Ciclo de trabalho (duty cycle) do PWMxA = 0
pwm_configA.cmpr_b = 50; // Ciclo de trabalho (duty cycle) do PWMxb = 0
pwm_configA.counter_mode = MCPWM_UP_DOWN_COUNTER; // Para MCPWM assimetrico
pwm_configA.duty_mode = MCPWM_DUTY_MODE_0; // Define ciclo de trabalho em nível alto

mcpwm_gpio_init(MCPWM_UNIT_0, MCPWM0A, Pulse_Pin_1);
mcpwm_gpio_init(MCPWM_UNIT_0, MCPWM0B, Pulse_Pin_2);
mcpwm_init(MCPWM_UNIT_0, MCPWM_TIMER_0, &pwm_configA); // Define PWM0A & PWM0B com as configurações acima

int DutyCycle = 50;
mcpwm_set_duty(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_OPR_A, DutyCycle);
mcpwm_set_duty(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_OPR_B, 100 - DutyCycle); // Configura a porcentagem do PWM no Operador B (Ciclo de trabalho)
mcpwm_set_duty_type(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_OPR_A, MCPWM_DUTY_MODE_0);
mcpwm_set_duty_type(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_OPR_B, MCPWM_DUTY_MODE_1);