espressif/esp-idf #3362: Airkiss has a bug

Feng573795502, 2019-04-25:
The phone has been connected for a timeout, but the log shows that it has been successfully connected.

Alvin1Zhang, 2019-04-25:
@Feng573795502 Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Thanks.

Feng573795502, 2019-04-26:
Change to airkiss using examples\smart_config
`void smartconfig_example_task(void * parm)
{
    EventBits_t uxBits;
    esp_smartconfig_fast_mode(0);
    ESP_ERROR_CHECK( esp_smartconfig_set_type(SC_TYPE_ESPTOUCH_AIRKISS) );  /* KISS */
    ESP_ERROR_CHECK( esp_smartconfig_start(sc_callback) );
    while (1) {
        uxBits = xEventGroupWaitBits(s_wifi_event_group, CONNECTED_BIT | ESPTOUCH_DONE_BIT, true, false, portMAX_DELAY); 
        if(uxBits & CONNECTED_BIT) {
            ESP_LOGI(TAG, "WiFi Connected to ap");
        }
        if(uxBits & ESPTOUCH_DONE_BIT) {
            ESP_LOGI(TAG, "smartconfig over");
            esp_smartconfig_stop();
            vTaskDelete(NULL);
        }
    }
}`
log:
[0;32mI (297) phy: phy_version: 4100, 6fa5e27, Jan 25 2019, 17:02:06, 0, 0[0m
I (297) wifi: mode : sta (24:0a:c4:bb:25:18)
I (347) smartconfig: SC version: V2.8.0

[08:39:00.617]收←◆I (4447) wifi: ic_enable_sniffer
[0;32mI (4447) sc: SC_STATUS_FINDING_CHANNEL[0m

[08:39:21.668]收←◆I (25497) smartconfig: TYPE: AIRKISS
I (25497) smartconfig: T|PHONE MAC: fc:18:3c:30:62:f2
I (25497) smartconfig: T|AP MAC:54:75:95:00:be:8f
[0;32mI (25497) sc: SC_STATUS_GETTING_SSID_PSWD[0m

[08:39:24.824]收←◆I (28647) smartconfig: T|pswd: toprie1.com
I (28647) smartconfig: T|ssid: TOPRIE_AP1
I (28647) wifi: ic_disable_sniffer
[0;32mI (28647) sc: SC_STATUS_LINK[0m
[0;32mI (28657) sc: SSID:TOPRIE_AP1[0m
[0;32mI (28657) sc: PASSWORD:toprie1.com[0m

[08:39:24.956]收←◆I (28787) wifi: new:<11,0>, old:
[08:39:25.049]收←◆<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (28787) wifi: state: init -> auth (b0)
I (28797) wifi: state: auth -> assoc (0)
I (28807) wifi: state: assoc -> run (10)
I (28817) wifi: connected with TOPRIE_AP1, channel 11, bssid = 54:75:95:00:be:8f
I (28827) wifi: pm start, type: 1

[08:39:27.308]收←◆[0;32mI (31137) event: sta ip: 192.168.1.64, mask: 255.255.255.0, gw: 192.168.1.1[0m
[0;32mI (31137) sc: WiFi Connected to ap[0m

[08:39:30.326]收←◆[0;32mI (34157) sc: SC_STATUS_LINK_OVER[0m
[0;32mI (34157) sc: Phone ip: 63.88.158.251
[0m
[0;32mI (34157) sc: smartconfig over[0m

IPHONE XR + ESP32 lyrat v4.3

At this point the phone is still waiting until the eject device connection times out.

Feng573795502, 2019-04-28:
> @Feng573795502 Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Thanks.

Change to airkiss using examples\smart_config
void smartconfig_example_task(void * parm) { EventBits_t uxBits; esp_smartconfig_fast_mode(0); ESP_ERROR_CHECK( esp_smartconfig_set_type(SC_TYPE_ESPTOUCH_AIRKISS) ); /* KISS */ ESP_ERROR_CHECK( esp_smartconfig_start(sc_callback) ); while (1) { uxBits = xEventGroupWaitBits(s_wifi_event_group, CONNECTED_BIT | ESPTOUCH_DONE_BIT, true, false, portMAX_DELAY); if(uxBits & CONNECTED_BIT) { ESP_LOGI(TAG, "WiFi Connected to ap"); } if(uxBits & ESPTOUCH_DONE_BIT) { ESP_LOGI(TAG, "smartconfig over"); esp_smartconfig_stop(); vTaskDelete(NULL); } } }
log:
�[0;32mI (297) phy: phy_version: 4100, 6fa5e27, Jan 25 2019, 17:02:06, 0, 0�[0m
I (297) wifi: mode : sta (24:0a:c4:bb:25:18)
I (347) smartconfig: SC version: V2.8.0

[08:39:00.617]收←◆I (4447) wifi: ic_enable_sniffer
�[0;32mI (4447) sc: SC_STATUS_FINDING_CHANNEL�[0m

[08:39:21.668]收←◆I (25497) smartconfig: TYPE: AIRKISS
I (25497) smartconfig: T|PHONE MAC: fc:18:3c:30:62:f2
I (25497) smartconfig: T|AP MAC:54:75:95:00:be:8f
�[0;32mI (25497) sc: SC_STATUS_GETTING_SSID_PSWD�[0m

[08:39:24.824]收←◆I (28647) smartconfig: T|pswd: toprie1.com
I (28647) smartconfig: T|ssid: TOPRIE_AP1
I (28647) wifi: ic_disable_sniffer
�[0;32mI (28647) sc: SC_STATUS_LINK�[0m
�[0;32mI (28657) sc: SSID:TOPRIE_AP1�[0m
�[0;32mI (28657) sc: PASSWORD:toprie1.com�[0m

[08:39:24.956]收←◆I (28787) wifi: new:<11,0>, old:
[08:39:25.049]收←◆<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (28787) wifi: state: init -> auth (b0)
I (28797) wifi: state: auth -> assoc (0)
I (28807) wifi: state: assoc -> run (10)
I (28817) wifi: connected with TOPRIE_AP1, channel 11, bssid = 54:75:95:00:be:8f
I (28827) wifi: pm start, type: 1

[08:39:27.308]收←◆�[0;32mI (31137) event: sta ip: 192.168.1.64, mask: 255.255.255.0, gw: 192.168.1.1�[0m
�[0;32mI (31137) sc: WiFi Connected to ap�[0m

[08:39:30.326]收←◆�[0;32mI (34157) sc: SC_STATUS_LINK_OVER�[0m
�[0;32mI (34157) sc: Phone ip: 63.88.158.251
�[0m
�[0;32mI (34157) sc: smartconfig over�[0m

IPHONE XR + ESP32 lyrat v4.3

At this point the phone is still waiting until the eject device connection times out.

liuzfesp, 2019-05-24:
HI @zhangyanjiaoesp, please help.

zhangyanjiaoesp, 2019-11-06:
@Feng573795502 Can you have a try with the latest IDF version? We have some bugfixs for samrtconfig.

closed 2020-02-13 by espressif-bot