espressif/esp-idf #3363: Example http_server / file_serving dont transfer the right content

arnold-b, 2019-04-25:
Hello, 

I´m using espidf master, Visual Code
If i try the sample http_server / file_serving, i dont get the right content. 

As i think, every recieving is just 77Byte 

How to do ?
there one sample: 

[image]

Original Size should be: 2775 bytes

Log: 
 (528911) file_server: /led.html

I (528921) file_server: Set TYPE ----------------------------------
I (528921) file_server: html found 

I (528931) file_server: Sending file : /spiffs/led.html  (2775 bytes)...
E (528941) file_server: File read 0
D (528941) httpd_txrx: httpd_send_all: sent = 70
D (528951) httpd_txrx: httpd_send_all: sent = 2
D (528951) httpd_txrx: httpd_send_all: sent = 3
D (528951) httpd_txrx: httpd_send_all: sent = 2
I (528961) file_server: File sending complete
D (528961) httpd_txrx: httpd_send_all: sent = 3
D (528971) httpd_txrx: httpd_send_all: sent = 2
D (528971) httpd_sess: httpd_sess_process: httpd_req_delete
D (528981) httpd_sess: httpd_sess_process: success
D (528981) httpd: httpd_server: doing select maxfd+1 = 62
D (529171) httpd: httpd_server: processing socket 61
D (529171) httpd_sess: httpd_sess_process: httpd_req_new
D (529171) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (529171) httpd_txrx: httpd_recv_with_opt: received length = 0
D (529181) httpd_parse: read_block: connection closed
D (529181) httpd: httpd_server: closing socket 61
D (529191) httpd_sess: httpd_sess_delete: fd = 61
D (529191) httpd: httpd_server: doing select maxfd+1 = 56
D (531521) httpd: httpd_server: processing listen socket 54
D (531521) httpd: httpd_accept_conn: newfd = 62
D (531521) httpd_sess: httpd_sess_new: fd = 62
D (531521) httpd: httpd_accept_conn: complete
D (531531) httpd: httpd_server: doing select maxfd+1 = 63
D (599721) httpd: httpd_server: processing socket 62
D (599721) httpd_sess: httpd_sess_process: httpd_req_new
D (599721) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (599721) httpd_txrx: httpd_recv_with_opt: received length = 0
D (599731) httpd_parse: read_block: connection closed
D (599731) httpd: httpd_server: closing socket 62
D (599741) httpd_sess: httpd_sess_delete: fd = 62
D (599741) httpd: httpd_server: doing select maxfd+1 = 56

Source:

static esp_err_t set_content_type_from_file(httpd_req_t *req,char * turi)
{
	
	std::string cmd = "";

ESP_LOGI(TAG_FILESERVER, "Set TYPE ----------------------------------");
//	char * turi;   // Temporaty because didnt find why sometimes ? on the end
//	turi = strdup(req->uri);
	
	if (turi[strlen(turi) - 1] == '?') // Remove if last char == ? Workaround 
		{ 
			ESP_LOGI(TAG_FILESERVER, "'?' found at position .\n");
			turi[strlen(turi) - 1] = '\0';
		
		 
		}
    if (IS_FILE_EXT(turi, ".pdf")) {
        return httpd_resp_set_type(req, "application/x-pdf");
    } else if (IS_FILE_EXT(turi, ".html")) {
        ESP_LOGI(TAG_FILESERVER, "html found \n");
        return httpd_resp_set_type(req, "text/html");
	} else if (IS_FILE_EXT(turi, ".htm")) {
		return httpd_resp_set_type(req, "text/html");
	} else if (IS_FILE_EXT(turi, ".css")) {
		return httpd_resp_set_type(req, "text/css");
	} else if (IS_FILE_EXT(turi, ".js")) {
		return httpd_resp_set_type(req, "application/javascript");
	} else if (IS_FILE_EXT(turi, ".jpeg")) {
        return httpd_resp_set_type(req, "image/jpeg");
    } else if (IS_FILE_EXT(turi, ".png")) {
        
        return httpd_resp_set_type(req, "image/png");        
    } else if (IS_FILE_EXT(turi, ".jpg")) {
        return httpd_resp_set_type(req, "image/jpeg");        
    } else if (IS_FILE_EXT(turi, ".ico")) {
        return httpd_resp_set_type(req, "image/x-icon");
    } else if (IS_FILE_EXT(turi, ".xml")) {
        return httpd_resp_set_type(req, "text/xml");
    } else if (IS_FILE_EXT(turi, ".zip")) {
        return httpd_resp_set_type(req, "application/x-zip");
    } else if (IS_FILE_EXT(turi, ".html.gz")) {
        ESP_LOGI(TAG_FILESERVER, "html.gz found \n");

        return httpd_resp_set_type(req, "text/html");

    } else if (IS_FILE_EXT(turi, ".gz")) {
        return httpd_resp_set_type(req, "application/x-gzip");   
    }
    /* This is a limited set only */
    /* For any other type always set as plain text */
    return httpd_resp_set_type(req, "text/html");
	
}

/* Send HTTP response with the contents of the requested file */
static esp_err_t http_resp_file(httpd_req_t *req)
{
    char filepath[FILE_PATH_MAX];
    FILE *fd = NULL;
    struct stat file_stat;

    /* Retrieve the base path of file storage to construct the full path */
    strcpy(filepath, ((struct file_server_data *)req->user_ctx)->base_path);

    /* Concatenate the requested file path */
	char * turi;  // Temporaty because didnt find why sometimes ? on the end
	turi = strdup(req->uri);
	
	if (turi[strlen(turi) - 1] == '?') // Remove if last char == ? Workaround 
		{ 
			ESP_LOGI(TAG_FILESERVER, "'?' found at position .\n");
			turi[strlen(turi) - 1] = '\0';
		
		 
		}
	ESP_LOGI(TAG_FILESERVER, "%s\n", req->uri);
	
	
    strcat(filepath, turi);
    //httpd_resp_set_hdr(req, "Content-Encoding", "deflate");
    if (stat(filepath, &file_stat) == -1) {
        strcat(filepath ,".gz");
        //httpd_resp_set_hdr(req, "Content-Encoding", "gzip");
        if ((stat(filepath, &file_stat) == -1)) {
        ESP_LOGE(TAG_FILESERVER, "Failed to stat file : %s", filepath);
        /* Respond with 404 Not Found */
        httpd_resp_send_err(req, HTTPD_404_NOT_FOUND, "File does not exist");
        return ESP_FAIL;
        }
    }

    fd = fopen(filepath, "r");
    if (!fd) {
        ESP_LOGE(TAG_FILESERVER, "Failed to read existing file : %s", filepath);
        /* Respond with 500 Internal Server Error */
        httpd_resp_send_err(req, HTTPD_500_INTERNAL_SERVER_ERROR, "Failed to read existing file");
        return ESP_FAIL;
    }
	set_content_type_from_file(req,filepath);
	//struct httpd_req_aux *ra = (httpd_req_aux*) req->aux;
	//char * content = ra->content_type;
	//ESP_LOGI(TAG_FILESERVER, "Sending file : %s type:%s (%ld bytes)...", filepath, content,file_stat.st_size);
	ESP_LOGI(TAG_FILESERVER, "Sending file : %s  (%ld bytes)...", filepath,file_stat.st_size);
    

    /* Retrieve the pointer to scratch buffer for temporary storage */
    char *chunk = ((struct file_server_data *)req->user_ctx)->scratch;
    size_t chunksize;
    do {
        /* Read file in chunks into the scratch buffer */
        chunksize = fread(chunk, 1, SCRATCH_BUFSIZE, fd);
        ESP_LOGE(TAG_FILESERVER, "File read %d",chunksize);

        /* Send the buffer contents as HTTP response chunk */
        if (httpd_resp_send_chunk(req, chunk, chunksize) != ESP_OK) {
            fclose(fd);
            ESP_LOGE(TAG_FILESERVER, "File sending failed!");
            /* Abort sending file */
            httpd_resp_sendstr_chunk(req, NULL);
            /* Respond with 500 Internal Server Error */
            httpd_resp_send_err(req, HTTPD_500_INTERNAL_SERVER_ERROR, "Failed to send file");
            return ESP_FAIL;
        }

        /* Keep looping till the whole file is sent */
    } while (chunksize != 0);

    /* Close file after sending complete */
    fclose(fd);
    ESP_LOGI(TAG_FILESERVER, "File sending complete");

    /* Respond with an empty chunk to signal HTTP response completion */
    httpd_resp_send_chunk(req, NULL, 0);
    return ESP_OK;
}

anurag-kar, 2019-04-25:
@arnold-b From the log it can be seen that `fread()` is returning 0

```
chunksize = fread(chunk, 1, SCRATCH_BUFSIZE, fd);
ESP_LOGE(TAG_FILESERVER, "File read %d",chunksize);
```

Next the call to `httpd_req_send_chunk()` sends only the HTTP headers and no body as chunksize is zero, that's why you have those 77 bytes.

The question is why fread() returns

arnold-b, 2019-04-25:
It´s nothing changd on this. Thats the original source from example

anurag-kar, 2019-04-25:
@arnold-b Could you tell me how you upload the SPIFFS data?

arnold-b, 2019-04-25:
I uploaded it on 2 diffrent ways. one with : 

mkspiffs -c data -s 0x280000 image.bin
python esptool.py --chip esp32 --port COM6 --baud 256000 write_flash -z 0x110000 image.bin

and the current one over Arduino and vmicro. for upload they use esptool.exe 
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,      data, nvs,     0x9000,  0x5000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 1M,
storage,  data, spiffs,0x191000   ,      0x26F000  

thanks

arnold-b, 2019-04-25:
I also tried in menuconfig  DSPIFFS_OBJ_META_LEN=4 and DSPIFFS_OBJ_META_LEN=0

anurag-kar, 2019-04-25:
@arnold-b Check this issue.

Before that, just to confirm that the file server is working all right, try the following:

From the monitor log note what the IP assigned to your ESP32 is on the local network. It will look something like this:
```
I (4710) tcpip_adapter: sta ip: 10.0.0.84, mask: 255.255.254.0, gw: 10.0.0.1
```
Now open a webrowser  and in the address bar type `http://10.0.0.84`.
You should see a webpage with a table and an upload button on the top left corner. Just upload your html using that (anything with name other than index.html which is reserved). Now the table should display your html file . Try opening it. Let me know if that works

arnold-b, 2019-04-25:
tcpip_adapter: sta ip: 192.168.178.111, mask: 255.255.255.0, gw: 192.168.178.1
OK, then i need to format first spiffs:

arnold-b, 2019-04-25:
If i do like this. then the Webpage is OK

anurag-kar, 2019-04-25:
@arnold-b I suppose this has to do with the how the SPIFFS image was generated. Please refer to https://github.com/espressif/esp-idf/issues/1437

arnold-b, 2019-04-25:
My Partition is 
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,      data, nvs,     0x9000,  0x5000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 1M,
storage,  data, spiffs,0x110000   ,      0x26F000  

so i use now 
Setting OBJ_META_LEN is set to zero: 
then
./mkspiffs -c data -b 4096 -p 256 -s 0x26F000 spiffs.bin

and then 
esptool.py --chip esp32 --port COM6 --baud 1500000 write_flash -z 0x110000 spiffs.bin

But its still not working. Something i´m doing wrong

One question more: is it normal, that on the webpage the files are showed without sub directories. Its only a flat list ?

arnold-b, 2019-04-25:
CONFIG_SPIFFS_MAX_PARTITIONS=4
CONFIG_SPIFFS_CACHE=y
CONFIG_SPIFFS_CACHE_WR=y
# CONFIG_SPIFFS_CACHE_STATS is not set
CONFIG_SPIFFS_PAGE_CHECK=y
CONFIG_SPIFFS_GC_MAX_RUNS=10
CONFIG_SPIFFS_GC_STATS=y
CONFIG_SPIFFS_PAGE_SIZE=256
CONFIG_SPIFFS_OBJ_NAME_LEN=64
CONFIG_SPIFFS_USE_MAGIC=y
CONFIG_SPIFFS_USE_MAGIC_LENGTH=y
CONFIG_SPIFFS_META_LENGTH=0
CONFIG_SPIFFS_DBG=y

anurag-kar, 2019-04-25:
> One question more: is it normal, that on the webpage the files are showed without sub directories. Its only a flat list ?

Yes spiffs doesn't support directories.

arnold-b, 2019-04-25:
Now i did this and still problem:
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,      data, nvs,     0x9000,  0x5000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 1M,
storage,  data, spiffs,0x110000   ,      1M,  

C:\ESP32\Rollo>mkspiffs -c test -b 4096 -p 256 -s 0x100000 spiffs.bin
/CMakeLists.txt
/CMakeLists.txt.old
/json.html
/led.html
/leddlg.html
/lg.html
/linear-min-non-zero.html
/linear.html
/m.html
/main.html
/Makefile
/menu.html.gz
/mod.html
/mq.html
/olfjava.txt
/outrange.html
/pwmdlg.html

C:\ESP32\Rollo>esptool.py --chip esp32 --port COM6 --baud 1500000 write_flash -z 0x110000 spiffs.bin
esptool.py v2.6
Serial port COM6
Connecting........_
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, Coding Scheme None
MAC: 30:ae:a4:1e:8d:f0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 1500000
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 1048576 bytes to 143395...
Wrote 1048576 bytes (143395 compressed) at 0x00110000 in 3.3 seconds (effective 2570.0 kbit/s)...
Hash of data verified.

C:\ESP32\Rollo>mkspiffs.exe --version
mkspiffs ver. 0.2.3
Build configuration name: arduino-esp32
SPIFFS ver. 0.3.7-5-gf5e26c4
Extra build flags: -DSPIFFS_OBJ_META_LEN=4
SPIFFS configuration:
  SPIFFS_OBJ_NAME_LEN: 32
  SPIFFS_OBJ_META_LEN: 4  ---------------  It showes 4 but in SDKConfig its 0
  SPIFFS_USE_MAGIC: 1
  SPIFFS_USE_MAGIC_LENGTH: 1
  SPIFFS_ALIGNED_OBJECT_INDEX_TABLES: 0

SDKConfig
CONFIG_SPIFFS_USE_MAGIC_LENGTH=y
CONFIG_SPIFFS_META_LENGTH=0
CONFIG_SPIFFS_DBG=y

arnold-b, 2019-04-25:
Now i changed SPIFFS_OBJ_META_LEN=4 back to 4 and used mkspiffs again. 
Still the same Problem

anurag-kar, 2019-04-25:
@arnold-b Are you using mkspiffs executable downloaded from here https://github.com/igrr/mkspiffs/releases/download/0.2.3/mkspiffs-0.2.3-arduino-esp32-win32.zip ?

arnold-b, 2019-04-25:
Yes

anurag-kar, 2019-04-25:
@arnold-b I just tested with the linux executable of mkspiffs and it does work for me with the default configuration of for SPIFFS. I'll see if I can get any of the devs to try out the win32 executable

arnold-b, 2019-04-25:
can you post your image of the spiffs image and the partition information for it

anurag-kar, 2019-04-25:
This is my `partition_examples.csv`
```
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 1M,
storage,  data, spiffs,  ,        0xF0000, 
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 1M,
storage,  data, spiffs,  ,        0xF0000, 
```

The commands I use :
```
>> ./mkspiffs -c simple-webpage/ -b 4096 -p 256 -s 0xF0000 spiffs.bin
/README.md
/CODE_OF_CONDUCT.md
/test.html
/styles/style.css
/LICENSE
/images/firefox-icon.png

>> $IDF_PATH/components/esptool_py/esptool/esptool.py write_flash 0x110000 spiffs.bin
esptool.py v2.7-dev
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, Coding Scheme None
MAC: 30:ae:a4:14:d3:ac
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 983040 bytes to 66133...
Wrote 983040 bytes (66133 compressed) at 0x00110000 in 9.2 seconds (effective 852.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

>> make monitor
....
....
....
I (4208) example_connect: IPv4 address: 192.168.43.139
I (4208) example_connect: IPv6 address: fe80:0000:0000:0000:32ae:a4ff:fe14:d3ac
I (4218) example: Initializing SPIFFS
I (4318) example: Partition size: total: 896321, used: 67268
I (4318) file_server: Starting HTTP Server
```

Now if I simply open `http://192.168.43.139` on a web browser, the monitor log updates as follows:

```
I (72138) file_server: Found file : README.md (469 bytes)
I (72148) file_server: Found file : CODE_OF_CONDUCT.md (689 bytes)
I (72148) file_server: Found file : test.html (1082 bytes)
I (72158) file_server: Found file : styles/style.css (495 bytes)
I (72168) file_server: Found file : LICENSE (6555 bytes)
I (72278) file_server: Found file : images/firefox-icon.png (55480 bytes)
```

arnold-b, 2019-04-25:
thanks. Can you also post your spiffs.bin then i directly load it

anurag-kar, 2019-04-25:
@arnold-b Here you go. Unzip and then write to flash

[spiffs.bin.zip](https://github.com/espressif/esp-idf/files/3118323/spiffs.bin.zip)

arnold-b, 2019-04-25:
[image]
 I used your file with esptool.py --port COM6 write_flash 0x110000 spiffs.bin

and still the same problem

anurag-kar, 2019-04-25:
@arnold-b Strange. Make sure you have the correct partition table and size. If possible do a fresh build of https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/file_serving

arnold-b, 2019-04-25:
Just now i´m compiling a fresh sample

arnold-b, 2019-04-25:
Hello, 

i tried a fesh build. It worked. 
Then i made a fullclean of my projekt. and suddenly i can not even go in menuconfig anymore. Something went wrong on my build.

anurag-kar, 2019-04-25:
@arnold-b Please ensure you are using the latest IDF master and all your submodules are updated

arnold-b, 2019-04-25:
I downloaded the master one week agao. 

now i tried my spiffs.image with new example and it works. But my sample i can not compile anymore.

the main diffrence between your sample and my source is that i changed to c++ compiler (.cpp)

arnold-b, 2019-04-25:
Hello again, 

sorry that i waste your time. 
After i got my application running again, the file_server is working. I only can guess, that something was wrong on the SDKConfig. because thats the file i copied from the new fileserver projekt.

anurag-kar, 2019-04-25:
@arnold-b I'm glad it worked! No issues

arnold-b, 2019-04-25:
On Arduino i had the Problem that the esp32 ignored file requests after around 10 files without any error message. Your one works also on this.  10 files are easily reached. (CSS, Java, html and so on)

arnold-b, 2019-04-25:
Hello, 

thanks again. may you want to use this for your sample. I didnt test "zip" i only tested "gz"

static esp_err_t set_content_type_from_file(httpd_req_t *req,char * turi)
{
	
	std::string cmd = "";

ESP_LOGI(TAG_FILESERVER, "Set TYPE ----------------------------------");
	
	uint8_t  zipext = 0;
    char * tpath;
    char * tpath2;
    char * resp_type=0;

    esp_err_t error;
    
    // Temporaty because didnt find why sometimes ? on the end
	if (turi[strlen(turi) - 1] == '?') // Remove if last char == ? Workaround 
		{ 
			ESP_LOGI(TAG_FILESERVER, "'?' found at position .\n");
			turi[strlen(turi) - 1] = '\0';
		
		}
    tpath = strdup(turi);
    tpath2 = tpath+strlen(tpath);

    if (IS_FILE_EXT(tpath, ".gz")) {
        tpath2 -= 3;
        zipext = 1;
        resp_type =  (char*) "application/gzip";
        *tpath2 = '\0' ;
    } 
    if (IS_FILE_EXT(tpath, ".zip")) {
        tpath2 -= 4;
        zipext = 2;
        *tpath2 = '\0' ;
        resp_type = (char*) "application/zip";
    } 

    ESP_LOGI(TAG_FILESERVER, "Compress found %s--%d\n",tpath,zipext);

    if (IS_FILE_EXT(tpath, ".pdf")) {
        resp_type = (char*) "application/x-pdf";
    } else if (IS_FILE_EXT(tpath, ".html")) {
        resp_type = (char*) "text/html";
	} else if (IS_FILE_EXT(tpath, ".htm")) {
		resp_type = (char*) "text/html";
	} else if (IS_FILE_EXT(tpath, ".css")) {
		resp_type = (char*) "text/css";
	} else if (IS_FILE_EXT(tpath, ".js")) {
		resp_type = (char*) "application/javascript";
	} else if (IS_FILE_EXT(tpath, ".jpeg")) {
        resp_type = (char*) "image/jpeg";
    } else if (IS_FILE_EXT(tpath, ".png")) {
        resp_type = (char*) "image/png";        
    } else if (IS_FILE_EXT(tpath, ".jpg")) {
        resp_type = (char*) "image/jpeg";        
    } else if (IS_FILE_EXT(tpath, ".ico")) {
        resp_type = (char*) "image/x-icon";
    } else if (IS_FILE_EXT(tpath, ".xml")) {
        resp_type = (char*) "text/xml";
    }

    ESP_LOGI(TAG_FILESERVER, "Resp_type =`%s` \n",resp_type);
    /* This is a limited set only */
    /* For any other type always set as plain text */
    if (resp_type ==0)
    {
        if (zipext==1)
        {
            resp_type =  (char*) "application/gzip";
        } else if (zipext==2)
        {
            resp_type =  (char*) "application/zip";
        }
        else
        {   
            
            resp_type = (char*) "text/html";
        }

        
    }
    
    error= httpd_resp_set_type(req,resp_type ); // Must be set before content-Encoding

    if (zipext == 1) {
        httpd_resp_set_hdr(req, "Content-Encoding", "gzip");
    } else if (zipext==2) {
        httpd_resp_set_hdr(req, "Content-Encoding", "compress");
    }
    return error;
	
}

arnold-b, 2019-04-26:
Hello, now i found the problem. 

If i change CONFIG_SPIFFS_OBJ_NAME_LEN=32 to 64 then it don´t work anymore

igrr, 2019-04-26:
@arnold-b When using `mkspiffs`, you need to pay attention to SPIFFS configuration, it should be the same between the IDF and in mkspiffs. When you change CONFIG_SPIFFS_OBJ_NAME_LEN in IDF, you need to change the corresponding option in mkspiffs *and recompile it*. Please see https://github.com/igrr/mkspiffs#spiffs-configuration for details.

Alternatively, if you are using the master branch for development, you may use the new Python-based SPIFFS partition generator in IDF. It integrates with the IDF build system and automatically uses correct SPIFFS settings from sdkconfig. See the example here: https://github.com/espressif/esp-idf/tree/master/examples/build_system/spiffsgen

arnold-b, 2019-04-26:
Thank you igrr, 

i will try now

arnold-b, 2019-04-26:
Hello, 

What i did: 
change CONFIG_SPIFFS_OBJ_NAME_LEN=64
change Makefile => $(eval $(call spiffs_create_partition_image,storage,spiffs_image,FLASH_IN_PROJECT))
add to main/CMakeList.txt => spiffs_create_partition_image(storage ../HTML FLASH_IN_PROJECT)
idf.py flash
try to access

If this is right, 
i need to check whats wrong. because the files are downloaded, but Chrome is still loading.

I (296190) file_server: Sending file : /spiffs/main.html  (1422 bytes)...
E (296200) file_server: File read 1422
E (296210) file_server: File read 0
I (296210) file_server: File sending complete
I (296230) file_server: /style.css

I (296240) file_server: Set TYPE ----------------------------------
I (296240) file_server: Compress found /spiffs/style.css--0

I (296240) file_server: Resp_type =`text/css` 

I (296250) file_server: Sending file : /spiffs/style.css  (4712 bytes)...
E (296250) file_server: File read 4712
E (296250) file_server: File read 0
I (296250) file_server: File sending complete

arnold-b, 2019-04-26:
Hello @iggr, 
https://github.com/espressif/esp-idf/tree/master/examples/build_system/spiffsgen
is working. 

My current problem is diffrent now. 
ESP32 is losing requests
I will check and may make a new Case

thanks

closed 2019-04-26 by arnold-b