espressif/esp-idf #3366: Can't use WPA2-enterprise with EAP-TLS without identity

anthodury, 2019-04-25:
## Environment

- Development Kit: esp32 devkit v1
- Module or chip used: ESP32-WROOM-32
- IDF version: v3.3-beta2-2-gebdcbe8c6
- Build System: Make
- Compiler version: a1.22.0-80-g6c4433a
- Operating System: Linux VM on Windows host
- Power Supply: USB

## Problem Description

Hello,

I would like to authenticate to a WPA2-enterprise network using only certificates.
So I followed this example: https://github.com/espressif/esp-idf/tree/master/examples/wifi/wpa2_enterprise.
Then I set my SSID and the EAP mode 0 (For TLS) in menuconfig and changed the certificates file.
As I only want to use certificate and the README file states:

> Set identity of phase 1 which is optional

I commented the line "`ESP_ERROR_CHECK( esp_wifi_sta_wpa2_ent_set_identity((uint8_t *)EXAMPLE_EAP_ID, strlen(EXAMPLE_EAP_ID)) );`" 

The problem is that I get the error "wpa: EAP: Build Identity Resp-> identity was not available" if I don't provide this identity.

### Steps to repropduce

1. Use wpa2_enterprise example
2. Set SSID and EAP mode to 0 using menuconfig
3. Comment the "set identity" call

### Code to reproduce this issue

```cpp
// Comment set identity call
//ESP_ERROR_CHECK( esp_wifi_sta_wpa2_ent_set_identity((uint8_t *)EXAMPLE_EAP_ID, strlen(EXAMPLE_EAP_ID)) );
```

## Debug Logs

 (322) wpa: WPA2 ENTERPRISE VERSION: [v2.0] enable

I (322) example: esp_wifi_start()
I (422) phy: phy_version: 4100, 6fa5e27, Jan 25 2019, 17:02:06, 0, 0
I (422) wifi: mode : sta (cc:50:e3:99:64:94)
I (542) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (1522) wifi: state: init -> auth (b0)
I (1532) wifi: state: auth -> assoc (0)
I (1532) wifi: state: assoc -> run (10)
I (1532) wpa: wpa2_task prio:2, stack:6656

**E (1542) wpa: EAP: Build Identity Resp-> identity was not available**

E (1542) wpa: Response build fail, return.
**E (1542) wpa: EAP: Build Identity Resp-> identity was not available**

E (1552) wpa: Response build fail, return.
I (4422) example: ~~~~~~~~~~~
I (4422) example: IP:0.0.0.0
I (4422) example: MASK:0.0.0.0
I (4422) example: GW:0.0.0.0
I (4422) example: ~~~~~~~~~~~
I (6422) example: ~~~~~~~~~~~
I (6422) example: IP:0.0.0.0
I (6422) example: MASK:0.0.0.0
I (6422) example: GW:0.0.0.0
I (6422) example: ~~~~~~~~~~~

```

negativekelvin, 2019-04-26:
Did you try setting it to a null string?

negativekelvin, 2019-04-26:
You can try setting it to "anonymous".  I think technically it should default to null string or allow setting to null string but it doesn't.

anthodury, 2019-05-02:
Thank you for answer. Unfortunately it didn't work but I don't know if it's due to the ESP32 and the software or the certificates or the infrastructure so I will have to test with other devices like desktop computers. The issue can be close for now, I will come back later if I get some progress.

closed 2019-05-02 by anthodury

negativekelvin, 2019-05-03:
It may depend on how your server is configured and what your inner identity (username) is set to.  For example if your username is person@domain.com then your outer identity may have to be something 
like user@domain.com or anon@domain.com so that the server can process the realm (domain.com)