espressif/esp-idf #3368: Stack smashing also when using make

X-Ryl669, 2019-04-25:
This is related to #2990, but I'm using make (and not cmake).
I've just upgraded my esp-idf to commit a20d02b7f196c407bc9f39b781e31a0a4f665968

Got this when running the (previously working) application:
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6548
load:0x40078000,len:10648
load:0x40080400,len:6544
entry 0x40080738
I (29) boot: ESP-IDF v3.3-beta1-344-g548d1ac 2nd stage bootloader
I (29) boot: compile time 22:15:01
I (29) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (66) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (73) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (81) boot:  3 factory          factory app      00 00 00010000 000f0000
I (88) boot:  4 ota_0            OTA app          00 10 00100000 00100000
I (95) boot:  5 ota_1            OTA app          00 11 00200000 00100000
I (103) boot:  6 storage          Unknown data     01 82 00300000 00100000
I (111) boot: End of partition table
I (115) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x2bf2c (180012) map
I (186) esp_image: segment 1: paddr=0x0012bf54 vaddr=0x3ffbdb60 size=0x03628 ( 13864) load
I (192) esp_image: segment 2: paddr=0x0012f584 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/cyril/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (193) esp_image: segment 3: paddr=0x0012f98c vaddr=0x40080400 size=0x00684 (  1668) load
I (203) esp_image: segment 4: paddr=0x00130018 vaddr=0x400d0018 size=0x8c024 (573476) map
0x400d0018: _stext at ??:?

I (409) esp_image: segment 5: paddr=0x001bc044 vaddr=0x40080a84 size=0x14380 ( 82816) load
I (454) boot: Loaded app from partition at offset 0x100000
I (454) boot: Disabling RNG early entropy source...
I (455) cpu_start: Pro cpu up.
I (459) cpu_start: Application information:
I (463) cpu_start: Project name:     nvse
I (468) cpu_start: App version:      1a96271-dirty
I (474) cpu_start: Compile time:     Apr 25 2019 22:47:52
I (480) cpu_start: ELF file SHA256:  cf4f604ee36aba72...
I (486) cpu_start: ESP-IDF:          v4.0-dev-408-g643618b
I (492) cpu_start: Starting app cpu, entry point is 0x40081160
0x40081160: call_start_cpu1 at /Users/cyril/esp/esp-idf/components/esp32/cpu_start.c:270

I (484) cpu_start: App cpu up.
I (502) heap_init: Initializing. RAM available for dynamic allocation:
I (510) heap_init: At 3FFAE6E0 len 0000F480 (61 KiB): DRAM
I (515) heap_init: At 3FFD8C88 len 00007378 (28 KiB): DRAM
I (522) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (528) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (535) heap_init: At 40094E04 len 0000B1FC (44 KiB): IRAM
I (541) cpu_start: Pro cpu start user code

Stack smashing protect failure!

abort() was called at PC 0x4014a1b8 on core 0
0x4014a1b8: __stack_chk_fail at /Users/cyril/esp/esp-idf/components/esp_common/src/stack_check.c:36

ELF file SHA256: cf4f604ee36aba7288f5b20f2503ad0e07b303b6f286adb1a21cfa0517e9731d

Backtrace: 0x4008e80a:0x3ffe3b50 0x4008eb68:0x3ffe3b80 0x4014a1b8:0x3ffe3bb0 0x4014a195:0x3ffe3be0 0x400810a0:0x3ffe3c10 0x400813b7:0x3ffe3c40 0x400791b6:0x3ffe3c80 0x40079659:0x3ffe3cc0 0x400807e6:0x3ffe3df0 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20
0x4008e80a: invoke_abort at /Users/cyril/esp/esp-idf/components/esp32/panic.c:721

0x4008eb68: abort at /Users/cyril/esp/esp-idf/components/esp32/panic.c:721

0x4014a1b8: __stack_chk_fail at /Users/cyril/esp/esp-idf/components/esp_common/src/stack_check.c:36

0x4014a195: __esp_stack_guard_setup at /Users/cyril/esp/esp-idf/components/esp_common/src/stack_check.c:31

0x400810a0: do_global_ctors at /Users/cyril/esp/esp-idf/components/esp32/cpu_start.c:478
 (inlined by) start_cpu0_default at /Users/cyril/esp/esp-idf/components/esp32/cpu_start.c:382

0x400813b7: call_start_cpu0 at /Users/cyril/esp/esp-idf/components/esp32/cpu_start.c:258
```

X-Ryl669, 2019-04-25:
For information, I've removed the cpu_start.o and stack_check.o to see what where the compilation flags and go this:
```
$ make V=1
[...]
xtensa-esp32-elf-gcc -std=gnu99 -Os -ggdb -Wno-frame-address -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wwrite-strings -Wno-old-style-declaration  -DESP_PLATFORM -D IDF_VER=\"v4.0-dev-408-g643618b\" -MMD -MP   -D_GNU_SOURCE -DGCC_NOT_5_2_0=0 -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -I /Users/cyril/esp/esp-idf/components/esp32/include -I /Users/cyril/esp/esp-idf/components/freertos/include -I /Users/cyril/esp/esp-idf/components/esp_common/include -I /Users/cyril/esp/esp-idf/components/efuse/esp32/include -I /Users/cyril/esp/esp-idf/components/efuse/include -I /Users/cyril/esp/esp-idf/components/esp_rom/include -I /Users/cyril/esp/esp-idf/components/esp_wifi/include -I /Users/cyril/esp/esp-idf/components/soc/esp32/include -I /Users/cyril/esp/esp-idf/components/soc/include -I /Users/cyril/esp/esp-idf/components/heap/include -I /Users/cyril/esp/esp-idf/components/newlib/platform_include -I /Users/cyril/esp/esp-idf/components/newlib/include -I /Users/cyril/esp/esp-idf/components/nvs_flash/include -I /Users/cyril/esp/esp-idf/components/spi_flash/include -I /Users/cyril/esp/esp-idf/components/log/include -I /Users/cyril/esp/esp-idf/components/tcpip_adapter/include -I /Users/cyril/esp/nvse_app/main/include -I /Users/cyril/esp/esp-idf/components/lwip/include/apps -I /Users/cyril/esp/esp-idf/components/lwip/include/apps/sntp -I /Users/cyril/esp/esp-idf/components/lwip/lwip/src/include -I /Users/cyril/esp/esp-idf/components/lwip/port/esp32/include -I /Users/cyril/esp/esp-idf/components/lwip/port/esp32/include/arch -I /Users/cyril/esp/esp-idf/components/driver/include -I /Users/cyril/esp/esp-idf/components/bt/include -I /Users/cyril/esp/esp-idf/components/bt/bluedroid/api/include/api -I /Users/cyril/esp/esp-idf/components/vfs/include -I /Users/cyril/esp/esp-idf/components/nghttp/port/include -I /Users/cyril/esp/esp-idf/components/nghttp/nghttp2/lib/includes -I /Users/cyril/esp/esp-idf/components/wpa_supplicant/include -I /Users/cyril/esp/esp-idf/components/wpa_supplicant/port/include -I /Users/cyril/esp/esp-idf/components/smartconfig_ack/include -I /Users/cyril/esp/esp-idf/components/app_trace/include -I /Users/cyril/esp/esp-idf/components/mbedtls/port/include -I /Users/cyril/esp/esp-idf/components/mbedtls/mbedtls/include -I /Users/cyril/esp/nvse_app/components/libesphttpd/core -I /Users/cyril/esp/nvse_app/components/libesphttpd/espfs -I /Users/cyril/esp/nvse_app/components/libesphttpd/util -I /Users/cyril/esp/nvse_app/components/libesphttpd/include -I /Users/cyril/esp/nvse_app/components/libesphttpd/lib/heatshrink -I /Users/cyril/esp/esp-idf/components/esp_http_client/include -I /Users/cyril/esp/esp-idf/components/tcp_transport/include -I /Users/cyril/esp/esp-idf/components/esp-tls -I /Users/cyril/esp/esp-idf/components/pthread/include -I /Users/cyril/esp/esp-idf/components/esp_ringbuf/include -I /Users/cyril/esp/esp-idf/components/bootloader_support/include -I /Users/cyril/esp/esp-idf/components/app_update/include -I /Users/cyril/esp/esp-idf/components/mdns/include -I /Users/cyril/esp/esp-idf/components/esp_event/include -I /Users/cyril/esp/esp-idf/components/xtensa/include -I /Users/cyril/esp/esp-idf/components/xtensa/esp32/include -I /Users/cyril/esp/nvse_app/build/include  -I . -c /Users/cyril/esp/esp-idf/components/esp32/cpu_start.c -o cpu_start.o
[...]
```

X-Ryl669, 2019-04-25:
Ok, after looking at `components/esp32/component.mk`, got this:
```
cpu_start.o: CFLAGS := $(filter-out -fstack-protector%, $(CFLAGS))
```
but the command line has `-fstack-protector-all` and not `-fstack-protector` (maybe it's due to the stack protection level (maximum) I've set in menuconfig)
So, I guess the line in the component.mk file should read instead:
```
cpu_start.o: CFLAGS += -fno-stack-protector
```
And it works.

mahavirj, 2019-06-20:
@X-Ryl669 This should have been fixed by https://github.com/espressif/esp-idf/commit/e701f96bf5520492aaabcc3cdab692208497fe89, can you please help to check once?

X-Ryl669, 2019-06-20:
I will.

mahavirj, 2019-07-02:
@X-Ryl669 This fix is tested internally so I will go ahead and close the issue. Please feel free to update otherwise.

closed 2019-07-02 by mahavirj