espressif/esp-idf #3369: How can I achieve effective volume control with esp32 i2S BT sink ?

AymericBr, 2019-04-26:
Hello,

I'm working from the BT A2DP sink example with a PCM5102 DAC (via 12S) trying to achieve volume control on the DAC analog output.
For the moment, the volume change "instructions" from the source is sent and received via AVCRP but has no influence on the volume on output.
How can I manage to do so ?

Best regards,
A

blueMoodBHD, 2019-04-28:
Hi AymericBr,

Which volume change "instructions" did you use?  If you used "SET_ABSOLUTE_VOLUME", sink will received it and should change the volume locally.

Best regards,
B

k-a-z-u, 2019-05-13:
My question is slightly different, but should be related, and might help the OP.
I also use the A2DP sink with PCM5102A over I2S
If i connect with Linux (bluez) or Android, i >can< change the Volume (on the PC/Smartphone).
However, this Volume change seems to be handled somewhere internally within the IDF using integer arithmetic:
If i turn the volume down, within the int16_t data-callback I get samples using only a fraction of 16 bits.
While this allows for volume control, audio quality after the DAC suffers a lot.
I think, in the best case, all 16 bits should be used, and the PCM5102 is followed by some sort of digital potentiometer which is configured to match the requested volume?
However, in my case there is no "volume-changed" event at all, when i change the volume on my smartphone...

drewandre, 2020-04-01:
I'm seeing the same behavior on v4.0. Were you able to solve this?

closed 2022-07-11 by ginkgm