espressif/esp-idf #3371: ESP is loses html requests

arnold-b, 2019-04-26:
Hello, 

If there are many File requests over HTML, ESP is losing the requests. 
What can i do to find the problem ?

I´m using 
- ESP-IDF Master
- example https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/file_serving
 but i think its not a problem of the sample.

Development: Visual Studio Code

[image]
[image]

[image]

I (8572) tcpip_adapter: sta ip: 192.168.178.111, mask: 255.255.255.0, gw: 192.168.178.1
I (8572) TEST_WIFI: SYSTEM_EVENT_STA_GOT_IP
I (8572) TEST_WIFI: Got IP: '192.168.178.111'
I (8572) [SPIFFS example]: Initializing SNTP
I (8582) [SPIFFS example]: Waiting for system time to be set... (1/10)
I (9592) [SPIFFS example]: System time is set.
I (20872) file_server: /sy.html

I (21052) file_server: Set TYPE ----------------------------------
I (21052) file_server: Compress found /spiffs/sy.html--0

I (21052) file_server: Resp_type =`text/html` 

I (21052) file_server: Sending file : /spiffs/sy.html  (3496 bytes)...
E (21062) file_server: File read 3496
E (21072) file_server: File read 0
I (21072) file_server: File sending complete
I (21102) file_server: /style.css

I (21332) file_server: Set TYPE ----------------------------------
I (21332) file_server: Compress found /spiffs/style.css--0

I (21342) file_server: Resp_type =`text/css` 

I (21342) file_server: Sending file : /spiffs/style.css  (4712 bytes)...
E (21352) file_server: File read 4712
E (21362) file_server: File read 0
I (21362) file_server: File sending complete
I (21372) file_server: /css/bootstrap.css

I (21762) file_server: Set TYPE ----------------------------------
I (21762) file_server: Compress found /spiffs/css/bootstrap.css--1

I (21762) file_server: Resp_type =`text/css` 

I (21772) file_server: Sending file : /spiffs/css/bootstrap.css.gz  (15704 bytes)...
E (21782) file_server: File read 8192
E (21842) file_server: File read 7512
E (21842) file_server: File read 0
I (21842) file_server: File sending complete
I (21852) file_server: /css/menu.css

I (21892) file_server: Set TYPE ----------------------------------
I (21892) file_server: Compress found /spiffs/css/menu.css--0

I (21892) file_server: Resp_type =`text/css` 

I (21892) file_server: Sending file : /spiffs/css/menu.css  (1384 bytes)...
E (21902) file_server: File read 1384
E (21912) file_server: File read 0
I (21912) file_server: File sending complete
I (21922) file_server: /css/tabs.css

[... 224 lines cut ...]
I (28532) file_server: File sending complete
I (28562) file_server: /js/connectdlg.js

I (28712) file_server: Set TYPE ----------------------------------
I (28712) file_server: Compress found /spiffs/js/connectdlg.js--0

I (28712) file_server: Resp_type =`application/javascript` 

I (28722) file_server: Sending file : /spiffs/js/connectdlg.js  (3328 bytes)...
E (28732) file_server: File read 3328
E (28732) file_server: File read 0
I (28742) file_server: File sending complete
I (28752) file_server: /js/restartdlg.js

I (28912) file_server: Set TYPE ----------------------------------
I (28912) file_server: Compress found /spiffs/js/restartdlg.js--0

I (28912) file_server: Resp_type =`application/javascript` 

I (28922) file_server: Sending file : /spiffs/js/restartdlg.js  (1339 bytes)...
E (28932) file_server: File read 1339
E (28932) file_server: File read 0
I (28932) file_server: File sending complete

after this no more request

anurag-kar, 2019-04-26:
@arnold-b  Try increasing `backlog_conn` config option (default 5) in httpd_config_t structure passed to httpd_start()

```
httpd_handle_t server = NULL;
    httpd_config_t config = HTTPD_DEFAULT_CONFIG();
    config.uri_match_fn = httpd_uri_match_wildcard;
    config.backlog_conn = 10;

    ESP_LOGI(TAG, "Starting HTTP Server");
    if (httpd_start(&server, &config) != ESP_OK) {
        ESP_LOGE(TAG, "Failed to start file server!");
        return ESP_FAIL;
    }
```

arnold-b, 2019-04-26:
OK, thanks

arnold-b, 2019-04-26:
I´m just now on the way to change on compression and reduce the files. Its faster to load less files even they are bigger. I will test this in 3 hours

arnold-b, 2019-04-26:
Hello anurag-kar,
i tried 
config.backlog_conn = 10 
and
config.backlog_conn = 20;

both made no change on this. 

How could i debug where the requests are lost

httpd_handle_t server = NULL;
    httpd_config_t config = HTTPD_DEFAULT_CONFIG();

    /* Use the URI wildcard matching function in order to
     * allow the same handler to respond to multiple different
     * target URIs which match the wildcard scheme */
    config.uri_match_fn = httpd_uri_match_wildcard;
    config.backlog_conn = 20;

    ESP_LOGI(TAG_FILESERVER, "Starting HTTP Server");
    if (httpd_start(&server, &config) != ESP_OK) {
        ESP_LOGE(TAG_FILESERVER, "Failed to start file server!");
        return ESP_FAIL;
    }

anurag-kar, 2019-04-26:
From the log it seems that requests are not even reaching the server. Could
you use wireshark to log the traffic between the device and the host?

arnold-b, 2019-04-26:
I checked with Wireshark.

You are right. The requests stoped. After i increased the timeout in IE it worked. Its just to slow.
Now in IE it works. in CHrome i don´t find how to increase the Timeout

closed 2019-04-26 by arnold-b

anurag-kar, 2019-04-29:
It seems you need to improve your client side JS implementation to limit the number of HTTP GET requests that can be sent at a time. Also handle situations when socket creation fails due to timeout, without changing your default browser settings.

arnold-b, 2019-05-02:
Hey, 
thats i know. But for me it was a good testing to get on the limits. And it´s more easy during testing. 
For later it will be just 4 or 5 files per load.