espressif/esp-idf #3372: is mbedtls_aes_crypt_ofb deactivated in esp-idf

chegewara, 2019-04-26:
Hi. I know there is self test function in aes.c, but i just found that `mbedtls_aes_crypt_ofb` cant be used on esp32.
In theory `MBEDTLS_CIPHER_MODE_OFB` is defined here
https://github.com/espressif/mbedtls/blob/19eb57f7f7b2f4312d497ddbcb5f104fc6877c70/include/mbedtls/config.h#L558

but it has to be undefined somewhere else because compiling with `mbedtls_aes_crypt_ofb` is causing error:
```
error: implicit declaration of function 'mbedtls_aes_crypt_ofb' [-Werror=implicit-function-declaration]
```

Here is my test app:
https://github.com/chegewara/esp32-mbedtls-aes-test

ghemal, 2019-04-29:
mbedtls allows to use default config.h or user defined config file

#if !defined(MBEDTLS_CONFIG_FILE)
#include "mbedtls/config.h"
#else
#include MBEDTLS_CONFIG_FILE
#endif

For esp32 default config.h file is overridden by  esp_config.h in esp-idf/components/mbedtls/port/include/mbedtls/esp_config.h. You will have to define MBEDTLS_CIPHER_MODE_OFB in this file

chegewara, 2019-04-30:
Well, i am defining `MBEDTLS_CIPHER_MODE_OFB` and im getting this error:
```
D:/esp32_apps/snippets/test_aes/build/main\libmain.a(main.o):(.literal.ofb+0x4): undefined reference to `mbedtls_aes_crypt_ofb'
D:/esp32_apps/snippets/test_aes/build/main\libmain.a(main.o): In function `ofb':
D:/esp32_apps/snippets/test_aes/main/main.c:90: undefined reference to `mbedtls_aes_crypt_ofb'
D:/esp32_apps/snippets/test_aes/main/main.c:91: undefined reference to `mbedtls_aes_crypt_ofb'
```

ghemal, 2019-05-01:
OFB mode support is not yet added for AES ALT implementation (https://github.com/espressif/esp-idf/blob/master/components/mbedtls/port/esp32/aes.c). Since you have enabled AES hardware accelerator (CONFIG_MBEDTLS_HARDWARE_AES=y) there is no corresponding hardware API in esp-idf/components/mbedtls/port/include/aes_alt.h and that is why you get this error. For the time being you can continue to use software implementation

chegewara, 2019-05-01:
@ghemal Thanks you are trying to help, but i have mbedtls working. The only thing that is not working is ofb encryption, all other types are working. You can check it in my test code, link you can find in OP.

ghemal, 2019-05-01:
@chegewara I have looked at your code. In your sdkconfig CONFIG_MBEDTLS_HARDWARE_AES=y which means alternate implementation (using hardware acceleartor) is being used. I understand mbedtls is working for all modes except ofb. This is expected since other modes except OFB are supported through alternate implementation. All the alternate implementation (except OFB) using hardware acceleration are available in esp-idf/components/mbedtls/port/esp32/aes.c 

As you might be already aware the software implementation of mbedtls_aes_crypt_ofb() is available in esp-idf/components/mbedtls/mbedtls/library/aes.c but this is not implemented in above file

chegewara, 2019-05-01:
this is exactly what ive been asking for.
Is ofb implemented in esp-idf or not. 

Thanks @ghemal

projectgus, 2019-05-02:
This is a bug, and we should fix it.

closed 2019-07-16 by igrr