espressif/esp-idf #3373: httpd apis : unrealistic, no wildcard or root of directory support

dannybackx, 2019-04-27:
## Environment

- Development Kit:   N/A (esp32 mh-et live minikit)
- Kit version N/A
- Module or chip used:  ESP32
- IDF version : v3.2
- Build System:         Make
- Compiler version : 5.2.0
- Operating System:     Linux
- Power Supply:         external

## Problem Description

The webserver APIs only support queries that you specify exactly. When you run with
```
  u_hello.uri = "/hello";
  httpd_register_uri_handler(server, &u_hello);

```
the server will only respond kindly to exactly this query.
Queries for e.g. "/hello/there" will fail.
Several attempts to build a handler based on a wildcard such as
```
  u_hello.uri = "/hello/*";
  httpd_register_uri_handler(server, &u_hello);

```
failed. No catch-all api appears to exist, none of the support channels appears to be able to help out.

### Code to reproduce this issue
```
WebServer::WebServer() {
  esp_err_t err;
  httpd_config_t cfg = HTTPD_DEFAULT_CONFIG();
  cfg.server_port = config->getWebServerPort();

  ESP_LOGD(webserver_tag, "Start webserver(%d)", cfg.server_port);

  if ((err = httpd_start(&server, &cfg)) != ESP_OK) {
    ESP_LOGE(webserver_tag, "failed to start %s (%d)", esp_err_to_name(err), err);
    return;
  }

  httpd_uri_t u_hello = {
    "/hello",                   // URI handled
    HTTP_GET,                   // HTTP method
    hello_get_handler,          // Handler
    (void *)0                   // User context
  };

  httpd_register_uri_handler(server, &u_hello);
  u_hello.uri = "/hello/*";
  httpd_register_uri_handler(server, &u_hello);
  u_hello.uri = "/hello/%";
  httpd_register_uri_handler(server, &u_hello);

  // This is the initial URI usually queried.
  u_hello.uri = "/";
  u_hello.handler = hello_index_handler;
  httpd_register_uri_handler(server, &u_hello);

  u_hello.uri = "/hello";
  u_hello.method = HTTP_POST;
  u_hello.handler = hello_post_handler;
  httpd_register_uri_handler(server, &u_hello);
  return;
}
```

## Debug Logs
```
W (31912157) httpd_uri: httpd_uri: URI '/hello/there' not found
W (31912157) httpd_txrx: httpd_resp_send_err: 404 Not Found - This URI doesn't exist

```

igrr, 2019-04-27:
You can set uri_match_fn member of httpd_config structure to httpd_uri_match_wildcard: https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/protocols/esp_http_server.html#_CPPv4N12httpd_config12uri_match_fnE. File serving example demonstrates this: https://github.com/espressif/esp-idf/blob/master/examples/protocols/http_server/file_serving/main/file_server.c

dannybackx, 2019-04-27:
The member uri_match_fn is not in 3.2 .

anurag-kar, 2019-04-27:
@dannybackx There is workaround to achieving this with 3.2 : https://esp32.com/viewtopic.php?p=29992#p29992

In brief, your requests would look like `/hello?there`, the variable part of the path sent as a query string and retrieved using `httpd_req_get_url_query_str()`

anurag-kar, 2019-04-29:
I believe your query has been answered, so I'll go ahead and close this issue. Please feel free to reopen if the need arises.

closed 2019-04-29 by anurag-kar