espressif/esp-idf #3376: Problem with EMBED binary files from subfolder

vtamas00, 2019-04-27:
- Development Kit:     ESP32-DevKitC
- Kit version DevKitC: v1
- Module or chip used:  ESP-WROOM-32
- IDF version : v4.0-dev-447-g689032650
- Build System:         Make
- Compiler version: 1.22.0-80-g6c4433a
- Operating System:    Linux
- Power Supply:         USB

I would like to embed binary files to my project. It was working without problems in idf version v3.3-beta1-328-gabea9e4c0, but after I updated latest version I got (.literal.http_serve+0x4): undefined reference to `_binary_root_html_start' error message.

After some investigation I found the file embedding is working if the bin file located in the main folder, but if I move it to a subfolder the build system ignore it.

I tried to add the subfolder to the build with COMPONENT_PRIV_INCLUDEDIRS, COMPONENT_ADD_LINKER_DEPS, COMPONENT_SRCDIRS and COMPONENT_ADD_INCLUDEDIRS in the main/component.mk file without success.

negativekelvin, 2019-04-27:
https://github.com/espressif/esp-idf/blob/0d7f2d77c2eee51dce3fa7c8430a641c65cd9719/examples/protocols/https_server/main/component.mk#L6

vtamas00, 2019-04-28:
OK, it is a working workaround, but not elegant solution.

anurag-kar, 2019-04-29:
@vtamas00 Using COMPONENT_EMBED_TXTFILES (for files containing strings) or COMPONENT_EMBED_FILES (for binary files) is the recommended solution.

See https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html#embedding-binary-data

Since this is a non issue I'm going to close it. Please feel free to reopen if necessary

closed 2019-04-29 by anurag-kar

negativekelvin, 2019-04-29:
@anurag-kar one suggestion would be if the entry is a directory, embed all the files inside

anurag-kar, 2019-04-29:
@negativekelvin Yes that may be a feature worth having, though a directory containing a mix of text and binary files, may still require specifying them separately or treating the text files as binary.

AntoinedeChassey, 2019-11-28:
How can one include many files in `component.mk`? What is the convention?

```
COMPONENT_EMBED_TXTFILES :=  ${PROJECT_PATH}/file1
COMPONENT_EMBED_TXTFILES :=  ${PROJECT_PATH}/file2
 ... ???
```

Thanks!

negativekelvin, 2019-11-29:
https://github.com/espressif/esp-idf/search?q=COMPONENT_EMBED_TXTFILES&unscoped_q=COMPONENT_EMBED_TXTFILES