espressif/esp-idf #3377: "idf.py fullclean" is deleting contents of mbedtls/include/mbedtls folder.

william-ferguson-au, 2019-04-27:
## Environment
- IDF version 
   v3.3-beta2
   mbedtls-2.13.1-1-g19eb57f7

- Build System:         CMake
- Compiler version   1.22.0-80-g6c4433a
- Operating System:     Windows

## Problem Description
Executing "idf.py fullclean" not only removes the contents of the project/build folder. It also removes the contents of the esp-idf mbedtls\include\mbedtls folder.

### Expected Behavior
"idf.py fullclean" should only affect the project build folder.

### Actual Behavior
"idf.py fullclean" also removes the contents of the esp-idf mbedtls\include\mbedtls folder.

### Steps to reproduce
1. Clone esp-idf recursive. 
2. Confirm that the contents of \esp-idf\components\mbedtls\mbedtls\include\mbedtls are there.
3. From project folder execute 'idf.py fullclean'
4. Confirm that the contents of \esp-idf\components\mbedtls\mbedtls\include\mbedtls have been removed.

dobairoland, 2019-04-29:
Hi @william-ferguson-au. Unfortunately, I cannot reproduce your issue. I would like to ask you to give me some additional information.

I tried `master` and `release/v3.3` branches in MSYS and I have 78 regular files in that directory before and after doing `fullclean` as well.

Are you using MSYS or Windows command line?

Is it possible to reproduce your issue on your side also with an example project of IDF, e.g. `get-started/hello_world`?

What is the output of `git status` of IDF before and after `fullclean`?

william-ferguson-au, 2019-04-29:
Using Windows command line.
I can reproduce with ```get-started/hello_world```.

Steps (from hello_world folder)
1. idf.py build
2. idf.py fullclean

NB idf.py on its own isn't enough, it needs to have built first.

git status (before fullclean)
```
On branch tag_v3.3-beta2
nothing to commit, working tree clean
```

git status (after fullclean)
```On branch tag_v3.3-beta2
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)

        modified:   mbedtls (modified content)
```

dobairoland, 2019-04-30:
I reproduced the issue. It can only happen in Windows console line with Python 2. Thank you @william-ferguson-au for reporting the issue.

The fix will be merged ASAP into master and release branches as well.

william-ferguson-au, 2019-04-30:
Great, thanks for the confirm @dobairoland

closed 2019-05-05 by igrr

NeloHuerta, 2020-03-12:
I've experienced the same behavior using Python 3.8. I think the problem is related with the fact that idf.py build creates a link between /build folder and the mbedtls folder inside /esp-idf folder. 
idf.py fullclean seems to erase recursively all folders inside /build