espressif/esp-idf #3378: Unable to build bootloader with idf.py and minGW makefiles

william-ferguson-au, 2019-04-27:
## Environment

- IDF version 
    // v3.3-beta2

- Build System:         CMake
- Compiler version 
    // (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0

- Operating System:     Windows

## Problem Description

Executing ```idf.py -G "MinGW Makefiles" build``` (from the command line) causes everything to build except for the bootloader. Building using ninja works perfectly, but I want to be able to build from within CLion so I need to be able to use MinGW.

### Expected Behavior

All binaries get built, including bootloader.

### Actual Behavior

All binaries get built, except for bootloader.

### Steps to reproduce

1. 'idf.py -G "MinGW Makefiles" build'

## Build Log

```
[100%] Generating ../bootloader.bin
E:\Dev\Espressif\esp-idf\components\esptool_py\esptool\esptool.py --chip esp32 elf2image --flash_mode dio --flash_freq 40m --flash_size 4MB --elf-sha256-offset 0xb0 -o E:/Source/esp32/wt_trinity/build/bootloader/bootloader.bin bootloader.elf
process_begin: CreateProcess(NULL, env python E:\Dev\Espressif\esp-idf\components\esptool_py\esptool\esptool.py --chip esp32 elf2image --flash_mode dio --flash_freq 40m --flash_size 4MB --elf-sha256-offset 0xb0 -o E:/Source/esp32/wt_trinity/build/bootloader/bootloader.bin bootloader.elf, ...) failed.
make (e=2): The system cannot find the file specified.
esp-idf\CMakeFiles\bootloader.dir\build.make:59: recipe for target 'bootloader.bin' failed
mingw32-make[5]: *** [bootloader.bin] Error 2
mingw32-make[5]: Leaving directory 'E:/Source/esp32/wt_trinity/build/bootloader'
CMakeFiles\Makefile2:500: recipe for target 'esp-idf/CMakeFiles/bootloader.dir/all' failed
mingw32-make[4]: *** [esp-idf/CMakeFiles/bootloader.dir/all] Error 2
mingw32-make[4]: Leaving directory 'E:/Source/esp32/wt_trinity/build/bootloader'
Makefile:82: recipe for target 'all' failed
mingw32-make[3]: *** [all] Error 2
```

## Other items if possible

Contents of build\bootloader folder:
```
 Volume in drive E is Data
 Volume Serial Number is C4B9-301F

 Directory of E:\Source\esp32\wt_trinity\build\bootloader

28/04/2019  09:03 AM    <DIR>          .
28/04/2019  09:03 AM    <DIR>          ..
28/04/2019  09:03 AM           185,020 bootloader.elf
28/04/2019  09:03 AM           293,239 bootloader.map
28/04/2019  09:02 AM            16,745 CMakeCache.txt
28/04/2019  09:20 AM    <DIR>          CMakeFiles
28/04/2019  09:02 AM             1,661 cmake_install.cmake
28/04/2019  09:03 AM            48,397 compile_commands.json
28/04/2019  09:02 AM             3,289 component_depends.cmake
28/04/2019  09:02 AM    <DIR>          config
28/04/2019  09:03 AM                 0 dummy_main_src.c
28/04/2019  09:02 AM    <DIR>          esp-idf
28/04/2019  09:02 AM               997 flasher_args.json
28/04/2019  09:02 AM                17 flash_app_args
28/04/2019  09:02 AM                91 flash_bootloader_args
28/04/2019  09:02 AM                38 flash_partition_table_args
28/04/2019  09:02 AM               148 flash_project_args
28/04/2019  09:02 AM            22,440 Makefile
28/04/2019  09:02 AM             1,789 project_description.json
              14 File(s)        573,871 bytes
               5 Dir(s)  636,975,407,104 bytes free
```

renzbagaporo, 2019-04-29:
Hi @william-ferguson-au, I've tried building the hello_world example and building bootloader seems to succeed. Does building this example fail or succeed for you?

william-ferguson-au, 2019-04-29:
No. It also fails to build the boot loader for hello_world.
NB this is Windows command line

projectgus, 2019-04-30:
Hi @william-ferguson-au ,

There were some fixes for issues where the Python interpreter was not being correctly picked up by CMake that are not in v3.3-beta2 but they are in v3.3-beta3 (see also #3160, #2936). If you bump your IDF version to v3.3-beta3, does the build succeed?

Thanks,

Angus

renzbagaporo, 2019-04-30:
Ah. I was on beta3. So bumping your IDF version should fix the issue if that is an option @william-ferguson-au.

william-ferguson-au, 2019-04-30:
I don't see any esp-idf v3.3-beta.

Nothing here : https://github.com/espressif/esp-idf/releases

And 
```
git tag --list

...
v3.2
v3.2-beta1
v3.2-beta3
v3.2-dev
v3.2-rc
v3.3-beta1
v3.3-beta2
v3.3-dev
v4.0-dev
```

No v3.3-beta3

Where do I find it?

william-ferguson-au, 2019-05-02:
@projectgus @renzbagaporo am I missing something? 
Where do I find v3.3-beta3?

projectgus, 2019-05-03:
@william-ferguson-au Apologies, we tagged it in our internal repo but the tag didn't get pushed out by the CI system for some reason. Will fix this.

I've just pushed the tag manually so if you fetch from GitHub again, you'll be able to check it our. Or you can checkout the release/v3.3 branch which now has a few fixes on top of v3.3-beta3.

william-ferguson-au, 2019-05-13:
Thanks, have checked out v3.3-beta3 now.
```
E:\Dev\Espressif\esp-idf>git branch
  master
  tag_v3.3-beta2
* tag_v3.3-beta3
```

But when I build ```idf.py build``` it is reporting 
```
-- IDF_VER: v3.3-beta1-694-g6b3da6b18
```

Is that expected?

william-ferguson-au, 2019-05-20:
*bump*

So now that I have checked out v3.3-beta3 I can build the bootloader.

But when I build idf.py build it is reporting

`
-- IDF_VER: v3.3-beta1-694-g6b3da6b18
`
Shouldn't it reported v3.3-beta3 ??

projectgus, 2019-05-21:
Hi @william-ferguson-au,

v3.3-beta3 was accidentally tagged "un-annotated", which means that `git describe` shows the version you have but `git describe --tags` should show `v3.3-beta3`. The IDF build system should pass `--tags` when generating the `IDF_VER` variable value, though.

What do you get if you run the `git describe` and `git describe --tags` commands locally?

william-ferguson-au, 2019-05-21:
Hi @projectgus, I get exactly as you surmised:

```
E:\Dev\Espressif\esp-idf>git describe
v3.3-beta1-694-g6b3da6b18

E:\Dev\Espressif\esp-idf>git describe --tags
v3.3-beta3
```
So just sounds like the build system needs to be updated to pass ```--tags```

closed 2019-05-21 by william-ferguson-au

projectgus, 2019-05-23:
Thanks @william-ferguson-au. On second look CMake wasn't passing `--tags --dirty` so the CMake IDF_VER didn't match the GNU Make one. A fix is in internal review, will be merged ASAP.