espressif/esp-idf #3380: I2S APLL Clock is inaccurate at 44100Hz 16-bit 2ch

redchenjs, 2019-04-28:
## Environment

- Development Kit: Widora AIR v6.0
- Module or chip used: ESP32 Rev.1
- IDF version: v3.2
- Build System: CMake
- Compiler version: 1.22.0-80-g6c4433a
- Operating System: Linux
- Power Supply: USB

## Problem Description

When the i2s APLL clock is set to 44100Hz 16-bit 2ch, the real rate is 43945.238, which is inaccurate and causes  `BT_APPL: Pkt dropped` problem when using A2DP streaming.

### Expected Behavior

Real rate should be 44099.xxx or 44100.xxx

### Actual Behavior

Real rate is 43945.238

### Steps to repropduce

1. Get the `a2dp_sink` example from esp-idf v3.2.
2. Add `.use_apll = 1,` to `i2s_config` to enable APLL.
3. Build the example and flash it to esp32.
4. Connect to esp32 with your phone and play some music.

### Code to reproduce this issue

The `a2dp_sink` example from esp-idf v3.2 

## Debug Logs

```
I (16536) I2S: APLL: Req RATE: 44100, real rate: 43945.238, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11249981.000, SCLK: 1406247.625000, diva: 1, divb: 0
......
I (81856) BT_AV: Audio packet count 3800
I (83316) BT_AV: Audio packet count 3900
I (84766) BT_AV: Audio packet count 4000
I (86226) BT_AV: Audio packet count 4100
I (87686) BT_AV: Audio packet count 4200
W (88266) BT_APPL: Pkt dropped

I (89136) BT_AV: Audio packet count 4300
I (90596) BT_AV: Audio packet count 4400
I (92056) BT_AV: Audio packet count 4500
I (93506) BT_AV: Audio packet count 4600
W (93656) BT_APPL: Pkt dropped

I (94966) BT_AV: Audio packet count 4700
I (96416) BT_AV: Audio packet count 4800
W (97196) BT_APPL: Pkt dropped

I (97876) BT_AV: Audio packet count 4900
I (99336) BT_AV: Audio packet count 5000
I (100786) BT_AV: Audio packet count 5100
W (101896) BT_APPL: Pkt dropped

I (102246) BT_AV: Audio packet count 5200
I (103706) BT_AV: Audio packet count 5300
W (105016) BT_APPL: Pkt dropped
```

negativekelvin, 2019-04-28:
If apll parameters are 
sdm2 = 5
sdm1 = 8
sdm0 = 28
odiv = 0

then fout = 40000000 * (4 + 5 + 8/256 + 28/65536)) = 361267089.84375
apll_freq = 361267089.84375 / (2 * (0 + 2)) = 90316772.4609375

If clkm_num = 8, a = 0, b = 1
clkm_freq = 90316772.4609375 / 8 = 11289596.5576171875

If bck_div_num = 256 / (16 * 2) = 8
bck_freq = 1411199.5697021484375

and sample rate = 1411199.5697021484375 / 32 = 44099.986553192138671875

so yeah the driver is not doing a good job calculating apll parameters.

redchenjs, 2019-04-29:
As #1393 mentioned, after 63ddae5 the apll_predefine values were removed, and before that, the driver did have a hardcoded APLL parameter table and the 44100Hz rate is right, so why the apll_predefine values were removed?

redchenjs, 2019-04-30:
Now my workaround is:
```
        if (rate == 44100) {
            rtc_clk_apll_enable(1, 15, 8, 5, 6);
            ESP_LOGW(TAG, "enable workaround for i2s apll clock at 44100Hz 16-bit 2ch");
        } else {
            i2s_set_sample_rates(I2S_NUM_0, rate);
        }
```
It did help, there is no `BT_APPL: Pkt dropped` problem anymore.
Where the `rtc_clk_apll_enable()` function prototype is
```
void rtc_clk_apll_enable(bool enable, uint32_t sdm0, uint32_t sdm1, uint32_t sdm2, uint32_t o_div)
```
The hardcoded parameter values are from previous 63ddae5 removed `apll_predefine` table:
```
/**	
 * @brief Pre define APLL parameters, save compute time	
 *        | bits_per_sample | rate | sdm0 | sdm1 | sdm2 | odir	
 */	
static const int apll_predefine[][6] = {	
    {16, 11025, 38,  80,  5, 31},	
    {16, 16000, 147, 107, 5, 21},	
    {16, 22050, 130, 152, 5, 15},	
    {16, 32000, 129, 212, 5, 10},	
    {16, 44100, 15,  8,   5, 6},	
    {16, 48000, 136, 212, 5, 6},	
    {16, 96000, 143, 212, 5, 2},	
    {0,  0,     0,   0,   0, 0}	
};
```
@negativekelvin Thanks for your reply! I tried to use the values you calculated, but it seems that they won't work unless the i2s_set_clk() function is modified.

steenou, 2019-05-03:
Have you experienced glitches in the output like I have been experiencing? https://github.com/espressif/esp-idf/issues/3407 I've been seeing zeroes in the SD output which is audible as clicking and cracking

pad52, 2019-05-30:
Same issue here, still not resolved: https://github.com/espressif/esp-idf/issues/2634

ajita02, 2019-06-07:
Try the attached patch. It has solved the inaccurate apll clock rate issue. Kindly share the result obtained.. 
[0001-driver-i2s-fix-apll_clock_rate-for-different-sample-.zip](https://github.com/espressif/esp-idf/files/3264388/0001-driver-i2s-fix-apll_clock_rate-for-different-sample-.zip)

fknrdcls, 2019-06-07:
Just a couple of changes to the original calculation are needed as I already provided in #2634 
https://github.com/espressif/esp-idf/issues/2634#issuecomment-460503515

There is no need for trial and error my patch will get you as close as the PLL can do, which is usually 10-20ppm or something. (+/- the source clock ppm obviously) I have measured hundreds of arbitrary clock rates with a high resolution frequency counter and have been using this in Audio projects for over a year now. 
There is nothing wrong with the APLL it's a good PLL, this original code was erroneous in that it would overshoot the more significant sdm parameters because it would choose the value closer to the target even if it was slightly higher than the target value, forever setting the accuracy at the resolution of that coarser parameter. Choosing the closest value should only be done on the least significant parameter.

I also have a cheap way to measure the input frequency if you're running I2S in slave mode if anyone's interested.

ajita02, 2019-06-07:
> Try the attached patch. It has solved the inaccurate apll clock rate issue. The patch is w.r.t. latest IDF i.e. 4.0. Kindly help to share the result.
> [0001-driver-i2s-fix-apll_clock_rate-for-different-sample-.zip](https://github.com/espressif/esp-idf/files/3264388/0001-driver-i2s-fix-apll_clock_rate-for-different-sample-.zip)

The attached patch give is percentage difference of 0.001. i.e. It will differ only by 1 sample rate.

fknrdcls, 2019-06-07:
Yeah no the APLL can do better than that with this fix:
From: https://github.com/espressif/esp-idf/issues/2634#issuecomment-460503515

> For example:
>
>`I (712) I2S: APLL: Req RATE: 48000, real rate: 47999.992, BITS: 32, CLKM: 1, BCK_M: 8, MCLK: >24575996.000, SCLK: 3071999.500000, diva: 1, divb: 0
>`

MCLK is off by 4 parts in 24.576 Million
0.00001627604167%

So I guess I was wrong it's usually off less than 10 parts per million.

I'll say it again: It's a good proper Audio PLL, and it is much better than esspresif is apparently OK with.

The patch is simple, it was just a brain fart by the original developer, what is disappointing is that it STILL HASN'T BEEN FIXED.

fknrdcls, 2019-06-07:
Also still got that MP3 encoder if anyone's interested. Otherwise I have work to do.
48Khz 16bit Stereo 256kbps 37% CPU overall.

redchenjs, 2019-06-18:
@ajita02 @fknrdcls Hi, I compared the two patches. From the results, they are all very accurate. So it's time to submit them.
@ajita02 's patch:
```
I (524) I2S: APLL: Req RATE: 10675, real rate: 10674.993, BITS: 16, ERROR: 0.000064%
I (537) I2S: APLL: Req RATE: 10675, real rate: 10674.993, BITS: 24, ERROR: 0.000067%
I (552) I2S: APLL: Req RATE: 10675, real rate: 10674.988, BITS: 32, ERROR: 0.000110%

I (568) I2S: APLL: Req RATE: 11025, real rate: 11024.999, BITS: 16, ERROR: 0.000009%
I (583) I2S: APLL: Req RATE: 11025, real rate: 11024.996, BITS: 24, ERROR: 0.000035%
I (599) I2S: APLL: Req RATE: 11025, real rate: 11024.990, BITS: 32, ERROR: 0.000089%

I (614) I2S: APLL: Req RATE: 16000, real rate: 15999.993, BITS: 16, ERROR: 0.000043%
I (630) I2S: APLL: Req RATE: 16000, real rate: 15999.992, BITS: 24, ERROR: 0.000049%
I (645) I2S: APLL: Req RATE: 16000, real rate: 15999.993, BITS: 32, ERROR: 0.000043%

I (661) I2S: APLL: Req RATE: 22050, real rate: 22049.980, BITS: 16, ERROR: 0.000089%
I (676) I2S: APLL: Req RATE: 22050, real rate: 22049.971, BITS: 24, ERROR: 0.000130%
I (692) I2S: APLL: Req RATE: 22050, real rate: 22049.980, BITS: 32, ERROR: 0.000089%

I (707) I2S: APLL: Req RATE: 32000, real rate: 31999.986, BITS: 16, ERROR: 0.000043%
I (722) I2S: APLL: Req RATE: 32000, real rate: 31999.984, BITS: 24, ERROR: 0.000049%
I (738) I2S: APLL: Req RATE: 32000, real rate: 31999.988, BITS: 32, ERROR: 0.000037%

I (753) I2S: APLL: Req RATE: 44100, real rate: 44099.961, BITS: 16, ERROR: 0.000089%
I (769) I2S: APLL: Req RATE: 44100, real rate: 44099.943, BITS: 24, ERROR: 0.000130%
I (784) I2S: APLL: Req RATE: 44100, real rate: 44099.988, BITS: 32, ERROR: 0.000027%

I (800) I2S: APLL: Req RATE: 48000, real rate: 47999.977, BITS: 16, ERROR: 0.000049%
I (815) I2S: APLL: Req RATE: 48000, real rate: 47999.953, BITS: 24, ERROR: 0.000098%
I (831) I2S: APLL: Req RATE: 48000, real rate: 47999.977, BITS: 32, ERROR: 0.000049%

I (846) I2S: APLL: Req RATE: 96000, real rate: 95999.953, BITS: 16, ERROR: 0.000049%
I (861) I2S: APLL: Req RATE: 96000, real rate: 95999.906, BITS: 24, ERROR: 0.000098%
I (877) I2S: APLL: Req RATE: 96000, real rate: 95999.984, BITS: 32, ERROR: 0.000016%
```
@fknrdcls 's patch:
```
I (527) I2S: APLL: Req RATE: 10675, real rate: 10675.000, BITS: 16, ERROR: 0.000000%
I (538) I2S: APLL: Req RATE: 10675, real rate: 10675.003, BITS: 24, ERROR: 0.000030%
I (554) I2S: APLL: Req RATE: 10675, real rate: 10675.006, BITS: 32, ERROR: 0.000055%

I (569) I2S: APLL: Req RATE: 11025, real rate: 11024.991, BITS: 16, ERROR: 0.000080%
I (585) I2S: APLL: Req RATE: 11025, real rate: 11024.992, BITS: 24, ERROR: 0.000071%
I (600) I2S: APLL: Req RATE: 11025, real rate: 11024.991, BITS: 32, ERROR: 0.000080%

I (616) I2S: APLL: Req RATE: 16000, real rate: 15999.986, BITS: 16, ERROR: 0.000085%
I (631) I2S: APLL: Req RATE: 16000, real rate: 15999.987, BITS: 24, ERROR: 0.000081%
I (646) I2S: APLL: Req RATE: 16000, real rate: 16000.003, BITS: 32, ERROR: 0.000018%

I (662) I2S: APLL: Req RATE: 22050, real rate: 22049.982, BITS: 16, ERROR: 0.000080%
I (677) I2S: APLL: Req RATE: 22050, real rate: 22049.984, BITS: 24, ERROR: 0.000071%
I (693) I2S: APLL: Req RATE: 22050, real rate: 22049.994, BITS: 32, ERROR: 0.000027%

I (708) I2S: APLL: Req RATE: 32000, real rate: 32000.006, BITS: 16, ERROR: 0.000018%
I (724) I2S: APLL: Req RATE: 32000, real rate: 31999.974, BITS: 24, ERROR: 0.000081%
I (739) I2S: APLL: Req RATE: 32000, real rate: 32000.006, BITS: 32, ERROR: 0.000018%

I (755) I2S: APLL: Req RATE: 44100, real rate: 44099.988, BITS: 16, ERROR: 0.000027%
I (770) I2S: APLL: Req RATE: 44100, real rate: 44099.969, BITS: 24, ERROR: 0.000071%
I (786) I2S: APLL: Req RATE: 44100, real rate: 44099.988, BITS: 32, ERROR: 0.000027%

I (801) I2S: APLL: Req RATE: 48000, real rate: 47999.961, BITS: 16, ERROR: 0.000081%
I (816) I2S: APLL: Req RATE: 48000, real rate: 48000.016, BITS: 24, ERROR: 0.000033%
I (832) I2S: APLL: Req RATE: 48000, real rate: 47999.992, BITS: 32, ERROR: 0.000016%

I (847) I2S: APLL: Req RATE: 96000, real rate: 95999.984, BITS: 16, ERROR: 0.000016%
I (863) I2S: APLL: Req RATE: 96000, real rate: 96000.031, BITS: 24, ERROR: 0.000033%
I (878) I2S: APLL: Req RATE: 96000, real rate: 95999.984, BITS: 32, ERROR: 0.000016%
```
The unmodified version:
[... 8 lines cut ...]

I (604) I2S: APLL: Req RATE: 16000, real rate: 15999.986, BITS: 16, ERROR: 0.000085%
I (620) I2S: APLL: Req RATE: 16000, real rate: 15999.987, BITS: 24, ERROR: 0.000081%
I (635) I2S: APLL: Req RATE: 16000, real rate: 16000.003, BITS: 32, ERROR: 0.000018%

I (651) I2S: APLL: Req RATE: 22050, real rate: 22049.982, BITS: 16, ERROR: 0.000080%
I (666) I2S: APLL: Req RATE: 22050, real rate: 22049.984, BITS: 24, ERROR: 0.000071%
I (682) I2S: APLL: Req RATE: 22050, real rate: 21972.619, BITS: 32, ERROR: 0.350934%

I (697) I2S: APLL: Req RATE: 32000, real rate: 32000.006, BITS: 16, ERROR: 0.000018%
I (712) I2S: APLL: Req RATE: 32000, real rate: 31999.974, BITS: 24, ERROR: 0.000081%
I (728) I2S: APLL: Req RATE: 32000, real rate: 32000.006, BITS: 32, ERROR: 0.000018%

I (743) I2S: APLL: Req RATE: 44100, real rate: 43945.238, BITS: 16, ERROR: 0.350934%
I (759) I2S: APLL: Req RATE: 44100, real rate: 44099.969, BITS: 24, ERROR: 0.000071%
I (774) I2S: APLL: Req RATE: 44100, real rate: 43945.238, BITS: 32, ERROR: 0.350934%

I (790) I2S: APLL: Req RATE: 48000, real rate: 47999.961, BITS: 16, ERROR: 0.000081%
I (805) I2S: APLL: Req RATE: 48000, real rate: 46874.922, BITS: 24, ERROR: 2.343913%
I (821) I2S: APLL: Req RATE: 48000, real rate: 43945.238, BITS: 32, ERROR: 8.447420%

I (836) I2S: APLL: Req RATE: 96000, real rate: 87890.477, BITS: 16, ERROR: 8.447420%
I (852) I2S: APLL: Req RATE: 96000, real rate: 96000.031, BITS: 24, ERROR: 0.000033%
I (867) I2S: APLL: Req RATE: 96000, real rate: 107421.875, BITS: 32, ERROR: 11.897786%
```

closed 2019-06-30 by igrr

mbt28, 2020-03-30:
> The attached patch give is percentage difference of 0.001.

Hi,

How can I apply this patch file on Windows?

Thanks