espressif/esp-idf #3381: HTTP listen socket not closed in httpd_stop() leading to bind error 112 (EADDRINUSE)

henrix, 2019-04-28:
Hello,

I've created a web app based on the example http_server.
httpd is started and stopped via an external toggle switch.
If starting and stopping httpd without opening the website in a browser, everything is fine.
However, if I open the website (and close it again) and restart httpd, I get bind error 112 (EADDRINUSE). 
It seems that the listen socket is not closed properly if the website was opened once.
SO_REUSEADDR is enabled in sdkconfig.
Is there a way to get the listen socket via httpd API to close it properly or is something else needed?

Thanks!

anurag-kar, 2019-04-29:
@henrix `SO_REUSEADDR` is not set on the listening socket of httpd, so you need to wait about 30 seconds before starting httpd again. Here is a patch which should make this work by calling setsockopt() on the listener socket. In the meanwhile I'll see if this can feature can be added upstream.

[httpd_reuseaddr.patch.txt](https://github.com/espressif/esp-idf/files/3127017/httpd_reuseaddr.patch.txt)

henrix, 2019-04-29:
@anurag-kar Thanks a lot!

closed 2019-04-29 by henrix

ChenPoHsuan, 2019-07-30:
This patch causes error 112 on socket listen stage again.