espressif/esp-idf #3383: HTTP(S) request through a SOCK5 proxy?

Allineer, 2019-04-29:
Please, help me to understand, how to execute a HTTP(S) request through a SOCK5 proxy with auth?

Thanks.

Alvin1Zhang, 2019-04-29:
@Allineer Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Thanks.

Allineer, 2019-04-30:
## Environment

- Development Kit: none
- Kit version (for WroverKit/PicoKit/DevKitC): none
- Module or chip used:  ESP32-WROOM-32U
- IDF version (run ``git describe --tags`` to find it): v3.2-103-g4aa1058e8
- Build System: no matter
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): no matter
- Operating System: no matter
- Power Supply: no matter

## Problem Description

I have no problem, just question:
How to execute a HTTP(S) request through a SOCK5 proxy with auth?
Thanks.

shubhamkulkarni97, 2020-03-04:
@Allineer SOCK5 proxy is currently not supported in esp_http_client.

Alvin1Zhang, 2020-03-05:
@Allineer Thanks for reporting.

closed 2020-03-05 by Alvin1Zhang

jeanmatthieud, 2021-03-20:
Will it be supported at one point?

QGB, 2022-09-25:
```
2022/09/25 18:55:17 tcp:192.168.1.10:46458 rejected  proxy/socks: unknown Socks version: 67
```
只能  export https_proxy=http://192.168.1.3:1080
不能 
export https_proxy=socks5h://192.168.1.3:21080
export https_proxy=socks5://192.168.1.3:21080