espressif/esp-idf #3385: Unable to connect AP from "TP-LINK Archer VR2600" Router

ozzdemir, 2019-04-29:
## Environment

- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):  v3.2
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):  1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Power Supply:         [external 5V]

## Problem Description
We have an "TP-LINK  Archer VR2600" router that we try to connect to. After erase_flash and flashing the binaries, we sometimes connect (mostly fails) to ap with the logs given. Even if we succesfully connect, device is disconnects later on with reason codes 2, 4 and 6 (they change, but mostly 2 comes). If we fail connecting, the disconnection reason we get is also 2. We don't have any problem connecting to the ap when ap is a hotspot from an Android phone.

### Expected Behavior
It should connect and remain connected.
### Actual Behavior
It rarely connects and disconnects even if it connects.

## Debug Logs

I (204780) wifi: state: init -> auth (b0)
I (204807) wifi: state: auth -> assoc (0)
I (204828) wifi: state: assoc -> run (10)
I (204949) wifi: connected with Kuartis, channel 3
I (204950) wifi: pm start, type: 1
I (205986) wpa: PTK has been installed, it may be an attack, ignor it.
I (205986) wpa: GTK has been installed, it may be an attack, ignor it.
I (206333) wifi: n:3 1, o:3 1, ap:3 1, sta:3 1, prof:3
I (206334) wifi: station: f8:1a:67:0f:07:8f join, AID=1, bgn, 40U

W (213257) wifi: rx ampdu: duplicated addba request, potential compitability bug
W (213897) wifi: rx ampdu: duplicated addba request, potential compitability bug

W (65076) wifi: token mismatch, expect=2

[image]

william-ferguson-au, 2019-04-30:
I've had a report from the field for similar failed connections for a TP Link Archer C7 router.
Have one inbound to be able to verify here.

sysizlayan, 2019-04-30:
@william-ferguson-au Is it another reported issue on github? If so, if you can tag it here, we can merge the situations?
Thanks.

william-ferguson-au, 2019-04-30:
No I haven't reported it as an issue yet, because I haven't had the ability to reproduce, but it sounds like exactly the same. I'll follow this thread and update it with any data I get when the router shows up.

sysizlayan, 2019-05-09:
Is there any comment? @igrr

william-ferguson-au, 2019-05-20:
OK, I have tested an Archer C7 VR2800 and VR1750.
Both seem to provide a solid wifi connection. @sysizlayan how long did you need to wait until failure?

NB I am running esp-idf v3.3-beta3

sysizlayan, 2019-05-20:
Hi,
Could you please several connect - disconnect cycle? At the beginning, our routers also work fine. But after some connect - disconnect cycle (We are resetting the device during development), it begins to give reason 2 or reason 4 and does not allow connection. It continues until the router is restarted.

We have checked the blacklist page of the router, there was no item on it.

We are currently using 3.2 release of the IDF.

Thanks in advance :)

william-ferguson-au, 2019-05-20:
When you say " we are resetting the device"., which device? The router or the ESP32?

If the ESP32, how are you resetting it?

If you are programmatically disconnecting from WiFi are you calling esp_wifi_stop();
If not, then that could explain the errors you are seeing.

sysizlayan, 2019-05-21:
The device I have mentioned is esp32. When we are testing our firmware and seeing the specific log, for example, we are changing the firmware and upload the code to the ESP32. During firmware upload, the programmer resets the board, hence the wifi connection is broken. When the device boot up, it begins to connect to the router again.

The problem is that there is no problem for a long time. But after some time, the router does not allow the connection. I am seeing the handshake logs but not GOT_IP event.

Another thing is that, even if the connection is successful, it begins to be broken more frequently with reasons 4 or 6.

I am not sure this is because of the router or esp32. What makes me concerned about the ESP32 is that this logs:

I (205986) wpa: PTK has been installed, it may be an attack, ignor it.
I (205986) wpa: GTK has been installed, it may be an attack, ignor it.

We have not seen this logs before. Do you have any comment on that?

william-ferguson-au, 2019-05-21:
No I don't.
I'm keen to ensure this is not an issue, but I have been unable to reproduce the fault.
Maybe an app that connects to WiFi and then reboots some time after connecting, but doesn't reboot if it can't connect.

liuzfesp, 2019-05-21:
@william-ferguson-au could you help to check it with latest IDF please?

william-ferguson-au, 2019-05-21:
@liuzfesp I have confirmed with v3.3-beta3. What is version do you mean by latest?

I ran this in a loop for the last 9.5 hours. Connected to TP-Link AC1750.

- Startup
- Connect to Wifi
- if connected to WiFi: Wait 20 seconds and esp_restart();

It didn't fault once.

liuzfesp, 2019-05-22:
@william-ferguson-au, when I say latest, it means lastest v3.1 or v3.2 or v3.3 or master. For most of the WiFi bugs, we will fix it on IDF master first then backport to previous release. 

E.g, if the problem happens in IDF release/v3.1, then you can check it in latest v3.1. 

For problem that ESP32 unable to connect "TP-LINK Archer VR2600", if it's fixed in v3.3-beta3. It should have been fixed in v3.1/v3.2/master also.

Alvin1Zhang, 2019-05-24:
@william-ferguson-au Hi, thanks for reporting the issue. Would you help check whether your issue still exists or fixed on latest v3.2? Thanks.

william-ferguson-au, 2019-05-24:
@Alvin1Zhang I didn't report his issue @sysizlayan  did.
I am unable to reproduce it.

Alvin1Zhang, 2019-05-24:
@william-ferguson-au Thanks for the information. @sysizlayan @ozzdemir Would you help check whether your issue still exists or get fixed on latest v3.2? Thanks.

Alvin1Zhang, 2019-06-05:
@william-ferguson-au Thanks for reporting the issue, feel free to reopen if it still happens. Thanks.

closed 2019-06-05 by Alvin1Zhang