espressif/esp-idf #3386: esp32 wifi data frame unreadable by phone

tekpalb, 2019-04-29:
I have to develop an esp32 wifi mesh network,but using esp32's sta+ap mode both.phone is unable to read/decode the original data or unable to connect to internet,please do give any solution.
note-esp32 wifi mesh network(in sta+ap mode)

liuzfesp, 2019-07-16:
Hi @tek145, sorry for late response, could you describe your problem in more detailed.

Alvin1Zhang, 2019-10-11:
@tek145 Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

dongdong004, 2019-11-05:
Did you mean this issue?
When esp32 in mesh mode, the AP didn't allow non-mesh stations to be connected.
https://github.com/espressif/esp-mdf/issues/43

Alvin1Zhang, 2020-03-04:
@tek145 Thanks for reporting. Will close due to short of details and feedback, feel free to reopen with more details provided. Thanks.

closed 2020-03-04 by Alvin1Zhang