espressif/esp-idf #3387: Latest Master as of 04/29/19 - SPI_MASTER broke cant compile

filzek, 2019-04-29:
Hi,

The master update as of  04/29/19 - SPI_MASTER is broke and cant compile

spi_master.c:331:57: error: invalid operands to binary == (have 'struct <anonymous>' and 'void *')
         SPI_CHECK(atomic_load(&spihost[host]->device[x])==NULL, "not all CSses freed", ESP_ERR_INVALID_STATE);

********
and a lot of more errors in same file.

also, doing a make clean, and make will return a final error ldgen_libraries esp32

SDK is complete broken now.

sbela, 2019-04-30:
I have the same issue with the current version from the git repository

ammaree, 2019-04-30:
+1 same problem

bigbigblue, 2019-04-30:
+1 same problem

pjalocha, 2019-04-30:
+1 same problem

ammaree, 2019-05-02:
This issue is solved by the patch in #3394 
Thanks

ginkgm, 2019-05-07:
the fix has been merged into master now, could you have a try? @ammaree

ginkgm, 2019-05-21:
and also @filzek  @sbela

filzek, 2019-05-21:
It’s fixed!

closed 2019-05-21 by filzek