espressif/esp-idf #3388: esp-idf如何编译自己新建目录中的程序

gufeng224, 2019-04-29:
我想在自己新建的工程目录中，新建vendorapi然后编译里面的程序，应该怎么操作？麻烦帮忙解答一下，谢谢了

Alvin1Zhang, 2019-04-29:
@gufeng224 Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Thanks.

huming2207, 2019-04-30:
@gufeng224 Please ask this kind of general questions in the forum, not here. Meanwhile, this is GitHub and there are tons of users who can't read Chinese natively. So please do ask your question in English.

In the CMake build system can either try creating your component or put your `vendorapi` to your main component, then add something like this in the main component's CMakeList:

```cmake
file(GLOB_RECURSE VENDOR_API_SOURCES "vendorapi/*.cpp" "vendorapi/include/*.hpp")
file(GLOB_RECURSE VENDOR_API_INCLUDES LIST_DIRECTORIES true "vendorapi/include")

...
...

set(COMPONENT_SRCS "project_main.cpp" ${VENDOR_API_SOURCES})
set(COMPONENT_ADD_INCLUDEDIRS "" ${VENDOR_API_INCLUDES })
register_component()
```

Or alternatively, add your source files one by one to your `COMPONENT_SRCS` instead.

closed 2019-05-09 by Alvin1Zhang