espressif/esp-idf #3389: Too high stack usage by tasks

no1seman, 2019-04-29:
## Environment

- Development Kit:  ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used: ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it): ESP-IDF v4.0-dev-464-ge7a7d216b
- Build System:  Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a
- Operating System: Linux
- Power Supply: USB|external 5V&Battery

## Problem Description
When running simple task (for example BLINK task from my test project) there are too high HWM of tasks stack. 
If using xTaskCreatePinnedToCore(): HWM of 'BLINK' task is 1652 bytes, while typical stack usage after task startup is ~400 bytes.
If using  xTaskCreate(): HWM of 'BLINK' is 1620 bytes, while typical stack usage after task startup is ~396 bytes.
Also I noticed that HWM of sophisticated tasks (such as wifi, http server, mqtt_client) is much lower, look below into included file from one of my projects).
For example: 
wifi task - have 1532 bytes HWM (STA+AP mode), 
mqtt_task - HWM1388 bytes (nonSSL mqtt client),
httpd - HWM  1978 bytes (11 uri handlers, sophisticted logic including JSON generation).

So I can't understand what I'm doing wrong? 
If there any receipts to reduce HWM of tasks or if there some bugs in IDF SDK FreeRTOS core?

### Expected Behavior
HWM must be much lower.
### Actual Behavior

### Steps to repropduce
Clone the project and run it.

### Code to reproduce this issue
You can find all project here:
https://github.com/no1seman/esp32_tasks_test

## Debug Logs
if using xTaskCreatePinnedToCore():
...
I (91140) tasks_test: Task name: 'main', Stack size: 4096, Stack used: 420, HWM: 2300
I (91140) tasks_test: Task name: 'IDLE1', Stack size: 1536, Stack used: 420, HWM: 1116
I (91140) tasks_test: Task name: 'IDLE0', Stack size: 1536, Stack used: 428, HWM: 1108
**I (91150) tasks_test: Task name: 'BLINK', Stack size: 2048, Stack used: 396, HWM: 1652**
I (91160) tasks_test: Task name: 'Tmr Svc', Stack size: 2048, Stack used: 424, HWM: 1624
I (91170) tasks_test: Task name: 'ipc0', Stack size: 1024, Stack used: 420, HWM: 604
I (91180) tasks_test: Task name: 'esp_timer', Stack size: 4096, Stack used: 428, HWM: 3668
I (91190) tasks_test: Task name: 'ipc1', Stack size: 1024, Stack used: 456, HWM: 568
...
if using xTaskCreate():
I (10350) tasks_test: Task name: 'main', Stack size: 4096, Stack used: 420, HWM: 2300
I (10350) tasks_test: Task name: 'IDLE1', Stack size: 1536, Stack used: 420, HWM: 1116
I (10350) tasks_test: Task name: 'IDLE0', Stack size: 1536, Stack used: 428, HWM: 1108
**I (10360) tasks_test: Task name: 'BLINK', Stack size: 2048, Stack used: 396, HWM: 1620**
I (10370) tasks_test: Task name: 'Tmr Svc', Stack size: 2048, Stack used: 424, HWM: 1624
I (10380) tasks_test: Task name: 'ipc1', Stack size: 1024, Stack used: 456, HWM: 568
I (10390) tasks_test: Task name: 'ipc0', Stack size: 1024, Stack used: 420, HWM: 604
I (10390) tasks_test: Task name: 'esp_timer', Stack size: 4096, Stack used: 428, HWM: 3668
...

## Other items if possible
json from my project with tasks statistics:
[ps.zip](https://github.com/espressif/esp-idf/files/3128860/ps.zip)

chegewara, 2019-04-29:
Did you try to check stack usage with printf in any of task?

no1seman, 2019-04-29:
@chegewara 
If you look into projects code (you can find it here: https://github.com/no1seman/esp32_tasks_test) there is only one printf-like call:  ESP_LOGI(TAG, "Blink task stopped"); and it is never called, because I'm not finishing the task. So printf() is not a cause of my question.

chegewara, 2019-04-29:
Ok, but my point is, if you are using printf in your tasks then you need increase stack size by almost 1,5kB, thats why blink and other examples have stack so high.
You can find many issues here and on forum where it is explained.

anurag-kar, 2019-04-29:
@no1seman higher the high stack watermark indicates the minimum space unused on the stack. So the higher the value the better it is. I guess you understood it the other way.

See https://www.freertos.org/uxTaskGetStackHighWaterMark.html

no1seman, 2019-04-29:
@anurag-kar, thanks for thanks for the explanation
So, in my case that means that:
[Stack size] = 2048 bytes
[Stack usage] = 396 bytes
[Max stack usage] = [Stack size] - [HWM] = 2048 - 1652 = 396 bytes
Right?

anurag-kar, 2019-04-29:
Exactly! The value tallies up :
```
 (91150) tasks_test: Task name: 'BLINK', Stack size: 2048, Stack used: 396, HWM: 1652
```

no1seman, 2019-04-29:
Thanks for All! Closing issue...

closed 2019-04-29 by no1seman