espressif/esp-idf #3391: 如何检测esp32的每个GPIO中断？（出现一些中断未触发的情况）

w719367295, 2019-04-30:
#  我做了这些事情： 
设置了一个80微妙的高精度定时器用作输出IO口（GPIO5）控制，产生pwm波形。一个输入IO口（GPIO23）作为输入中断，作为控制信号控制输出IO口的波形输出。
# 问题： 
输入IO口的信号稳定发生，但是输入IO口的中断并没有稳定触发，会出现部分中断未触发的情况，导致输出IO口不能稳定的输出波形。是什么情况导致了这种事情的发生呢？高精度定时器是否会打断GPIO中断？请问我应该这么做才能让每一个输入信号都被检测到呢？
```c
volatile u32_t  zero_bit=0;

/***********************
GPIO23中断函数：testgpio_isr_handler(void* arg)
*************************/
void IRAM_ATTR testgpio_isr_handler(void* arg)
{
		zero_bit=0;
}

/********************
定时器中断函数：gpio_isr_handler_zerow(void* arg)
**************/
void IRAM_ATTR gpio_isr_handler_zerow(void* arg)
{ 
	
	if(zero_bit>105)
          gpio_set_level(5, 0);//把这个GPIO5输出低电平
        else if(zero_bit>=50)
	  gpio_set_level(5, 1);//把这个GPIO5输出高电平
	else 
	 gpio_set_level(5, 0);//把这个GPIO5输出低电平

       zero_bit++;
}

static int gpio_input_init(void)
{
   //GPIO口结构体定义
    gpio_config_t io_conf;
    //边沿触发
    io_conf.intr_type =GPIO_PIN_INTR_ANYEDGE;
    //选择为输入模式
    io_conf.mode = GPIO_MODE_INPUT;
    //配置GPIO_OUT寄存器
    io_conf.pin_bit_mask = (1<<23);//输入io口23
    //设置下拉
    io_conf.pull_down_en = 0;
    //设置上拉
    io_conf.pull_up_en = 1;
    //最后配置使能
    gpio_config(&io_conf);
    
    //输出IO口配置
    io_conf.mode = GPIO_MODE_OUTPUT;
    io_conf.intr_type =GPIO_PIN_INTR_DISABLE;
    //bit mask of the pins that you want to set,e.g.GPIO5
    io_conf.pin_bit_mask = (1<<5); //输出IO口5
    //disable pull-down mode
    io_conf.pull_down_en = 0;
    //disable pull-up mode
    io_conf.pull_up_en = 0;
    //configure GPIO with the given settings
    gpio_config(&io_conf);
    
    //配置中断
    gpio_install_isr_service(0);
    gpio_isr_handler_add(23, testgpio_isr_handler, NULL);
    
    return 0;
}

void app_main()
{
  gpio_input_init();
  esp_timer_handle_t test_p_handle = 0;
	//定义一个周期重复运行的定时器结构体
	esp_timer_create_args_t test_periodic_arg = { 
	.callback = &gpio_isr_handler_zerow, //设置回调函数
        .arg = NULL, //不携带参数
        .name = "gpio_pwm" //定时器名字

        };
	vTaskDelay(1000/ portTICK_RATE_MS);
	ESP_ERROR_CHECK(esp_timer_create(&test_periodic_arg , &test_p_handle));
	ESP_ERROR_CHECK(esp_timer_start_periodic(test_p_handle, 80));
}
```

koobest, 2019-04-30:
Hi, @w719367295 
GPIO23 的输入是不是也是 PWM? 频率是多少?

w719367295, 2019-04-30:
@koobest 是PWM输入，频率为50Hz,需要双边沿触发

koobest, 2019-04-30:
Hi, @w719367295 
我们进行了测试, 但是从抓取的波形看没有发现错误的波形, 都比较稳定, 也有可能是没有抓取到.

[image]

你们是想实现什么样的功能?我们看看是否可以用硬件来实现.这样比软件稳定一点,
thanks!!

w719367295, 2019-04-30:
@koobest 非常感谢！我再试试看能不能重现这个问题。

w719367295, 2019-04-30:
@koobest 哦！我们想要实现的是一个利用可控硅进行调光的一个开关，你们有什么比较好的方法吗？谢谢！

koobest, 2019-04-30:
Hi, @w719367295 
我对可控硅进行调光了解不多. 你们的电路是否是需要一路 100Hz 的 PWM 波? 通过调整 PWM 的相位角(50Hz 输入信号作为参考)达到调光的目的? 对占空比有没有要求.

w719367295, 2019-04-30:
@koobest 非常感谢！关于可控硅调光控制，我是通过控制可控硅导通时间以达到调整亮度的目的，实现过程：
[image]
现在使用可控硅调光的开发应该是比较多的，希望你们可以添加一些使用esp32进行可控硅调光的参考。
关于你说的MCPWM我还没有使用过，稍后我会去尝试使用。如果你有什么好的方法请告诉我！
 再次感谢！

Alvin1Zhang, 2019-05-09:
@w719367295 Feel free to reopen if the issue still exists. Thanks.

closed 2019-05-09 by Alvin1Zhang