espressif/esp-idf #3392: Allow VFS components to work with concurrent select calls from several tasks.

skandalfo, 2019-04-30:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): 
    // v4.0-dev-447-g689032650
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1-rc2) 8.2.0
- Operating System:     Linux
- Power Supply:         USB

## Problem Description

I'm currently using "simulated pipes" by connecting together two datagram sockets with ephemeral ports.

I'd like to implement my own VFS module to support signaling (set/clear flag) fds. Unfortunately the current interface involving start_select/end_select doesn't allow for it because end_select doesn't provide back any context pointer for the select call that just finished..

### Expected Behavior

start_select / end_select calls should be linked together with some state pointer, so that simultaneous waits (as opposed to "singleton select" in the uart vfs module) can be properly attributed/signalled.

### Actual Behavior

`s if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

dobairoland, 2019-08-08:
Hi @skandalfo! Thank you for pointing out this problem and sorry for the long silence regarding this. The fix is under internal review and got delayed. It will be published as soon as possible.

skandalfo, 2019-08-12:
Hi, @dobairoland, thanks for letting me know :-)

My plans were use select-enable synchronization primitives to properly port [cpp-async-rpc](https://github.com/google/cpp-async-rpc) without having to use lots of sockets.

closed 2019-08-21 by igrr