espressif/esp-idf #3394: Windows setup - make: *** No rule to make target 'flash'. Stop.

yvedant, 2019-04-30:
I have followed the official ESP IDF get started guide until the build and flash section. Until then all went fine, the make menuconfig command also runs fine. But when I run 'make flash' it takes time compiles everything and at the end give the stated error.

make: *** No rule to make target 'IDF_PATH/hello_world/build/ldgen_libraries', needed by 'IDF_PATH/hello_world/build/esp32/esp32.project.ld'. Stop.

please suggest me something thanks in advance

lettierimarco, 2019-04-30:
Have you correctly setted the IDF_PATH?

yvedant, 2019-04-30:
yes my path is correct

luuksteitner, 2019-05-01:
I'm having the exact same issue. There's one thing that I noticed that's odd about my installation, that could be the same on your machine, and may have to do with the issue: Everywhere where I check the documentation and instructions online, I see the IDF_PATH is located in the msys32/home/user-name/ directory. However, when I follow the installation instructions the esp-idf directory always ends up in "C:\Users\luuks\AppData\Roaming\SPB_Data\esp\esp-idf".
I have setup the IDF_PATH correctly, hence the build starts and continues for a few minutes but then always stops with the message:
"make: *** No rule to make target '/c/Users/luuks/AppData/Roaming/SPB_Data/esp/esp-idf/examples/get-started/hello_world/build/ldgen_libraries', needed by '/c/Users/luuks/AppData/Roaming/SPB_Data/esp/esp-idf/examples/get-started/hello_world/build/esp32/esp32.project.ld'."
The strange thing is, I have tried the same setup on another machine (also Windows 10 64bit), where it does build successfully. The only difference in setup is that on the machine where it doesn't work, the msys32 directory is "D:\msys32" in stead of "C:\msys32". I would not expect that to be an issue.

luuksteitner, 2019-05-01:
A quick update: I checked the make files in the directory IDF_PATH/make. It turned out the make files there had many differences between the two machines. Since there are differences in this part, I guess there can be differences in other parts as well. I copied the entire esp-idf directory from the PC with working setup to the PC where it didn't work. Then I cleaned the project, rebuild it and voila, it worked.

Now the big question is, why are the two installations so different? I had downloaded everything on PC 1 last weekend, where it didn't work. I downloaded everything on PC 2 monday morning where it did work. Suspecting a version change between downloads, I re-downloaded and installed everything on PC 2 on Monday evening, still without success. Apparently somewhere during the installation process something went wrong twice on PC1, and I have no idea what.

bigbigblue, 2019-05-01:
I am having the same issue, however on my machine the msys32 directory is C:\msys32
I also receive the same error when attempting to build the NODEMCU firmware for the ESP32 and as there is no pre-built NODEMCU firmware for the ESP32 available with the modules I need, I am completely stuck.
Any help would be very much appreciated.

popurlti, 2019-05-01:
> However, when I follow the installation instructions the esp-idf directory always ends up in "C:\Users\luuks\AppData\Roaming\SPB_Data\esp\esp-idf".

That happens because you have installed Cadence OrCAD or similar on your machine. Cadence uses the HOME environment variable (used by MSYS2, too) and sets it to the path you cited.

luuksteitner, 2019-05-01:
> That happens because you have installed Cadence OrCAD or similar on your machine. Cadence uses the HOME environment variable (used by MSYS2, too) and sets it to the path you cited.

Ah, that makes sense. I have installed Cadence on both machines.

luuksteitner, 2019-05-01:
In the mean time I have switched to PlatformIO in Visual Studio code. It was set up in minutes and works out of the box.

dimitry-boersma, 2019-05-01:
Hey, I'm currently facing the same problem, I have the IDF_PATH set correctly AND the esp-idf is in the place it should be... It seems to build the rest without problems, but it outputs the same error

renzbagaporo, 2019-05-01:
Sorry for the trouble this issue has caused. The fix for this has been merged in our internal repo. It should find its way in the Github repo in the next couple of days.

theproxy, 2019-05-01:
Is there any way we can workaround on our own until the fix gets to Github?

bigbigblue, 2019-05-01:
I have cloned the V3.2 repo of esp-idf and compiled against this and all works OK. YMMV of course

yvedant, 2019-05-01:
Thanks let me check!

theproxy, 2019-05-01:
That appears to work for me. Thank you @bigbigblue

renzbagaporo, 2019-05-02:
Issue is caused by this commit: https://github.com/espressif/esp-idf/commit/b1ecd75d83c3e72b18bf140ff19f3e4f06b31bdb, so reverting to commits earlier than this should fix this issue. Alternatively, you can use the patch:

[0001-make-fix-missing-rule-for-creating-ldgen_libraries-f.patch.txt](https://github.com/espressif/esp-idf/files/3136468/0001-make-fix-missing-rule-for-creating-ldgen_libraries-f.patch.txt)

altcuim, 2019-05-02:
> Issue is caused by this commit: [b1ecd75](https://github.com/espressif/esp-idf/commit/b1ecd75d83c3e72b18bf140ff19f3e4f06b31bdb), so reverting to commits earlier than this should fix this issue. Alternatively, you can use the patch:
> 
> [0001-make-fix-missing-rule-for-creating-ldgen_libraries-f.patch.txt](https://github.com/espressif/esp-idf/files/3136468/0001-make-fix-missing-rule-for-creating-ldgen_libraries-f.patch.txt)

It work, but i using manual edit make/ldgen.mk file.

closed 2019-05-05 by igrr

modtronix-com, 2019-09-05:
I had the same problem. If fixed it by changing following line in .cproject file from:
`<target name="flash" path="main" targetID="org.eclipse.cdt.build.MakeTargetBuilder">`
to
`<target name="flash" path="" targetID="org.eclipse.cdt.build.MakeTargetBuilder">`

Turn out that in eclipse, you have to click on the actual project name(not a folder) in the "project explorer", and then select "Build Targets" from content menu. I had the "main" folder selected in my "project explorer", and then selected "Project"-->"Build Target" from menu. This created a wrong flash entry in the .cproject file, with path="main"