espressif/esp-idf #3395: cmake cannot find esp_vfs_fat.h in components

Arthedian, 2019-04-30:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): 
    v3.3-beta2-202-g56afb3bd1
- Build System:        CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a
- Operating System:     Debian
- Power Supply:         USB

## Problem Description

I have my library in 

> components/fatCustom
In fatCustom.c is this Code:

> #include "fatCustom.h"
> #include "esp_vfs.h"
> #include "esp_vfs_fat.h"
> #include "esp_system.h"

CMake cannot find this esp_vfs_fat.h header. But if I put the same code into the main folder, than everything is ok. Why?

### Expected Behavior

successful include

### Actual Behavior

unsuccessful include

### Steps to reproduce

1. use this library in components

igrr, 2019-04-30:
This is because of the special handling on `main` component requirements. Please see https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system-cmake.html#component-requirements and https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system-cmake.html#renaming-main-component.

Arthedian, 2019-05-01:
Ok, but I am not suro what is the component name for this library to be put into COMPONENT_PRIV_REQUIRES

renzbagaporo, 2019-05-01:
The component name is the name of the directory containing the component CMakeLists.txt. In the case of esp_vfs_fat.h, it's `fatfs`.

Arthedian, 2019-05-01:
Thank you for the clarification :)

closed 2019-05-01 by Arthedian

CharlesSeartech, 2020-10-23:
Good day, I had the same problem. There is no mention in the documentation of the key words to use for the REQUIRES options.
Can you consider adding to the key words to the documentation or at least the mechanism.

Charles