espressif/esp-idf #3396: Eclipse commands. not found & python dependencies

theobouwman, 2019-04-30:
I followed the https://docs.espressif.com/projects/esp-idf/en/latest/get-started/eclipse-setup.html guide step-by-step but get this error:

```
16:50:33 **** Incremental Build of configuration Default for project row-application ****
make -j7 all 
make: xtensa-esp32-elf-gcc: Command not found
expr: syntax error
including /Users/theobouwman/dev/esp/esp-idf/components/app_update/Makefile.projbuild...
including /Users/theobouwman/dev/esp/esp-idf/components/bootloader/Makefile.projbuild...
including /Users/theobouwman/dev/esp/esp-idf/components/bootloader_support/Makefile.projbuild...
including /Users/theobouwman/dev/esp/esp-idf/components/efuse/Makefile.projbuild...
including /Users/theobouwman/dev/esp/esp-idf/components/esp32/Makefile.projbuild...
including /Users/theobouwman/dev/esp/esp-idf/components/esptool_py/Makefile.projbuild...
including /Users/theobouwman/dev/esp/esp-idf/components/mbedtls/Makefile.projbuild...
including /Users/theobouwman/dev/esp/esp-idf/components/nghttp/Makefile.projbuild...
including /Users/theobouwman/dev/esp/esp-idf/components/partition_table/Makefile.projbuild...
including /Users/theobouwman/dev/esp/esp-idf/components/spiffs/Makefile.projbuild...
including /Users/theobouwman/dev/esp/esp-idf/components/ulp/Makefile.projbuild...
including /Users/theobouwman/dev/esp/esp-idf/components/unity/Makefile.projbuild...
/bin/sh: xtensa-esp32-elf-gcc: command not found
/bin/sh: xtensa-esp32-elf-gcc: command not found
Toolchain path: 
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
python /Users/theobouwman/dev/esp/esp-idf/tools/check_python_dependencies.py
The following Python requirements are not satisfied:
future>=0.15.2
cryptography>=2.1.4
Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.
Alternatively, you can run "/usr/bin/python -m pip install --user -r /Users/theobouwman/dev/esp/esp-idf/requirements.txt" for resolving the issue.
make: *** No rule to make target `/Users/theobouwman/dev/esp/projects/row-application/build/include/config/auto.conf', needed by `/Users/theobouwman/dev/esp/projects/row-application/build/partitions_singleapp.bin'.  Stop.
"make -j7 all" terminated with exit code 2. Build might be incomplete.

16:50:34 Build Failed. 4 errors, 0 warnings. (took 550ms)
```

My `.bash_profile` is this:
```
export GOPATH=~/dev/go
export ANDROID_HOME="/Users/theobouwman/Library/Android/sdk"
export ANDROID_NDK="/Userstheobouwman/Library/Android/sdk/ndk-bundle"

export PATH="$PATH:/bin"
export PATH="$PATH:~/dev/grpc/bins/opt"
export PATH="$PATH:~/dev/istio-0.7.1/bin"
export PATH="$PATH:/usr/local/sbin"
export PATH="$PATH:~/openwhisk/bin"
export PATH="$PATH:~/.composer/vendor/bin"
export PATH="$PATH:$ANDROID_HOME/tools:$ANDROID_HOME/platform-tools"
export PATH="$PATH:~/Library/Android/sdk/ndk-bundle"
export PATH="$PATH:~/dev/flutter/bin"
export PATH="$PATH:/Users/theobouwman/.pub-cache/bin"
export PATH="$PATH:$GOPATH"
export PATH="$PATH:$GOPATH/bin"
export PATH="$PATH:/Users/theobouwman/dev/vcpkg"
export PATH="$PATH:/opt/local/bin"
export PATH="$PATH:/Users/theobouwman/dev/gcc-arm-none-eabi-8-2018-q4-major/bin"

export PATH="$PATH:/Library/Frameworks/Python.framework/Versions/3.6/bin"

export PATH="$HOME/.cargo/bin:$PATH"
export PATH="/usr/local/opt/opencv@2/bin:$PATH"
export PATH="/usr/local/opt/openssl/bin:$PATH"

# JAVA
#export CLASSPATH=".:/usr/local/lib/antlr-4.7.1-complete.jar:$CLASSPATH"
#alias antlr4='java -Xmx500M -cp "/usr/local/lib/antlr-4.7.1-complete.jar:$CLASSPATH" org.antlr.v4.Tool'
#alias grun='java -Xmx500M -cp "/usr/local/lib/antlr-4.7.1-complete.jar:$CLASSPATH" org.antlr.v4.gui.TestRig'

# For compilers to find zlib you may need to set:
#export LDFLAGS="-L/usr/local/opt/zlib/lib"
#export CPPFLAGS="-I/usr/local/opt/zlib/include"

# Terminal
export PS1="\[\033[36m\]\u\[\033[m\]@\[\033[32m\]\h:\[\033[33;1m\]\w\[\033[m\]\$ "
export CLICOLOR=1
export LSCOLORS=ExFxBxDxCxegedabagacad
alias ls='ls -GFh'

# ARM
export GNUARMEMB_TOOLCHAIN_PATH="~/dev/gcc-arm-none-eabi-8-2018-q4-major/"
export ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb

# ESP
export ESPRESSIF_TOOLCHAIN_PATH="$HOME/dev/esp/xtensa-esp32-elf"
export IDF_PATH="$HOME/dev/esp/esp-idf"
export ESP_IDF_PATH=$IDF_PATH
export PATH="$PATH:$ESPRESSIF_TOOLCHAIN_PATH/bin:$ESP_IDF_PATH"

# Zephyr
export ZEPHYR_BASE="$HOME/dev/zephyrproject/zephyr"

# Haskell Stack Completion
#eval "$(stack --bash-completion-script stack)"
```

When I use `make flash` in the root of my project it does work.

Alvin1Zhang, 2019-05-06:
@theobouwman Hi, thanks for reporting the issue. We will look into this. Thanks.

david-cermak, 2019-05-06:
Hi @theobouwman, 

It looks like the PATH environmental variable is not correctly setup in eclipse (in contrast to your terminal as shown in `.bash_profile` you've posted)

Have you configured this variable in the following step https://docs.espressif.com/projects/esp-idf/en/latest/get-started/eclipse-setup.html#project-properties? Can you please check the value (or post here the content of the `PATH` variable) in `ProjectProperties` -> `C/C++ Build` -> `Environment`?

[image]

theobouwman, 2019-05-07:
@david-cermak this is the `PATH` variable value in the Eclipse Environment settings `/bin:/usr/bin:/usr/sbin:/sbin:/Users/theobouwman/dev/esp/xtensa-esp32-elf/bin`

david-cermak, 2019-05-07:
@theobouwman Thanks for the update. Still the same error?

From the above error message it seems both toolchain and python cannot be found. I can understand the python part, as there could be many pythons on system and the one used to install dependencies (https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#step-4-install-the-required-python-packages) might be different from the one executed from eclipse. Can you please run `which python` and make sure you get that path prepended in your `PATH` in eclipse (must be before any other path containing python on your system, e.g. `/usr/bin` usually has python on macOS).

No idea about the `xtensa-esp32-elf-gcc: Command not found` part, as you clearly reference the toolchain bin path in eclipse.

Alvin1Zhang, 2019-05-20:
@theobouwman Hi, thanks for reporting the issue. Would you help share if any updates for the issue? Thanks.

Alvin1Zhang, 2019-10-11:
@theobouwman Thanks for reporting, will close the ticket for now due to short of feedback. Feel free to reopen if you still have the issue or have more updates. Thanks.

closed 2019-10-11 by Alvin1Zhang

siddyboy11, 2021-02-21:
@david-cermak  i have trouble getting past make menuconfig with mingw32.
I get error bin/sh : xtensa-esp32-gcc command not found 
I have tried setting path using export as well as setting path in environmental variables aslo
But still the error remains