espressif/esp-idf #3397: Different ciphered text from aes128cbc mbedtls on different esp32

lettierimarco, 2019-04-30:
```
void encrypt(char * plainText, char * key, unsigned char * outputBuffer){

  mbedtls_aes_context aes;
  for(uint8_t i=0;i<16;i++){iv[i]=i;} //reset IV
  mbedtls_aes_init( &aes );
  mbedtls_aes_setkey_enc( &aes, (const unsigned char*) key, strlen(key) * 8 );
	mbedtls_aes_crypt_cbc(&aes,
			MBEDTLS_AES_ENCRYPT,
				                    32,
				                    iv,
									(const unsigned char*)plainText,
									outputBuffer);
  mbedtls_aes_free( &aes );
}

void decrypt(unsigned char * chipherText, char * key, unsigned char * outputBuffer){
  for(uint8_t i=0;i<16;i++){iv[i]=i;} //reset IV
  mbedtls_aes_context aes;

  mbedtls_aes_init( &aes );
  mbedtls_aes_setkey_dec( &aes, (const unsigned char*) key, strlen(key) * 8 );
	mbedtls_aes_crypt_cbc(&aes,
			MBEDTLS_AES_DECRYPT,
				                    32,
				                    iv,
									(const unsigned char*)chipherText,
									outputBuffer);
	mbedtls_aes_free( &aes );
}

void app_main() {
	esp_err_t err = nvs_flash_init();
	if (err == ESP_ERR_NVS_NO_FREE_PAGES || err == ESP_ERR_NVS_NEW_VERSION_FOUND) {
		// NVS partition was truncated and needs to be erased
		// Retry nvs_flash_init
		ESP_ERROR_CHECK(nvs_flash_erase());
		err = nvs_flash_init();
	}
	char plainText[]={ 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0 };
	    ESP_LOGI(MAIN_TAG,"STRLEN %i",strlen(key));

		unsigned char cipherTextOutput[32];
		unsigned char decipheredTextOutput[32];

		encrypt(plainText, key, cipherTextOutput);
		decrypt(cipherTextOutput, key, decipheredTextOutput);

		printf("\nOriginal plain text:");
		for (int i = 0; i < 32; i++) {
		printf("%02x",plainText[i]);
		}
		printf("\n");
printf("\nCiphered text:");
		for (int i = 0; i < 32; i++) {
		printf("%02x",cipherTextOutput[i]);
		}
		printf("\n");
		printf("\nDeciphered text:");
			for (int i = 0; i < 32; i++) {
				printf("%02x",decipheredTextOutput[i]);
			}
			printf("\n");
}
```
		
On esp32 DOIT devkit:
`Ciphered text:f1f6a87029d431dc4cbd467b8755a9a7f3d84c18d6536c7ee58414f17a426aac
`
On official esp32 module:
`Ciphered text:f92314823339888c0eae4abbdc22f6c0737d25ef14cee29ec15382b150a772c2
`

chegewara, 2019-04-30:
Could you add key value you are using in this test?

lettierimarco, 2019-04-30:
@chegewara 
`char key[] = {0x0F,0x05,0x06,0x0C,0x0,0x04,0x01,0x0F,0x09,0x09,0x00,0x0E,0x05,0x03,0x0,0x74};
`

chegewara, 2019-04-30:
This is the problem:
`mbedtls_aes_setkey_dec( &aes, (const unsigned char*) key, strlen(key) * 8 );`

You are using strlen with key that has fifth element 0x00, so your encoding is using 40 as key encoding type (which is completely wrong).

I cant explain why you are getting different hash on 2 different esp32 boards, but start with fixing this issue.

lettierimarco, 2019-05-01:
Also without the strlen the chipered text is different.
Is the aes calculation connected to hardware configuration or some fuse bit?
I've 4 DOIT devboards return the same result, and 3 espressif module with same result but different from doit ones.
Obviusly the firmware uploaded is the same.
Someone have any idea?

chegewara, 2019-05-01:
@projectgus Help please

projectgus, 2019-05-01:
Well spotted @chegewara :)

Hi @lettierimarco,

Even if hardware AES acceleration is used, there isn't any dependency on the hardware efuse or other internals.

There isn't enough code in the issue post to build the full sample (some variable declarations are missing, for example). Can you please post the full .c file (include the strlen fix), and also your sdkconfig file? You can put them in a .zip file and attach them in a comment if you like.

lettierimarco, 2019-05-02:
attached the sdkconfig and main
[aes-test.zip](https://github.com/espressif/esp-idf/files/3136978/aes-test.zip)

projectgus, 2019-05-02:
> Now the output on ESP32 module is:
> Original plain text:0102030405060708090a0b0c0d0e0f1001000000000000000000000000000000

> And on ESP32 DOIT:
> Original plain text:0102030405060708090a0b0c0d0e0f1000000000000000000000000000000000

Are you sure there two devices are running the same firmware? These plaintext strings aren't the same (look at the byte at index 16.)

lettierimarco, 2019-05-02:
@projectgus sorry for the wrong copy of the output.
yes is the same firmware.

ESP32 DOIT:
```
Original plain text:0102030405060708090a0b0c0d0e0f1001000000000000000000000000000000
Ciphered text:ecec89c2ad80b1e2f1355d8c5796944a9bab4f4d43b094369cdab88620a7a9cd
Deciphered text:0102030405060708090a0b0c0d0e0f1001000000000000000000000000000000

```
output on espressif ESP32 module is:

```
Original plain text:0102030405060708090a0b0c0d0e0f1001000000000000000000000000000000
Ciphered text:b2002c00d755af2806090f4d8894fb2300868871cecb290f5811a6855ccdc0de
Deciphered text:0102030405060708090a0b0c0d0e0f1001000000000000000000000000000000
```

lettierimarco, 2019-05-02:
i've solved, you were right was implementation issue.
sorry

closed 2019-05-02 by lettierimarco

projectgus, 2019-05-02:
@lettierimarco Glad you got everything working.