espressif/esp-idf #3399: A2DP sink output left/right channel swapped

redchenjs, 2019-05-01:
## Environment

- Development Kit: Widora AIR v6.0
- Module or chip used: ESP32 Rev.1
- IDF version: v3.2
- Build System: CMake
- Compiler version: 1.22.0-80-g6c4433a
- Operating System: Linux
- Power Supply: USB

## Problem Description

When ESP32 is used for A2DP streaming, the A2DP output left/right channels are swapped. I have tested the `i2s` example and the `I2S` driver works as expected, this problem seems to be related to A2DP data output.

### When playing left channel only audio, the LA output is:
[image]

### When playing right channel only audio, the LA output is:
[image]

### The expected left/right channel behavior according to TRM:
[image]

### Steps to repropduce

1. Get the `a2dp_sink` example from esp-idf v3.2.
2. Build the example and flash it to esp32.
3. Connect to esp32 with your phone and play the test files.
[test_files.zip](https://github.com/espressif/esp-idf/files/3133899/test_files.zip)

### Code to reproduce this issue

The `a2dp_sink` example from esp-idf v3.2

redchenjs, 2019-05-01:
Well, the `a2d_data_cb()` is just a wrapper for `i2s_write()`, so the problem might be SBC decoding output:
```
void bt_app_a2d_data_cb(const uint8_t *data, uint32_t len)
{
    size_t bytes_written;
    i2s_write(0, data, len, &bytes_written, portMAX_DELAY);
    if (++s_pkt_cnt % 100 == 0) {
        ESP_LOGI(BT_AV_TAG, "Audio packet count %u", s_pkt_cnt);
    }
}
```

blueMoodBHD, 2019-05-13:
Hi @redchenjs,  We have do some test, and the PCM decoded by the SBC actually has the correct left and right channel order. 
And the problem appear on the I2S side, our Drvier code defaults to the left channel in the high position, only need to set I2S_CONF_REG (0x0008) when I2S configuration the I2S_TX_MSB_RIGHT bit of the register, L/R, can be flipped over.

closed 2019-08-05 by igrr