espressif/esp-idf #3406: Can't use all OpenSSL function: undefined reference to `ERR_get_error'

anthodury, 2019-05-02:
## Environment

- Development Kit: esp32 devkit v1
- Module or chip used: ESP32-WROOM-32
- IDF version: v3.3-beta2-2-gebdcbe8c6
- Build System: Make
- Compiler version: a1.22.0-80-g6c4433a
- Operating System: Linux VM on Windows host
- Power Supply: USB

## Problem Description

Hello,

I would like to use some OpenSSL functions like: 
`int ERR_get_error(void);` defined here : https://github.com/espressif/esp-idf/blob/master/components/openssl/include/openssl/ssl.h but an error came during linking.
However I can call `TLSv1_client_method();` without problems. Do you know what some functions can't be called?

### Steps to repropduce

1. Use for example the hello_world project
2. In the main hello_world_main.c file, add #include "openssl/ssl.h" then call the function ERR_get_error();

### Code to reproduce this issue

```cpp
// Include openssl header
#include "openssl/ssl.h"

....

void app_main()
{
// Call the function : this one doesn't work
ERR_get_error();

// Call the function : this one work
TLSv1_client_method();

...
```

## Debug Logs

LD build/hello-world.elf
/home/anthony/esp/hello_world/build/main/libmain.a(hello_world_main.o):(.literal.app_main+0x2c): undefined reference to `ERR_get_error'
/home/anthony/esp/hello_world/build/main/libmain.a(hello_world_main.o): In function `app_main':
/home/anthony/esp/hello_world/main/hello_world_main.c:18: undefined reference to `ERR_get_error'
collect2: error: ld returned 1 exit status
/home/anthony/esp/esp-idf/make/project.mk:481: recipe for target '/home/anthony/esp/hello_world/build/hello-world.elf' failed
make: *** [/home/anthony/esp/hello_world/build/hello-world.elf] Error 1

negativekelvin, 2019-05-03:
The openssl shim layer to mbedtls only provides a subset of openssl functions.  I dont think mbedtls has an error queue.

anthodury, 2019-05-03:
Ok but the components "nghttp" and "asio" seem to use this function and don't return linking error.

`
anthony@debian:~/esp/esp-idf$ grep . -rnw -e "ERR_get_error"
Binary file ./.vscode/ipch/b0af6edccebd54bd/openssl_client_example_main.ipch matches
Binary file ./.vscode/ipch/eb1ea1651a82ecd/openssl_server_example_main.ipch matches
Binary file ./.vscode/ipch/920c82cf14c8dc1/nghttp.ipch matches
./components/openssl/include/openssl/ssl.h:377:int ERR_get_error(void);
./components/nghttp/nghttp2/doc/sources/tutorial-client.rst:55:             ERR_error_string(ERR_get_error(), NULL));
./components/nghttp/nghttp2/examples/libevent-client.c:332:         ERR_error_string(ERR_get_error(), NULL));
./components/nghttp/nghttp2/examples/libevent-client.c:353:         ERR_error_string(ERR_get_error(), NULL));
./components/nghttp/nghttp2/examples/client.c:384:    dief("SSL_set_fd", ERR_error_string(ERR_get_error(), NULL));
./components/nghttp/nghttp2/examples/client.c:389:    dief("SSL_connect", ERR_error_string(ERR_get_error(), NULL));
./components/nghttp/nghttp2/examples/client.c:545:    dief("SSL_CTX_new", ERR_error_string(ERR_get_error(), NULL));
./components/nghttp/nghttp2/examples/client.c:550:    dief("SSL_new", ERR_error_string(ERR_get_error(), NULL));
./components/nghttp/nghttp2/examples/libevent-server.c:148:         ERR_error_string(ERR_get_error(), NULL));
./components/nghttp/nghttp2/examples/libevent-server.c:158:         ERR_error_string(ERR_get_error(), NULL));
./components/nghttp/nghttp2/examples/libevent-server.c:190:         ERR_error_string(ERR_get_error(), NULL));
./components/nghttp/nghttp2/src/HttpServer.cc:2111:      std::cerr << ERR_error_string(ERR_get_error(), nullptr) << std::endl;
./components/nghttp/nghttp2/src/HttpServer.cc:2133:      std::cerr << ERR_error_string(ERR_get_error(), nullptr) << std::endl;
./components/nghttp/nghttp2/src/HttpServer.cc:2154:                << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/HttpServer.cc:2168:                  << ERR_error_string(ERR_get_error(), nullptr) << std::endl;
./components/nghttp/nghttp2/src/HttpServer.cc:2176:                  << ERR_error_string(ERR_get_error(), nullptr) << std::endl;
./components/nghttp/nghttp2/src/shrpx_tls.cc:748:    LOG(FATAL) << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/shrpx_tls.cc:782:               << " failed: " << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/shrpx_tls.cc:804:               << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/shrpx_tls.cc:817:                 << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/shrpx_tls.cc:823:                 << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/shrpx_tls.cc:835:              << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/shrpx_tls.cc:843:                 << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/shrpx_tls.cc:857:               << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/shrpx_tls.cc:870:               << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/shrpx_tls.cc:875:               << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/shrpx_tls.cc:885:                   << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/shrpx_tls.cc:896:                   << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/shrpx_tls.cc:941:                 << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/shrpx_tls.cc:950:                 << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/shrpx_tls.cc:1028:    LOG(FATAL) << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/shrpx_tls.cc:1053:               << " failed: " << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/shrpx_tls.cc:1061:              << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/shrpx_tls.cc:1068:                 << ": " << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/shrpx_tls.cc:1081:                 << ": " << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/shrpx_tls.cc:1092:                 << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/shrpx_tls.cc:1122:               << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/shrpx_tls.cc:1858:               << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/shrpx_tls.cc:1871:               << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/shrpx_connection.cc:404:                  << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/shrpx_connection.cc:640:                  << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/shrpx_connection.cc:693:        LOG(INFO) << "SSL_read: " << ERR_error_string(ERR_get_error(), nullptr);
./components/nghttp/nghttp2/src/nghttp.cc:667:                  << ERR_error_string(ERR_get_error(), nullptr) << std::endl;
./components/nghttp/nghttp2/src/nghttp.cc:2257:                << ERR_error_string(ERR_get_error(), nullptr) << std::endl;
[... cut ...]
./components/asio/asio/asio/include/asio/ssl/impl/context.ipp:933:        static_cast<int>(::ERR_get_error()),
./components/asio/asio/asio/include/asio/ssl/impl/context.ipp:976:        static_cast<int>(::ERR_get_error()),
./components/asio/asio/asio/include/asio/ssl/impl/context.ipp:1004:      static_cast<int>(::ERR_get_error()),
./components/asio/asio/asio/include/asio/ssl/impl/context.ipp:1028:      static_cast<int>(::ERR_get_error()),
./components/asio/asio/asio/include/asio/ssl/impl/context.ipp:1049:      static_cast<int>(::ERR_get_error()),`

negativekelvin, 2019-05-03:
https://github.com/espressif/esp-idf/blob/master/components/asio/component.mk#L4
https://github.com/espressif/esp-idf/blob/master/components/nghttp/component.mk#L7

So those files are excluded from the build

anthodury, 2019-05-03:
ok. So as some functions are not usable, wouldn't be a good idea to remove these functions from the wrapper headers?

closed 2020-08-03 by espressif-bot