espressif/esp-idf #3408: [ble mesh] provisioning

chegewara, 2019-05-02:
Maybe i am doing something wrong, but ble_mesh_node example needs to be re-provisioned after reset. Is there something that we need to do to store provisioning data in nvs? Maybe its some setting in menuconfig?

Isl2017, 2019-05-04:
Hi, Chegewara

I think you shall enable "BT_MESH_SETTINGS" in menuconfig to enable NVS storing feature.

chegewara, 2019-05-04:
Hi,
sorry but i cant find option in menuconfig. `BT_MESH_SETTINGS` is active, because without it ble mesh wont work at all, but after reset esp32 node i have to re-provision it.

WCCWCC, 2019-05-05:
Hi Chegewara，
    You should open this option in menuconfig：`Blutooth Mesh support --> Store Bluetooth Mesh key and configuration persistenly`.

chegewara, 2019-05-05:
It works, but:
- after using option reset node from nRF connect i have unprovisioned node again and when ive been trying to re-provisioning it with different app key this happened:
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40001277  PS      : 0x00060730  A0      : 0x80127f5c  A1      : 0x3ffcf660
A2      : 0x00000000  A3      : 0x3ffcf820  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x0000003d  A7      : 0x40404040  A8      : 0x73000000  A9      : 0x00000000
A10     : 0x00000003  A11     : 0x3ffdff70  A12     : 0x00000000  A13     : 0x00000001
A14     : 0x00060f20  A15     : 0x00000004  SAR     : 0x00000018  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400012e5  LEND    : 0x40001309  LCOUNT  : 0x80127f43

ELF file SHA256: 35e8d2eee7e2d28b40725d9654220d0ccb72d26e71d2d15100c980c093f58757

Backtrace: 0x40001277:0x3ffcf660 0x40127f59:0x3ffcf670 0x40127bbf:0x3ffcf7f0 0x40127683:0x3ffcf820 0x401276cb:0x3ffcf870 0x40126da6:0x3ffcf890 0x4012774d:0x3ffcf8b0 0x4012478f:0x3ffcf8d0 0x4014095a:0x3ffcf8f0 0
x4013ff1e:0x3ffcf990 0x4012576e:0x3ffcf9c0 0x40125999:0x3ffcf9e0 0x40123ca8:0x3ffcfa00 0x40143987:0x3ffcfcf0 0x40107ac1:0x3ffcfd30 0x4010bfbb:0x3ffcfd50 0x4010c51f:0x3ffd0000 0x4010c74d:0x3ffd0020 0x4010cb3b:0x
3ffd0050 0x4010cd7a:0x3ffd0070 0x40116866:0x3ffd0090 0x401053d0:0x3ffd00d0 0x401055fe:0x3ffd00f0 0x4008f2b9:0x3ffd0120
0x40127f59: csi_load_check at D:/esp32_ble_mesh/home/darek/esp/esp-idf/components/bt/ble_mesh/mesh_core/settings/settings_fcb.c:176 (discriminator 1)

0x40127bbf: settings_save_one at D:/esp32_ble_mesh/home/darek/esp/esp-idf/components/bt/ble_mesh/mesh_core/settings/settings_store.c:77

0x40127683: store_pending_net at D:/esp32_ble_mesh/home/darek/esp/esp-idf/components/bt/ble_mesh/mesh_core/settings.c:1559

0x401276cb: store_pending at D:/esp32_ble_mesh/home/darek/esp/esp-idf/components/bt/ble_mesh/mesh_core/settings.c:1559

0x40126da6: k_work_submit at D:/esp32_ble_mesh/home/darek/esp/esp-idf/components/bt/ble_mesh/mesh_core/settings.c:1559
 (inlined by) schedule_store at D:/esp32_ble_mesh/home/darek/esp/esp-idf/components/bt/ble_mesh/mesh_core/settings.c:872

0x4012774d: bt_mesh_store_net at D:/esp32_ble_mesh/home/darek/esp/esp-idf/components/bt/ble_mesh/mesh_core/settings.c:1559

0x4012478f: bt_mesh_provision at D:/esp32_ble_mesh/home/darek/esp/esp-idf/components/bt/ble_mesh/mesh_core/mesh_main.c:84

0x4014095a: prov_data at D:/esp32_ble_mesh/home/darek/esp/esp-idf/components/bt/ble_mesh/mesh_core/prov.c:1180

0x4013ff1e: bt_mesh_pb_gatt_recv at D:/esp32_ble_mesh/home/darek/esp/esp-idf/components/bt/ble_mesh/mesh_core/prov.c:1618

0x4012576e: proxy_complete_pdu at D:/esp32_ble_mesh/home/darek/esp/esp-idf/components/bt/ble_mesh/mesh_core/proxy.c:462

0x40125999: proxy_recv at D:/esp32_ble_mesh/home/darek/esp/esp-idf/components/bt/ble_mesh/mesh_core/proxy.c:557

0x40123ca8: bt_mesh_bta_gatts_cb at D:/esp32_ble_mesh/home/darek/esp/esp-idf/components/bt/ble_mesh/mesh_core/mesh_bearer_adapt.c:1808

0x40143987: bta_gatts_send_request_cback at D:/esp32_ble_mesh/home/darek/esp/esp-idf/components/bt/bluedroid/bta/gatt/bta_gatts_act.c:531

0x40107ac1: gatt_sr_send_req_callback at D:/esp32_ble_mesh/home/darek/esp/esp-idf/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761

0x4010bfbb: gatts_process_write_req at D:/esp32_ble_mesh/home/darek/esp/esp-idf/components/bt/bluedroid/stack/gatt/gatt_sr.c:1159

0x4010c51f: gatts_process_attribute_req at D:/esp32_ble_mesh/home/darek/esp/esp-idf/components/bt/bluedroid/stack/gatt/gatt_sr.c:1468

0x4010c74d: gatt_server_handle_client_req at D:/esp32_ble_mesh/home/darek/esp/esp-idf/components/bt/bluedroid/stack/gatt/gatt_sr.c:1663

0x4010cb3b: gatt_data_process at D:/esp32_ble_mesh/home/darek/esp/esp-idf/components/bt/bluedroid/stack/gatt/gatt_main.c:1147

0x4010cd7a: gatt_le_data_ind at D:/esp32_ble_mesh/home/darek/esp/esp-idf/components/bt/bluedroid/stack/gatt/gatt_main.c:1147

0x40116866: l2c_rcv_acl_data at D:/esp32_ble_mesh/home/darek/esp/esp-idf/components/bt/bluedroid/stack/l2cap/l2c_main.c:274

0x401053d0: btu_hci_msg_process at D:/esp32_ble_mesh/home/darek/esp/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:598

0x401055fe: btu_task_thread_handler at D:/esp32_ble_mesh/home/darek/esp/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:598

0x4008f2b9: vPortTaskWrapper at D:/esp32_ble_mesh/home/darek/esp/esp-idf/components/freertos/port.c:403
```
then esp32 restarted and is no longer useful with this app flashed:
```
E (820) BLE_MESH: net_key_set, No subnet with NetKeyIndex 0x000
ASSERTION FAIL [rc == 0] @ D:/esp32_ble_mesh/home/darek/esp/esp-idf/components/bt/ble_mesh/mesh_core/settings/settings_store.c:47:
        settings_load_cb, Failed to load setting valueE (5730) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (5730) task_wdt:  - IDLE0 (CPU 0)
E (5730) task_wdt: Tasks currently running:
E (5730) task_wdt: CPU 0: Btc_task
E (5730) task_wdt: CPU 1: IDLE1
E (10730) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10730) task_wdt:  - IDLE0 (CPU 0)
E (10730) task_wdt: Tasks currently running:
E (10730) task_wdt: CPU 0: Btc_task
E (10730) task_wdt: CPU 1: IDLE1
E (15730) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (15730) task_wdt:  - IDLE0 (CPU 0)
E (15730) task_wdt: Tasks currently running:
[... cut ...]
E (20730) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (20730) task_wdt:  - IDLE0 (CPU 0)
E (20730) task_wdt: Tasks currently running:
E (20730) task_wdt: CPU 0: Btc_task
E (20730) task_wdt: CPU 1: IDLE1
E (25730) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (25730) task_wdt:  - IDLE0 (CPU 0)
E (25730) task_wdt: Tasks currently running:
E (25730) task_wdt: CPU 0: Btc_task
E (25730) task_wdt: CPU 1: IDLE1
```

WCCWCC, 2019-05-05:
Hi Chegewara，

      I hope that you can provide more detailed information so that I can locate the problem. Or you tell us your process, I can reproduce the problem.If you are convenient, you can send the entire project to me.

     Note：You need to `make erase_flash` when you re-do it.

chegewara, 2019-05-05:
This is code to reproduce it:
https://github.com/espressif/esp-idf/tree/ble_mesh_release/esp-ble-mesh-v0.6.1/examples/bluetooth/ble_mesh/ble_mesh_node

and this is procedure i am using to reproduce it 100%:
> after using option reset node from nRF connect i have unprovisioned node again and when im trying to re-provisioning it 

Yes, i know i have to use `erase_flash` when this issue occurred.

This is my sdkconfig:
```
#
# Automatically generated file; DO NOT EDIT.
# Espressif IoT Development Framework Configuration
#
CONFIG_IDF_TARGET="esp32"

#
# SDK tool configuration
#
CONFIG_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_PYTHON="python"
CONFIG_MAKE_WARN_UNDEFINED_VARIABLES=y

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
CONFIG_APP_EXCLUDE_PROJECT_VER_VAR=
CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR=

#
# Bootloader config
#
CONFIG_LOG_BOOTLOADER_LEVEL_NONE=
CONFIG_LOG_BOOTLOADER_LEVEL_ERROR=
CONFIG_LOG_BOOTLOADER_LEVEL_WARN=
CONFIG_LOG_BOOTLOADER_LEVEL_INFO=y
CONFIG_LOG_BOOTLOADER_LEVEL_DEBUG=
CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE=
CONFIG_LOG_BOOTLOADER_LEVEL=3
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V=
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
CONFIG_BOOTLOADER_FACTORY_RESET=
CONFIG_BOOTLOADER_APP_TEST=
CONFIG_BOOTLOADER_WDT_ENABLE=y
CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE=
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
CONFIG_APP_ROLLBACK_ENABLE=

#
# Security features
#
CONFIG_SECURE_SIGNED_APPS_NO_SECURE_BOOT=
CONFIG_SECURE_BOOT_ENABLED=
CONFIG_FLASH_ENCRYPTION_ENABLED=

#
# Serial flasher config
#
CONFIG_ESPTOOLPY_PORT="/dev/ttyUSB0"
CONFIG_ESPTOOLPY_BAUD_115200B=y
CONFIG_ESPTOOLPY_BAUD_230400B=
CONFIG_ESPTOOLPY_BAUD_921600B=
CONFIG_ESPTOOLPY_BAUD_2MB=
CONFIG_ESPTOOLPY_BAUD_OTHER=
CONFIG_ESPTOOLPY_BAUD_OTHER_VAL=115200
CONFIG_ESPTOOLPY_BAUD=115200
CONFIG_ESPTOOLPY_COMPRESSED=y
CONFIG_FLASHMODE_QIO=
CONFIG_FLASHMODE_QOUT=
[... 946 lines cut ...]
CONFIG_UNITY_ENABLE_FLOAT=y
CONFIG_UNITY_ENABLE_DOUBLE=y
CONFIG_UNITY_ENABLE_COLOR=
CONFIG_UNITY_ENABLE_IDF_TEST_RUNNER=y
CONFIG_UNITY_ENABLE_FIXTURE=

#
# Virtual file system
#
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y

#
# Host File System I/O (Semihosting)
#
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128

#
# Wear Levelling
#
CONFIG_WL_SECTOR_SIZE_512=
CONFIG_WL_SECTOR_SIZE_4096=y
CONFIG_WL_SECTOR_SIZE=4096
```

WCCWCC, 2019-05-06:
Hi chegewara,

You can use the `ble_mesh_client_model` project first, and the project has detailed steps in the corresponding `tutorial` directory of the project . `nRF connect App` I have not tested it before.

We will find the problem and reply to you as soon as possible, using other apps, there is no situation you said.

closed 2019-07-29 by chegewara