espressif/esp-idf #3409: [ble mesh] can´t provision ble_mesh_node example using "nRF mesh" android app after upgrading to v0.6.1 ble mesh SDK

mudo007, 2019-05-02:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-DevKit DO IT 30 pin]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version :  v4.0-dev-297-g54ecdca32
- Build System:         [Make]
- Compiler version: 1.22.0-80-g6c4433a5
- Operating System:     [Windows|]
- Power Supply:         [USB]

## Problem Description
Using the ble_mesh_node example (under \examples\bluetooth\ble_mesh\ble_mesh_node) from the ble_mes_release/esp-ble-mesh-v0.6.1, I am no longer able to provision the Esp32 node using the "nrf Mesh" android app. I can see the device by scanning, but when I try to provision it, the app displays the "connecting" message for about 10 seconds, and then quickly displays the error message: "Error on connection state change"

When running the same example from the v0.5 branch, I was able to provision and get/set the generic onoff server states without issue using the nRF mesh app. 

### Expected Behavior
Device is provisioned and it is possible to set/get the generic onoff states using the nRF mesh android APP, the serial monitor connected to the esp32 shows that provisioning has been performed

### Actual Behavior
 app displays the "connecting" message for about 10 seconds, and then quickly displays the error message: "Error on connection state change". No message appears on the esp32 serial monitor.

### Steps to repropduce

1. compile and flash the Esp32 with the ble_mesh_node as it is
2. open the nRF mesh android app, click on the plus sign.
3. the ESP-BLE_MESH device shows on the scan list.
4. Click on the esp-ble-mesh entry

### Code to reproduce this issue
There's no added code. using the stock ble_mesh_node example

## Debug Logs

I (30) boot: ESP-IDF v4.0-dev-297-g54ecdca32-dirty 2nd stage bootloader
I (30) boot: compile time 16:58:30
I (31) boot: Enabling RNG early entropy source...
I (36) boot: SPI Speed      : 40MHz
I (40) boot: SPI Mode       : DIO
I (44) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00100000
I (82) boot: End of partition table
I (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2b1d4 (176596) map
I (157) esp_image: segment 1: paddr=0x0003b1fc vaddr=0x3ffbdb60 size=0x033cc ( 13260) load
I (162) esp_image: segment 2: paddr=0x0003e5d0 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/msys32/home/helld/esp/esp-idf/components/freertos/xtensa_vectors.S:177
9

I (164) esp_image: segment 3: paddr=0x0003e9d8 vaddr=0x40080400 size=0x01638 (  5688) load
I (175) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x8916c (561516) map
0x400d0018: _flash_cache_start at ??:?

I (378) esp_image: segment 5: paddr=0x000c918c vaddr=0x40081a38 size=0x0fed0 ( 65232) load
0x40081a38: ets_update_cpu_frequency at C:/msys32/home/helld/esp/esp-idf/components/esp32/clk.c:214

I (416) boot: Loaded app from partition at offset 0x10000
I (416) boot: Disabling RNG early entropy source...
I (416) cpu_start: Pro cpu up.
I (420) cpu_start: Application information:
I (425) cpu_start: Project name:     ble_mesh_node
I (430) cpu_start: App version:      1
I (435) cpu_start: Compile time:     May  2 2019 16:58:18
I (441) cpu_start: ELF file SHA256:  e649ae80fa205494...
I (447) cpu_start: ESP-IDF:          v4.0-dev-297-g54ecdca32-dirty
I (454) cpu_start: Starting app cpu, entry point is 0x40081040
0x40081040: call_start_cpu1 at C:/msys32/home/helld/esp/esp-idf/components/esp32/cpu_start.c:267

I (0) cpu_start: App cpu up.
I (464) heap_init: Initializing. RAM available for dynamic allocation:
I (471) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (477) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (483) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (489) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (495) heap_init: At 3FFCA2D0 len 00015D30 (87 KiB): DRAM
I (501) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (508) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (514) heap_init: At 40091908 len 0000E6F8 (57 KiB): IRAM
I (520) cpu_start: Pro cpu start user code
I (203) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (204) ble_mesh_node: Initializing...
I (214) BTDM_INIT: BT controller compile version [b638fbb]
I (214) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (314) phy: phy_version: 4100, 6fa5e27, Jan 25 2019, 17:02:06, 0, 0
I (764) ble_mesh_node: esp_ble_mesh_prov_cb, event = ESP_BLE_MESH_PROV_REGISTER_COMP_EVT
I (764) ble_mesh_node: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code 0
I (764) ble_mesh_node: BLE Mesh Node initialized
I (774) ble_mesh_node: esp_ble_mesh_prov_cb, event = ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT
I (784) ble_mesh_node: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code 0

attachments:
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/3140008/sdkconfig.zip)
[ble_mesh_node.zip](https://github.com/espressif/esp-idf/files/3140010/ble_mesh_node.zip)

chegewara, 2019-05-02:
Hi,
thats very strange, because i dont have any problems with nRF mesh, but im having issues with provisioning using espressif EspBleMesh app on android. With v0.5 ive got no problem to provision with that app, but now when android app is trying to change mtu then log on screen shows its only 23, which is not enough to provision ble mesh node.

mudo007, 2019-05-03:
Hello.
That should be very strange indeed, because I just tested with the Espressif EspBleMesh app and it worked without issues, and that should eliminate the possibility of a faulty ESP32.

mudo007, 2019-05-06:
I decided to fully investigate this "weird" issue, and tred to perform provisioning using a combination of:
-  sdk v0.6.1 node example/sdk v0.5 node example
- old board (used for a week with leds connected directly)/ new board (only used one led with a 290 ohm resistor)
- "nRF mesh"  / "Bluetooth Mesh by Silicon labs" android app for provisioning
- my phone (motorola x Force) / friend's phone (moto g5), always deleting app data before performing the test.

The results are:
-the sdk version used and the board selected did not influence on the results
-I could provision the node  using any phone with the silicon labs app
-I could not provision the nodes using my phone (moto x force) with the nRF mesh app, but I could provision it using the Silicon labs app.

So, It looks like the failure lies within my phone, and not with the espressif BLE stack. 

Closing it as there's nothing to be done with the expressif BLE mesh stack.

closed 2019-05-06 by mudo007