espressif/esp-idf #3411: Build failing after run make app

glanderson42, 2019-05-03:
OS: Ubuntu 18.04
Compiler version: 5.2.0
IDF Commit hash: f49e2afe9
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a

I first run ```make app``` command in my project, it built the app successfully, but later when I tried to run ```make app``` command I got the following error:
```
Toolchain path: /home/glanderson42/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
port/crypto_hash_mbedtls/crypto_hash_sha256_mbedtls.d:78: *** missing separator.  Stop.
port/crypto_hash_mbedtls/crypto_hash_sha256_mbedtls.d:78: *** missing separator.  Stop.
/home/glanderson42/esp/esp-idf/make/project.mk:530: recipe for target 'component-libsodium-build' failed
make: *** [component-libsodium-build] Error 2
```
I got it for every command, forexample ```make all```  ```make -j all flash monitor``` . I tried to reinstall the full blockchain and esp-idf but I still getting this error.

igrr, 2019-05-04:
Could you please attach `build/port/crypto_hash_mbedtls/crypto_hash_sha256_mbedtls.d` file to this issue?

After that, you can probably fix the problem by removing the build directory `rm -rf build` and running `make` again.

Alvin1Zhang, 2019-05-06:
@glanderson42 Hi, would you help share if any updates for this issue? Thanks.

glanderson42, 2019-05-06:
@igrr @Alvin1Zhang here is the crypto_hash_sha256_mbedtls.d file
[crypto_hash_sha256_mbedtls.zip](https://github.com/espressif/esp-idf/files/3147130/crypto_hash_sha256_mbedtls.zip)

Alvin1Zhang, 2019-05-06:
@glanderson42 Thanks for the details.

igrr, 2019-06-05:
It looks like the dependency file was half-written, possibly as a result of compilation process being terminated:

```
<snip>
/home/glanderson42/esp/esp-idf/components/mbedtls/mbedtls/include/mbedtls/sha256.h:

/home/glanderson42/esp/esp-idf/components/mbedtls/port/include/mbedtls/esp_config.h:

/home/glanderson42/Work/esp32-core/build/include/sdkconfig.h:

/home/glanderson42/
```
(note the last line is cut off)

Normally when the make process is terminated by SIGINT, all child processes are also terminated by SIGINT. When SIGINT is received by GCC, it cleans up half-written files. However if GCC process is terminated forcefully (e.g. by SIGKILL), then half-written files may still remain, causing such an issue.

If the issue re-appears in other cases than described, please feel free to reopen.

closed 2019-06-05 by igrr