espressif/esp-idf #3412: Can't run make menuconfig

sshwarts, 2019-05-03:
I yesterday pulled down the esp-idf to update my copy to the latest. I can make my project, but when I try to run make menuconfig I now get this:

On OSX (Mohave):

Toolchain path: /Users/scotts/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /Users/scotts/esp/esp-idf/requirements.txt are satisfied.
bison -t -l -p zconf -o zconf.tab.c /Users/scotts/esp/esp-idf/tools/kconfig/zconf.y
make[1]: *** [zconf.tab.c] Abort trap: 6

(I can't even find the zconf.tab.c in my file system -- but I'd guess this is a generated file)

sshwarts, 2019-05-03:
Follow on: hmm, can't run bison at all. bison --version gives Abort trap:6

sshwarts, 2019-05-03:
Removed bison and reinstalled using brew. Bison works now however, when running make menuconfig I get this:

In file included from /Users/scotts/esp/esp-idf/tools/kconfig/zconf.tab.c:2459:
/Users/scotts/esp/esp-idf/tools/kconfig/confdata.c:113:23: error: use of undeclared identifier 'PATH_MAX'
        static char fullname[PATH_MAX+1];
                             ^
/Users/scotts/esp/esp-idf/tools/kconfig/confdata.c:741:15: error: use of undeclared identifier 'PATH_MAX'
        char dirname[PATH_MAX+1], tmpname[PATH_MAX+1], newname[PATH_MAX+1];
                     ^
/Users/scotts/esp/esp-idf/tools/kconfig/confdata.c:741:36: error: use of undeclared identifier 'PATH_MAX'
        char dirname[PATH_MAX+1], tmpname[PATH_MAX+1], newname[PATH_MAX+1];
                                          ^
/Users/scotts/esp/esp-idf/tools/kconfig/confdata.c:741:57: error: use of undeclared identifier 'PATH_MAX'
        char dirname[PATH_MAX+1], tmpname[PATH_MAX+1], newname[PATH_MAX+1];
                                                               ^
/Users/scotts/esp/esp-idf/tools/kconfig/confdata.c:766:10: error: use of undeclared identifier 'newname'
        sprintf(newname, "%s%s", dirname, basename);
                ^
/Users/scotts/esp/esp-idf/tools/kconfig/confdata.c:766:10: error: use of undeclared identifier 'newname'
/Users/scotts/esp/esp-idf/tools/kconfig/confdata.c:769:11: error: use of undeclared identifier 'tmpname'; did you mean 'tmpnam'?
                sprintf(tmpname, "%s.tmpconfig.%d", newname, (int)getpid());
                        ^~~~~~~
                        tmpnam
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/secure/_stdio.h:47:50: note:
      expanded from macro 'sprintf'
  __builtin___sprintf_chk (str, 0, __darwin_obsz(str), __VA_ARGS__)
                                                 ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/secure/_common.h:39:54: note:
      expanded from macro '__darwin_obsz'
#define __darwin_obsz(object) __builtin_object_size (object, _USE_FORTIFY_LEVEL > 1 ? 1 : 0)
                                                     ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/stdio.h:188:7: note:
      'tmpnam' declared here
char    *tmpnam(char *);
         ^
In file included from /Users/scotts/esp/esp-idf/tools/kconfig/zconf.tab.c:2459:
/Users/scotts/esp/esp-idf/tools/kconfig/confdata.c:769:11: error: use of undeclared identifier 'tmpname'; did you mean 'tmpnam'?
                sprintf(tmpname, "%s.tmpconfig.%d", newname, (int)getpid());
                        ^~~~~~~
                        tmpnam
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/secure/_stdio.h:47:28: note:
      expanded from macro 'sprintf'
  __builtin___sprintf_chk (str, 0, __darwin_obsz(str), __VA_ARGS__)
                           ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/stdio.h:188:7: note:
      'tmpnam' declared here
char    *tmpnam(char *);
         ^
In file included from /Users/scotts/esp/esp-idf/tools/kconfig/zconf.tab.c:2459:
/Users/scotts/esp/esp-idf/tools/kconfig/confdata.c:769:39: error: use of undeclared identifier 'newname'
                sprintf(tmpname, "%s.tmpconfig.%d", newname, (int)getpid());
                                                    ^
/Users/scotts/esp/esp-idf/tools/kconfig/confdata.c:770:15: error: use of undeclared identifier 'tmpname'; did you mean 'tmpnam'?
                out = fopen(tmpname, "w");
                            ^~~~~~~
                            tmpnam
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/stdio.h:188:7: note:
      'tmpnam' declared here
char    *tmpnam(char *);
         ^
In file included from /Users/scotts/esp/esp-idf/tools/kconfig/zconf.tab.c:2459:
/Users/scotts/esp/esp-idf/tools/kconfig/confdata.c:772:4: error: use of undeclared identifier 'tmpname'; did you mean 'tzname'?
                *tmpname = 0;
                 ^~~~~~~
                 tzname
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/time.h:97:14: note:
      'tzname' declared here
extern char *tzname[];
             ^
In file included from /Users/scotts/esp/esp-idf/tools/kconfig/zconf.tab.c:2459:
[... 9 lines cut ...]
char    *tmpnam(char *);
         ^
In file included from /Users/scotts/esp/esp-idf/tools/kconfig/zconf.tab.c:2459:
/Users/scotts/esp/esp-idf/tools/kconfig/confdata.c:822:10: error: use of undeclared identifier 'newname'
                rename(newname, dirname);
                       ^
/Users/scotts/esp/esp-idf/tools/kconfig/confdata.c:823:14: error: use of undeclared identifier 'tmpname'; did you mean 'tmpnam'?
                if (rename(tmpname, newname))
                           ^~~~~~~
                           tmpnam
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/stdio.h:188:7: note:
      'tmpnam' declared here
char    *tmpnam(char *);
         ^
In file included from /Users/scotts/esp/esp-idf/tools/kconfig/zconf.tab.c:2459:
/Users/scotts/esp/esp-idf/tools/kconfig/confdata.c:823:23: error: use of undeclared identifier 'newname'
                if (rename(tmpname, newname))
                                    ^
/Users/scotts/esp/esp-idf/tools/kconfig/confdata.c:827:49: error: use of undeclared identifier 'newname'
        conf_message(_("configuration written to %s"), newname);
                                                       ^
/Users/scotts/esp/esp-idf/tools/kconfig/confdata.c:837:12: error: use of undeclared identifier 'PATH_MAX'
        char path[PATH_MAX+1];
                  ^
18 errors generated.

igrr, 2019-05-04:
@sshwarts Thanks for reporting this. The errors you are getting look similar to what was reported in https://github.com/espressif/esp-idf/issues/1281. Could you please run `make -C $IDF_PATH/tools/kconfig clean` and then `make menuconfig` again? Extra question, did you happen to update XCode recently?
(Also, let's continue troubleshoothing this in #1281 thread)

projectgus, 2020-07-14:
Closing as this was resolved in the linked issue, and in recent versions of ESP-IDF we no longer require a host compiler at all (to avoid these kind of issues - thankfully)

closed 2020-07-14 by projectgus