espressif/esp-idf #3413: timer ISR not able to run PWM modulation functions at interrupt

bobkasterop, 2019-05-03:
## Environment

- Development Kit:      [ESP32-DevKitC
- Kit version [v4]
- Module or chip used:  [ESP32-WROOM-3]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description
I am trying to run some functions inside an ISR triggered on a Timer. My problem is that when i call the functions inside of the ISR, the ESP just prints error messages and keep on restarting. I tried just setting a GPIO high inside the ISR instead and this works fine. So for some reason im not allowed to run the desired functions inside the ISR...
the functions i try to run is alternating PWM signals based on some calculations done in the function. its basically space vector modulation so a good amount of math which i have tried to cut computation time off as much as possible. Does anyone know why i cant run these functions in the ISR? 
### Expected Behavior
Functions running everytime interrupt occurs.
### Actual Behavior
ESP reboots 

### Code to reproduce this issue

```cpp
static void new_timer_init(int timer_idx,
    bool auto_reload, double timer_interval_sec)
{
    /* Select and initialize basic parameters of the timer */
    timer_config_t config;
    config.divider = TIMER_DIVIDER;
    config.counter_dir = TIMER_COUNT_UP;
    config.counter_en = TIMER_PAUSE;
    config.alarm_en = TIMER_ALARM_EN;
    config.intr_type = TIMER_INTR_LEVEL;
    config.auto_reload = auto_reload;
    timer_init(TIMER_GROUP_0, timer_idx, &config);

    /* Timer's counter will initially start from value below.
       Also, if auto_reload is set, this value will be automatically reload on alarm */
    timer_set_counter_value(TIMER_GROUP_0, timer_idx, 0x00000000ULL);

    /* Configure the alarm value and the interrupt on alarm. */
    timer_set_alarm_value(TIMER_GROUP_0, timer_idx, timer_interval_sec * TIMER_SCALE);
    timer_enable_intr(TIMER_GROUP_0, timer_idx);
    timer_isr_register(TIMER_GROUP_0, timer_idx, isr_handler,
        (void *) timer_idx, ESP_INTR_FLAG_IRAM, NULL);

    timer_start(TIMER_GROUP_0, timer_idx);
}

//timer handler
static void IRAM_ATTR isr_handler(void *para)
{
	int timer_idx = (int) para;

	uint32_t intr_status = TIMERG0.int_st_timers.val;
	TIMERG0.hw_timer[timer_idx].update = 1;
	uint64_t timer_counter_value = ((uint64_t) TIMERG0.hw_timer[timer_idx].cnt_high) << 32 | TIMERG0.hw_timer[timer_idx].cnt_low;

//Functions to run

     VF_control();
     SVM();

    /* Clear the interrupt
       and update the alarm time for the timer with without reload */
    if ((intr_status & BIT(timer_idx)) && timer_idx == TIMER_0) {
        TIMERG0.int_clr_timers.t0 = 1;
        timer_counter_value += (uint64_t) (TIMER_INTERVAL0_SEC * TIMER_SCALE);
        TIMERG0.hw_timer[timer_idx].alarm_high = (uint32_t) (timer_counter_value >> 32);
        TIMERG0.hw_timer[timer_idx].alarm_low = (uint32_t) timer_counter_value;
    } else if ((intr_status & BIT(timer_idx)) && timer_idx == TIMER_1) {
        TIMERG0.int_clr_timers.t1 = 1;
    } else {
        //evt.type = -1; // not supported even type
    }

    /* After the alarm has been triggered
      we need enable it again, so it is triggered the next time */
    TIMERG0.hw_timer[timer_idx].config.alarm_en = TIMER_ALARM_EN;
    //timer_flag = false;
}

```

negativekelvin, 2019-05-03:
You didn't provide the exception and backtrace.  You marked isr_handler as IRAM_ATTR but is every nested function call marked as IRAM_ATTR?

koobest, 2019-05-07:
@bobkasterop ,
When the interrupt occurs, is there a task that is erasing flash? In this case, if there is a floating point operation in ISR, it will cause crash. Maybe not caused by this reason.  Can you provide the log?

thanks !!

Alvin1Zhang, 2019-05-23:
@bobkasterop Thanks for reporting the issue. @negativekelvin Thanks for the notes. @bobkasterop Would you please help provide logs, exception and backtrace? Thanks.

Alvin1Zhang, 2019-06-05:
@bobkasterop Would you please help provide logs, exception and backtrace? Thanks.

Alvin1Zhang, 2019-07-15:
@bobkasterop Thanks for reporting the issue, will close this issue due to few updates. Feel free to reopen if you still have the issue, thanks.

closed 2019-07-15 by Alvin1Zhang