espressif/esp-idf #3414: FreeRTOS task affinity, bug or memory corruption?

ammaree, 2019-05-03:
## Environment

- Development Kit:      Various custom
- Kit version 2 different versions
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): various last 2+ months
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     Windows
- Power Supply:         external 3.3V

## Problem Description
Our application creates 9 tasks, all using tskNO_AFFINITY.
8 of the tasks consistently  gets configured correctly whereas 1 task "Sensor", and always the same task, gets created using core affinity = 1.

It does not matter where in the sequence the task get started, when moved from our 8 to 9 or 4, the result remain the same, gets CoreID=1

When we force the affinity to 0, it gets CoreID=0 and all the rest remain tskNO_AFFINITY.
When we force the affinity to 1, it gets CoreID=1 and all the rest remain tskNO_AFFINITY.

### Expected Behavior

The task remain configured with tskNO_AFFINITY if created with tskNO_AFFINITY.

### Code to reproduce this issue

`	if (wifi_mode == WIFI_MODE_STA) {
		xRtosTaskCreate(vSntpTask, "SNTP", sntpSTACK_SIZE, &sTSZ.usecs, sntpPRIORITY, NULL, tskNO_AFFINITY) ;
	}
	xRtosTaskCreate(vTaskControl, "Ctrl", controlSTACK_SIZE, NULL, controlPRIORITY, NULL, tskNO_AFFINITY) ;
	vRtosWaitRunState(taskCONSOLE) ;					// ONLY when control task done init do we continue
#if		(configCONSOLE_HTTP == 1)
	xRtosTaskCreate(vTaskHttp, "HTTP", httpSTACK_SIZE, NULL, httpPRIORITY, NULL, tskNO_AFFINITY) ;
#endif
	if (wifi_mode == WIFI_MODE_STA) {
#if		(configCONSOLE_TELNET == 1)
		xRtosTaskCreate(vTaskTelnet, "TNET", tnetSTACK_SIZE, NULL, tnetPRIORITY, NULL, tskNO_AFFINITY) ;
#endif
		xRtosTaskCreate(vTaskMqttTx, mqttTX_NAME, mqttTX_STACK_SIZE, NULL, mqttTX_PRIORITY, NULL, tskNO_AFFINITY) ;
		xRtosTaskCreate(vTaskMqttRx, mqttRX_NAME, mqttRX_STACK_SIZE, NULL, mqttRX_PRIORITY, NULL, tskNO_AFFINITY) ;
#if		(configHAL_XXX_XXX_OUT > 0)
		xRtosTaskCreate(vTaskActuator, "Actuator", ACTUATE_STACK_SIZE, NULL, ACTUATE_PRIORITY, NULL, tskNO_AFFINITY) ;
#endif
		xRtosTaskCreate(vTaskEvents, "Events", eventsSTACK_SIZE, NULL, eventsPRIORITY, &EventsHandle, tskNO_AFFINITY) ;
//		xRtosTaskCreate(vTaskSensors, "Sensors", sensorsSTACK_SIZE, NULL, sensorsPRIORITY, NULL, tskNO_AFFINITY) ;
		xRtosTaskCreate(vTaskSensors, "Sensors", sensorsSTACK_SIZE, NULL, sensorsPRIORITY, NULL, 0) ;
	} else {
		SL_WARN("AP mode SNTP/mqttXX/Events/Sensors[Actuators/TNET] not started") ;
	}

 and the function doing the creation is:

int32_t	xRtosTaskCreate(TaskFunction_t pxTaskCode,
						const char * const pcName,
						const uint32_t usStackDepth,
			            void * pvParameters,
						UBaseType_t uxPriority,
						TaskHandle_t * pxCreatedTask,
						const BaseType_t xCoreID) {
#if		(ESP32_PLATFORM == 1)
	#if		defined(CONFIG_FREERTOS_UNICORE)
	UNUSED(xCoreID) ;
	int32_t iRetVal = xTaskCreatePinnedToCore(pxTaskCode, pcName, usStackDepth, pvParameters, uxPriority, pxCreatedTask, 0) ;
	#else
	int32_t iRetVal = xTaskCreatePinnedToCore(pxTaskCode, pcName, usStackDepth, pvParameters, uxPriority, pxCreatedTask, xCoreID ) ;
	#endif

#else
	UNUSED(xCoreID) ;
	int32_t iRetVal = xTaskCreate(pxTaskCode, pcName, usStackDepth, pvParameters, uxPriority, pxCreatedTask) ;

#endif
	return iRetVal ;
}
`
## Debug Logs

```
------------------------------------- tskNO_AFFINITY ---------------------------------
 # | Pc/Pb |-TaskName-| S | LoStk|MCU| %Util| System Ticks |Stack Base|-Task TCB-|
 1 | 22/22 |esp_timer | B | 2,976| 0 |  0.10|       154,826|0x3ffaeaf0|0x3ffaf8f4|
 2 | 24/24 |ipc0      | B | 1,148| 0 |  0.00|         2,359|0x3ffbcda4|0x3ffafc10|
 3 | 24/24 |ipc1      | B | 1,096| 1 |  0.01|        20,398|0x3ffbd3a8|0x3ffafddc|
 6 |  0/ 0 |IDLE0     | P |   792| 0 | 86.19|   124,263,741|0x3ffbfac8|0x3ffbffcc|
 7 |  0/ 0 |IDLE1     | P |   788| 1 | 99.39|   143,288,067|0x3ffc013c|0x3ffc0640|
 8 |  4/ 4 |Tmr Svc   | B | 2,197| 0 |  0.00|         8,859|0x3ffc08ac|0x3ffc16b4|
12 | 18/18 |tiT       | B | 2,304| X |  0.40|       587,489|0x3ffc1da0|0x3ffc2ba4|
13 | 20/20 |sys_evt   | B | 1,568| 0 |  0.00|         1,004|0x3ffc30a0|0x3ffc3aa4|
14 | 23/23 |wifi      | B | 1,396| 0 |  1.45|     2,093,857|0x3ffc451c|0x3ffc5320|
15 |  2/ 2 |SNTP      | B | 1,172| X |  0.00|         7,087|0x3ffccfac|0x3ffcd9b0|
16 |  2/ 2 |Ctrl      | B |   780| X |  0.68|       988,783|0x3ffcdb20|0x3ffcef24|
17 |  4/ 4 |HTTP      | B | 1,420| X |  0.02|        32,606|0x3ffcf634|0x3ffcff38|
18 |  3/ 3 |TNET      | R |   892| X |  0.42|       606,191|0x3ffdc2c4|0x3ffd04ac|
19 |  5/ 5 |mqttTX    | B | 3,492| X |  0.01|        26,102|0x3ffdcdfc|0x3ffde100|
20 |  5/ 5 |mqttRX    | B | 1,752| X |  0.24|       353,558|0x3ffd09f8|0x3ffdf314|
21 |  8/ 8 |Actuator  | B | 2,196| X |  2.42|     3,496,107|0x3ffd16fc|0x3ffdf54c|
22 |  7/ 7 |Events    | B |   960| X |  0.04|        60,432|0x3ffd2300|0x3ffdf72c|
23 |  6/ 6 |Sensors   | B | 1,436| 0 |  8.53|    12,304,065|0x3ffd2e84|0x3ffdf9fc|
Util:  0=10.08%  1=0.01%  X=4.23%  T=14.32%

------------------------------------ Forced to '0' --------------------------------------
15 |  2/ 2 |SNTP      | B | 1,044| X |  0.02|         7,060|0x3ffccfac|0x3ffcd9b0|
16 |  2/ 2 |Ctrl      | B |   780| X |  3.07|       995,610|0x3ffcdb20|0x3ffcef24|
17 |  4/ 4 |HTTP      | B | 1,424| X |  0.01|         6,376|0x3ffcf690|0x3ffdbcd4|
18 |  3/ 3 |TNET      | R | 1,296| X |  0.05|        17,662|0x3ffdc250|0x3ffdbe44|
19 |  5/ 5 |mqttTX    | B | 3,500| X |  0.03|         9,817|0x3ffdcb54|0x3ffd0518|
20 |  5/ 5 |mqttRX    | B | 1,948| X |  0.23|        76,933|0x3ffd0af4|0x3ffdfb08|
21 |  8/ 8 |Actuator  | B | 2,200| X |  2.06|       669,612|0x3ffd17f8|0x3ffdfc78|
22 |  7/ 7 |Events    | B |   964| X |  0.09|        31,320|0x3ffd23fc|0x3ffdfde8|
23 |  6/ 6 |Sensors   | B | 1,472| 0 |  7.30|     2,363,538|0x3ffd2f80|0x3ffdde58|
Util:  0=11.62%  1=0.06%  X=5.97%  T=17.65%

------------------------------  Forced to '1'  -------------------------------------------
15 |  2/ 2 |SNTP      | B | 1,172| X |  0.00|         7,024|0x3ffccfac|0x3ffcd9b0|
16 |  2/ 2 |Ctrl      | B |   780| X |  0.86|     1,037,864|0x3ffcdb20|0x3ffcef24|
17 |  4/ 4 |HTTP      | B | 1,408| X |  0.02|        24,545|0x3ffcf690|0x3ffcff94|
18 |  3/ 3 |TNET      | R |   880| X |  0.18|       224,050|0x3ffdc250|0x3ffd0508|
19 |  5/ 5 |mqttTX    | B | 3,488| X |  0.03|        38,981|0x3ffdcbb0|0x3ffddeb4|
20 |  5/ 5 |mqttRX    | B | 1,940| X |  0.22|       264,849|0x3ffdf1cc|0x3ffd0af4|
21 |  8/ 8 |Actuator  | B | 2,204| X |  2.88|     3,461,696|0x3ffd0c64|0x3ffd1868|
22 |  7/ 7 |Events    | B | 1,225| X |  0.05|        71,863|0x3ffd19d8|0x3ffd255c|
23 |  6/ 6 |Sensors   | P | 1,476| 1 |  8.67|    10,394,561|0x3ffd26cc|0x3ffd3194|
Util:  0=1.60%  1=8.68%  X=4.56%  T=14.84%
```

igrr, 2019-05-04:
Tasks get pinned automatically when FPU is used, could that be the case here? See https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/freertos-smp.html#floating-point-aritmetic

ammaree, 2019-05-05:
Absolutely the case, thanks for clarifying so quickly.

closed 2019-05-05 by ammaree