espressif/esp-idf #3416: spiffsgen.py incorrect path delimiter on windows

vetaldudnik, 2019-05-03:
## Problem Description

spiffsgen.py generates spiffs image with incorrect path delimiters on windows

### Expected Behavior

Expected filename in spiffs filesystem is /spiffs/path/to/file.jpg

### Actual Behavior

Actual filename is /spiffs\path\to\file.jpg

renzbagaporo, 2019-05-05:
Thanks for reporting this issue @vetaldudnik! A fix for this has been submitted to our internal repo. In the meantime, you can use the patch:

[0001-spiffs-fix-path-issue-with-using-spiffsgen-on-Window.patch.zip](https://github.com/espressif/esp-idf/files/3145105/0001-spiffs-fix-path-issue-with-using-spiffsgen-on-Window.patch.zip)

renzbagaporo, 2019-05-05:
Sorry, didn't notice the associated PR for this issue. Will be happy to accept that instead! :smile:

Alvin1Zhang, 2020-01-22:
@vetaldudnik Thanks for reporting. Feel free to reopen if the issue still happens. Thanks.

closed 2020-01-22 by Alvin1Zhang