espressif/esp-idf #3417: Build failed on v4.0-dev-459-gba1ff1692

vetaldudnik, 2019-05-03:
## Environment

- IDF version: v4.0-dev-459-gba1ff1692
- Build System: Make
- Operating System: Windows

## Problem Description

Build failed on v4.0-dev-459-gba1ff1692.
Executing python -m pip install --user -r $IDF_PATH/requirements.txt does not solve the problem

### Steps to repropduce

1. Update esp-idf to v4.0-dev-459-gba1ff1692
2. Run make clean or make clean 

## Debug Logs

Console output after make all:

The following Python requirements are not satisfied:
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22
Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.
Alternatively, you can run "D:/Programs/msys/mingw32/bin/python.exe -m pip install --user -r D:/workspaces/esp32_workspace/esp-idf\\requirements.txt" for resolving the issue.
make: *** No rule to make target 'check_python_dependencies', needed by 'all'.  Stop.
"make -j4 all" terminated with exit code 2. Build might be incomplete.

negativekelvin, 2019-05-04:
https://github.com/espressif/esp-idf/issues/3346

dobairoland, 2019-05-04:
Hi @vetaldudnik.

What is the produced output if you run the following command?
```
D:/Programs/msys/mingw32/bin/python.exe -m pip install --user -r D:/workspaces/esp32_workspace/esp-idf\requirements.txt
```

vetaldudnik, 2019-05-04:
Hi, @dobairoland.

Hmm... I think I understand what the problem is.

Command
`python -m pip install --user -r D:/workspaces/esp32_workspace/esp-idf/requirements.txt`
must be run under bash or sh. In this case some python modules installed in home folder like this:
**d:\programs\msys\home\user_name\\.local\lib\python2.7\site-packages**
but if this command run from Windows CMD home dir is
**c:\users\user_name\\.local\lib\python2.7\site-packages**

I just pasted this command into Windows CMD and install some packeges into wrong home dir

Maybe on Windows when any requirements are not satisfied we need advise user to run

**bash -c "D:/Programs/msys/mingw32/bin/python.exe -m pip install --user -r D:/workspaces/esp32_workspace/esp-idf/requirements.txt"** ?

dobairoland, 2019-05-06:
Hi @vetaldudnik!

Thanks for sharing your experiences. Unfortunately, I'm unable to reproduce the issue. If I use MSYS' Python then the packages go to the other directory (as you pointed out) but are available after that (i.e. the requirements are met). I tried with and without the `--user` option in various combination with my globally installed Python interpreter. 

Isn't it possible that your `python` used from Windows command line is not the MSYS one? Could you please share the output of `where python`? Is it possible that if you run `python -m pip ...` then not the MSYS' Python interpreter is invoked.

Could you please also share the outputs of the following commands? 
```
echo %PATH%
D:/Programs/msys/mingw32/bin/python.exe -c "import site; print(site.getuserbase())"
python -c "import site; print(site.getuserbase())"
```

vetaldudnik, 2019-05-06:
`where python` invoked from cmd and bash gives the same output:
`D:\Programs\msys\mingw32\bin\python.exe`

so I have one python interpreter

`echo %PATH%`
`C:\Program Files (x86)\STMicroelectronics\st_toolset\asm;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utility\ST-LINK Utility;C:\Users\Udtech-785\AppData\Local\Microsoft\WindowsApps;D:\Programs\msys\usr\bin;D:\Programs\msys\mingw32\bin;D:\Programs\msys\opt\gcc-arm-none-eabi\bin;D:\Programs\msys\opt\gcc-avr\bin;D:\Programs\msys\opt\gcc-xtensa-esp32-elf\bin;D:\Programs\msys\opt\openocd\bin;`

`D:/Programs/msys/mingw32/bin/python.exe -c "import site; print(site.getuserbase())"`
and 
`python -c "import site; print(site.getuserbase())"`
invoked from windows cmd gives the same output:
`C:\Users\user_name\.local`

but if I invoke this from bash the output is:
D:\Programs\msys\home\username\.local

vetaldudnik, 2019-05-06:
Maybe it matters, before using esp-idf I installed clear msys2 environment from https://www.msys2.org/
then download gcc-xtensa-esp32-elf from espressif.com copy it to /opt and then I did these steps:

1. pacman --noconfirm -Syu
2. pacman --noconfirm -S --needed gcc git make ncurses-devel flex bison gperf vim gnu-netcat mingw-w64-i686-python2 mingw-w64-i686-python2-pip mingw-w64-i686-python2-cryptography
3. python -m pip install --upgrade pip
4. python -m pip install --user -r $IDF_PATH/requirements.txt

dobairoland, 2019-05-06:
Thank you for providing all of these useful information. I don't see any suspicious thing beside that you've ugraded `pip`. I've seen in many cases that it can break the Python installation in so many ways. Note that here https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html#toolchain-setup is a ZIPed MSYS environment which we create with https://github.com/espressif/esp-idf/blob/master/tools/windows/windows_install_prerequisites.sh. It contains commands very similar to yours except that we don't upgrade `pip`.

There is also an installer (https://docs.espressif.com/projects/esp-idf/en/latest/get-started-cmake/windows-setup.html#esp-idf-tools-installer) which is much better for Windows command line and MSYS will not be supported in the future. Further improvement for handling Python packages are coming soon.

If I understand it correctly then your issue has been solved by invoking `bash`. I will not recommend the use of this generally because (as I outlined) Windows command line should not depend on MSYS. I'm closing this issue because I don't think that there is anything which can be done to prevent such issue in the future.

closed 2019-05-06 by dobairoland